procount/pinn #408: Can only replace an os with an installed os

Ae3NerdGod, 2020-08-11:
I went to use the replace feature in maintenance mode, and it only allows me to replace an OS with OSs that are already installed on the sdcard. Is this purposefully done? That's kinda the exact opposite functionality I expected it to be, does it reinstall the OS cleanly, Or copy direct from the partitions of the SD card?

I'd love the ability to install a new OS non-destructively, as in leave the other OS's alone and "replace" one with a new OS.

I guess I could install a project space, untar the os into that, and then clone the project space, but I don't see why pinn couldnt achieve this on its own, not to mention then I'd have to make a modified partition-setup script to work from userspace instead of recovery.

procount, 2020-08-11:
That is not correct.
Did you follow the instructions in the readme.pimn?
First select the OSes you want to install from the main menu.
Then go to maintenance and select the OSes you want to replace.
Now when you click replace you can select the new is to replace each existing one in the dropdowns.

Ae3NerdGod, 2020-08-11:
Ahhhhhhhhh, that makes sense. My bad, so it was only showing the ones I had installed, Because they were checked from being installed. Just curious, is that to avoid the "new OS" menu being huge?

procount, 2020-08-11:
Yes, also just trying to have a consistent user interface.

closed 2020-08-12 by Ae3NerdGod