procount/pinn #409: Multithreaded decompression

Ae3NerdGod, 2020-08-11:
Finally got pinn to compile, made a small change but it performs so we'll i thought I'd bring it up.

I added -T 0 to the xz command. 

On sdcard it doubles the speed of most os installs: 7-8MB/s increases to around 12-19MB/s

On usb SSD, it more than trippled for me: 7-10 originaly, 24-33 after the multi thread parameter

Those are both with stock clocks on pi4b8gb, they get a few mb/s higher with overclocking.

The experience has went from taking an hour to install two OSs to doing 4, easy, in 10 minutes. I think it's well worth adding multi threading to the code, and I can look up the needed parameters for each command, test them out (I only did xz for now), and make a pull request if you're interested.

procount, 2020-08-11:
I already use multithreaded for compression. Surprised it makes such a difference for decompression too. Is that from a local USB or from an internet download?

lurch, 2020-08-11:
Sounds good, as long as it still works on the single-core Pis like the Zero.

I'd always assumed that the SD-card write-speed was the bottleneck, but with the Pi 4B everything is much faster :)

Ae3NerdGod, 2020-08-11:
@procount local area network server goes that fast, GitHub is going my normal download speeds, sourceforge also unless I get a overloaded or far away mirror.

I usually don't install from USB

And in your defense, any pi2 or lower probably wouldn't show much of a speed up on stock clocks, at least not with xz

Ae3NerdGod, 2020-08-11:
@lurch at least for xz the 0 in -T 0 means "use the amount of cores available" so it should not affect, I do have a "fast" build if you'd like to try it. I don't have a zero

http://bowtie-ent.com/pinng/pinn-fastxz-V3.3.4.2.zip

Again, I only sped up xz for testing on that build. #lazy

Ae3NerdGod, 2020-08-11:
I've noticed it's faster than just a tar also, I think because more data gets to the pi faster technically, and gets put in ram first before it gets to bsdtar. Xz is able to feed bsdtar faster than my internet/lan

procount, 2020-08-24:
I've added this to v3.3.4.3 just released. However, it only seems to provide an improvement with files that have been compressed in a multi-threaded manner, otherwise it just uses a single thread as before.

closed 2020-08-25 by procount

lurch, 2020-08-25:
> it only seems to provide an improvement with files that have been compressed in a multi-threaded manner, otherwise it just uses a single thread as before.

Interesting.

I did a bit of googling and found [this old answer](https://stackoverflow.com/questions/22244962/multiprocessor-support-for-xz) which mentions "Threaded decompression ... will only work on files that contain multiple blocks with size information in block headers. All files compressed in multi-threaded mode meet this condition, but files compressed in single-threaded mode don't..."
And it seems (from related search results) that files compressed in multi-threaded mode actually come out slightly larger than files compressed in single-threaded mode. So it's the classic do-you-optimise-for-speed (of decompression) or do-you-optimise-for-size (of download) tradeoff :winking_face: