procount/pinn #410: retro.sh not working in PINN?

msinn, 2020-08-11:
Hi,

I read in one of the issues that retro.sh isn't working with PINN.
I wanted to use it to install software from the internet after the setup of the operating system was done. Is there another way to achieve this?

procount, 2020-08-12:
Which issue?
Retro.sh is for a very specific use case and may not be what you need anyway.
Have you read https://github.com/procount/pinn/blob/master/noobsconfig.md ?

msinn, 2020-08-12:
I understood a comment from you from Nov 2017 in that direction. In between I found your comment

> > Where do I put retro.sh? The noobsconfig documentation says it goes in the root of the RECOVERY partition, but that might not be accurate in PINN.
>
> It's the same. (I assume retro.sh still works, haven't tried it in a while).

in issue #251.

I am gonna try it.

> Have you read https://github.com/procount/pinn/blob/master/noobsconfig.md ?

Yes, I have read it but found no reference to retro.sh. I found retro.sh in the NOOBSConfig documentation.

> Retro.sh is for a very specific use case and may not be what you need anyway.

What I want is a script that is run on (first) boot after the installation of PINN is completed. In this script I need to do some ``git clone`` and some configuration afterwards.

procount, 2020-08-12:
Then you can forget retro.sh, it is not needed.

You will need to write the script yourself and figure out how you are going to run it on first boot (many ways).
Once you have done that, create a "flavour" that will install your script and first boot configurations from a tar file as the noobs config.md explains.

msinn, 2020-08-16:
Thanks procount,

I managed to get my files copied an using rc.local to get an installation script running afterwards. I have managed with the RaspPiOS on the SD card. 

I have tried to set up a repository (analog to your pinn-os) and install from there. That works fine for a standard RaspPiOS, but I have no idea how to integrate my ``<flavour>_root`` when setting up for a network install.

Is it possible (and if yes: How?)

msinn, 2020-08-16:
I noticed something else in between:

copying files by ``<flavour>_root.txt`` works great, but now I wanted to change the user and group of a file:

```
inst_shng.sh /home/pi 0755 pi pi
```

The file gets copied to /home/pi, the attributes get set, but the file still belongs to root:root... Did I get something wrong?

procount, 2020-08-24:
I fixed a small bug relating to the .txt file in v3.3.4.3 Just released. Maybe it will help?

msinn, 2020-09-02:
In the mean time I got it working using v3.3.4.3 and bits and peaces of information I gathered from your readmes, multiple issue threads and a lot of testing.

It took me some time and I was running in circles because the information peaces I found only described part of the solution and completing it with my thoughts how it would work didn't work out. Some peaces of information (in my mind) even contradicted information that I found in other places.

What I have running now is the following:
- In a github repo I am hosting my own repo_list.json ane an os_list_v3.json.
- In the os_list_v3.json I have the url to load raspian from downloads.raspberrypi.org and defined the flavours I want.
- On the sd card I have the customization in the os folder (It took me some time to find out that the custimization files belonged directly in the os folder and not in a subfolder and that configpath had to be /mnt/os/)

What I have still to do:
- For now it works, but I want the customization on github too (instead of putting it on the sd card).

Thanks a lot for the good tool. The documentation needs some improvement though. If I can help with that, please let me know.

procount, 2020-09-02:
I'm glad you have it partially working.
It should work as you wanted it to, but I haven't had anything to test it thoroughly with. If you have a public URL you can provide me with, I can try testing it with your customisation.
If you are only hosting one os_list_v3.json file, then you can dispense with the repo_list.json and just reference the os_list_v3.json with an alt_image_source parameter instead.