procount/pinn #412: Install new OS without cleaning the rest

ciberharry97, 2020-08-15:
Is it possible to install a new OS while keeping the rest untouched?

procount, 2020-08-16:
Yes. Have a read of README_pinn.md and look at the replace feature.

ciberharry97, 2020-08-16:
The thing is I don’t want to replace one of my os and I didn’t install project spaces. So is it possible to add like new partitions for the free space on the SD card?

ciberharry97, 2020-08-16:
Maybe doing a backup of the os installed and then reinstall them adding the new os?

procount, 2020-08-16:
Yes, that's about the only way.

SpacingBat3, 2020-08-18:
You can always manually resize some of the partitions to make a free space and then manually install any os you wish from the *.img file (you just to create 2 partitions (BOOT and ROOT) and add the proper entry in your /settings/installed-os.json config file (you can do this from rescue shell or after mounting SETTINGS partition).
I've got no problems then to learn by myself how entries in this config works. BTW, you can use this method to install most of the OSes unsupported by PINN now like TwisterOS (I did once, even added an icon).

@ciberharry97, I suggest now closing this issue (unless you got any other problems with adding new OS without removing exising one).

ciberharry97, 2020-08-18:
Okey thanks! @SpacingBat3, I'll try it manually then.

closed 2020-08-18 by ciberharry97

lurch, 2020-08-18:
> BTW, you can use this method to install most of the OSes unsupported by PINN now like TwisterOS (I did once, even added an icon).

As long as you're manually able to perform the appropriate steps that `partition_setup.sh` normally does (e.g. editing `/boot/cmdline.txt` and `/etc/fstab` to point to the correct partition numbers).