procount/pinn #413: Running PINN on USB drive with GUID Partition Table on the Raspberry Pi 4

SpacingBat3, 2020-08-15:
I'd like to install PINN on the GPT. As PINN by itself boots up correctly on the GPT, it doesn't allow me to boot into any other OSes (I've got errors such as: *GPT: No bootable partition found.* or *Partition 6 \ Can't open as FAT*).

# I installed it on the GPT by this _tricky_ way:
- install PINN normally on the HDD;
- save RECOVERY and SETTINGS partitions as img/iso file (using `dd` command);
- reformat partition table to the gpt;
- mount RECOVERY and SETTINGS images so I've got an access to its files
- create some empty partitions on the disk from /dev/sdx1 to/dev/sdx5
- formatt /dev/sdx1 as FAT partition and /dev/sdx5 as ext4 partition
- copy all files from RECOVERY image to /dev/sdx1 partition and all files from SETTINGS partition to the /dev/sdx5 partition.
- at this state PINN should boot correctly, but it will prompt to reformat drive. **Doing this breaks the recovery partition, making disk unusable to boot.**;
- can't boot installed OSes on the drive (you can "install" them by copping from the PINN when it's installed normally or by installing them manually using img file);
- not able to install oses anymore trought the PINN installed on the GPT and even if so, you won't be able to boot them correctly.

# Other questions:
- why the PINN forces the usage of MBR partition scheme and doesn't try to install itself on the GPT?
- why it can't boot other OSes even if it is possible to boot the GPT partitions on the RPi (otherwise PINN wouldn't boot)?

procount, 2020-08-15:
Neither NOOBS nor PINN understand GPT. They were only designed for MBR and would need modifying to work with GPT.

Ae3NerdGod, 2020-08-16:
I'm also pretty sure earlier versions of the pi can't read gpt partition schemes at boot time. 

Not to mention there is no real advantage to using gpt in this situation. Extended partitions solve the limitations of MBR.

Ae3NerdGod, 2020-08-16:
Ps @SpacingBat3 , hybrid partitions work to some extent. Occasionally won't boot recovery or installed OS the first time.

SpacingBat3, 2020-08-17:
@Ae3NerdGod Raspberry Pi __can__ boot from the GPT, as PINN boots itself correctly, without any issues. The problem is that it can't boot the other OSes (aka "switch" the os to the one I've installed). I didn't tested if it's possible to boot one OS on the GPT, but if the PINN or NOOBS could, others should also.

The reason I really wanted to boot from the GPT is that I've also wanted my HDD to not only use it for the Pi, but rather for all kinds of different machines, including normal PCs with motherboard that supports UEFI. MBR now limits me to "dual boot" with Legacy BIOS only. I've also got used to the GPT "Partitions names", which allowed me to describe any of the partitions without using the shortcuts (as labeling on some filesystems might have character restrictions), but that's obviously not the main reason.

procount, 2020-08-18:
PINN certainly won't be able to install OSes to a GPT because it doesn't know how to create GPT partitions.
However, if you manage to manually install them on GPT, then PINN _might_ be able to boot them, IFF you can reference the partitions using /dev, label name, or partuuid. GPT references won't work.

Installing PINN to an SD card with Boot partitions on there and root partitions on the GPT might be more successful 🤷‍♂️

SpacingBat3, 2020-08-18:
Thank you all for your messages. As the result, here comes small FAQ and my answers to light up this entire situation.
## My Answers:
> PINN certainly won't be able to install OSes to a GPT because it doesn't know how to create GPT partitions.

Yep. That's what I understand. It would in this situation reformat entire disk to use MBR partition table. I've also noticed it could break itself when trying to reformat it after I installed it manually on the GPT

> However, if you manage to manually install them on GPT, then PINN __might__ be able to boot them, IFF you can reference the partitions using /dev, label name, or partuuid. GPT references won't work.

I actually used "the simplest" reference (/dev/sdxX) and PINN tries to switch the partition (as I get the message: PARTITION X), but the bootloader fails recognising it as bootable (or even saying it's not FAT even if it is in real).

> Installing PINN to an SD card with Boot partitions on there and root partitions manually installed/copied to the GPT might be more successful

That's working, I've used this partitioning sheme for quite long before the USB-MSD gone stable, but now I'm experimenting to boot without my microSD, as it's performance is horrible. Also I really wanted to use it for my phone, as it's

### Things I didn't check:
- if NOOBS would work (propably not, as NOOBS is similar to PINN),
- if booting 1 OS would work (without PINN or NOOBS; in my opinion it should as PINN did)
- if BerryBoot would work (propably yes, as it's not using multiple boot partitions, simplifying there boot process – but it's not actively developed and even I'm not sure if it support USB boot).

### @Ae3NerdGod: Reasons why not Hybrid Partition Table (excluding yours):
- Using projective MBR ~~might~~ break the PINN if you wish to still boot it as GPT partition (i.e. *not* adding it to the MBR scheme);
- Limited to 3 partitions, so the installation scheme can only look like: 2 os + PINN BOOT (better use with BerryBoot as it doesn't require having multiple boot paritions since it's real bootloader);
- Currently no way to bypass the limitation above as GPT doesn't support creating extended partitions (it's in theory possible to create one, but practically there's any easy way to do that and even if there was, there's no guarantee that it is possible to make everything working as it should);

## Other thoughts, partially related with the problem:
A this state, my *crazy* idea is waiting for the Pi to get the GPT booting fixed (as it's *kinda boots, kinda not* at this state) or the [UEFI Firmware](https://github.com/pftf/RPi4) made by the community will get more stable than now – either way, this issue will stay open here. I'm also going to check if NOOBS has similiar issue posted and if it's not, I'm going to repost it and mention it there. I'm not sure if this issue should be closed if NOOBS might have similar problems, so it stays here open now even if it's not going to be fixed by @procount. 

Anyway, thank you all for your answers and interest.

lurch, 2020-08-18:
AFAIK it's only the Pi4 that supports booting from a GPT-partitioned disk? And even then I think that's only a "feature" that was added by a [bootloader EEPROM update](https://github.com/raspberrypi/rpi-eeprom) created _after_ the Pi4 was first released.
One of the key principles of NOOBS and PINN is to seamlessly work on all Pis, so they're unlikely to natively support GPT as it would break backwards-compatibility.
Just in case you're not aware, PINN is a fork of NOOBS, and NOOBS was originally developed back in the Pi1 days, which is why it's all so MBR-specific :wink:  (read e.g. [this](https://github.com/raspberrypi/noobs/wiki/Standalone-partitioning-explained) and [this](https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained) )

> The reason I really wanted to boot from the GPT is that I've also wanted my HDD to not only use it for the Pi, but rather for all kinds of different machines, including normal PCs with motherboard that supports UEFI.

Be **very** careful with doing that. I'm not sure how PINN works these days, but I know that e.g. NOOBS is only able to "reinstall" an OS by wiping the entire drive and then installing things afresh. And I imagine you'd get a bit upset if NOOBS or PINN wiped out the OSes that you were also using on other PCs :laughing: 

If you do want to continue exploring this journey, you'll probably want to read https://github.com/raspberrypi/noobs/issues/551 and https://github.com/raspberrypi/rpi-eeprom/issues/206 and see if you're able to "reboot" from different partitions on a GPT disk.

Ae3NerdGod, 2020-08-18:
>Limited to 3 partitions
>Currently no way to bypass the limitation above as GPT doesn't support creating extended partitions

What?
Then what's the purpose of hybrid partitions, because that exactly what gparted did when it went all hybrid; allowed the MBR scheme I had to have 6 partitions (counting the extended as one partition) and a total of like 20 something partitions altogether.

I am also like 99% certain the rpi4 is the only one that can boot from gpt (at launch at least) because when I had a rpi2 I could not get an sd card I knew worked to boot for the life of me till I realized it was gpt and booted immediately after.

I'm still very confused why this would even be necessary? Just use MBR and an extended partition? I mean... I'm not calling you stupid or anything but ever heard of K.I.S.S? 
Keep
It
Simple,
Stupid