procount/pinn #417: Resize partitions

realOgenki, 2020-08-27:
One of main difficulties is manage partitions on SD cards easier. With PINN you can define it at first but I don't know change later. I think it will be a great enhancement if you can resize, delete or create partitions from PINN manager.

procount, 2020-08-27:
Yes, it would. But nether NOOBS (and hence nor PINN) were designed with this in mind.
It is quite difficult to do, so I doubt it would happen.
Your best bet is to use something like gparted to manage partitions, but if you alter the existing partition numbers, you would have to manually apply any fixups yourself.

realOgenki, 2020-08-27:
So the number of each of the partitions is important but not their size. It is right?
I thought that modifing recovery.cmdline file like [PINN System Resize](https://pinn.mjh.nz/) could be implemented.

procount, 2020-08-27:
Your link points to Matt's PINN resize webpage. But this only allows you to set partition sizes at installation. You cannot change the sizes later without manual intervention. For that you need to know how it all works.

realOgenki, 2020-08-27:
Ok, I will try to resize with gparted but now with your help I have a security backup. Bests regards.

closed 2020-10-14 by procount