procount/pinn #418: Replacing OS doesn't make correct partition labels.

Ae3NerdGod, 2020-09-07:
Normally, when you install multiple OS with the same labels for partitions, ie both are named boot and root, they would be changed to boot0 and root0 if another partition shares the name. When replacing an OS, it doesn't do that. At least not on usb.

procount, 2020-09-07:
Looking through the code, REINSTALL will re-use the existing partition labels, but REPLACE should do the same thing as a fresh install and rename partition labels if there is a name clash. USB shouldn't be any different. Can you confirm if this works properly on an SD card and the bug is restricted to USB? That should narrow down my testing and bug huntng a bit.

Ae3NerdGod, 2020-09-07:
that was the next thing i planned to do but my sandisk sd card got really hot and broke in half in my card reader??? never happened to me before... idk... think it was defective. getting a new one today, will get back to ya.

Ae3NerdGod, 2020-09-08:
@procount confirmed also happens on sdcard.

procount, 2020-09-08:
OK. I confirmed this. Very odd. I'm sure it used to work. Time for some debugging....

procount, 2020-09-09:
Resolved. Will be fixed in next version.

closed 2020-09-10 by procount