procount/pinn #420: img.XX support?

Ae3NerdGod, 2020-09-17:
Ive been experimenting with windows 10 arm, and I wanted to try it out on pinn. well I'm realizing tar doesn't support NTFS completely (something about acl) and I'm wondering if it would be easier to just add support for .img and compressed .img.xx files.. however I realize the issue of having to add partition resizing logic to the os install process. The only other alternative I see is using the partition setup script to install windows 10.

 Thoughts? Alternative suggestions?

Ae3NerdGod, 2020-09-17:
Ayeeee 420 blaze it

lurch, 2020-09-17:
IIRC NOOBS (and therefore presumably also PINN) does support "raw" partition images - have a look at how RISCOS works.

procount, 2020-09-17:
@lurch - You beat me to it!
Yes, PINN supports raw partition image files (with all manner of compressions: gz, xz, zip, lzo, bz2 etc), Partclone images (which may be appropriate for Windows :man_shrugging: ) as well as the usual tar files.
I only produce OSes that use the tar files because it allows more flexible partition sizes.
I guess raw images must be installed to a partition that is at least as big as the image, but would also need to resize themselves to fill the partition space if that was bigger.

Ae3NerdGod, 2020-09-18:
> have a look at how RISCOS works

> Yes, PINN supports raw partition image files (with all manner of compressions: gz, xz, zip, lzo, bz2 etc),

hmm, i was getting an error about the wrong file extension like before.... i guess ill double check the images and be strict with file/folder names and try again?

> , but would also need to resize themselves to fill the partition space if that was bigger. 

would it be possible to export the calculated "slice" (thats what the json calls it i think) size as a variable to the partition-setup script, so it can handle resizing?

lurch, 2020-09-18:
The partition-setup script gets told which partition-numbers are being used, and it seems to me that any tool which is advanced enough to be able to resize the filesystem on a partition, should find it easy to also read the MBR / EBR to work out how big the partition actually is?

procount, 2020-09-18:
> would it be possible to export the calculated "slice" (thats what the json calls it i think) size as a variable to the partition-setup script, so it can handle resizing?

Normally this is done by injecting a run-once boot script into the OS itself that runs resizefs or similar. No need to pass it any parameters at it just resizes to the partition size.

In my OSes I normally have to take this script out as it is not necessary for the tar files.

Ae3NerdGod, 2020-09-18:
I'm under the impression each partition gets made as the os is being installed. As in You're installing 3 os. Os 1 makes its two partitions, and while it's filling them the rest of the drive is unpartitioned? Meaning if I tried to expand the partition via the a method like raspbian does it would take up os 2 and 3's space in the process? Am I incorrect?

lurch, 2020-09-18:
You don't want to expand the partition (which is defined in the MBR/EBR), you only want to expand the size of the filesystem within the partition.

procount, 2020-09-18:
You are getting things mixed up.
Resizing is a two step process. 
When expanding, you first resize the partition, then you resize the filesystem within it (The reverse order for shrinking).

In the case of OS installation with PINN, the size of the partition is already determined. So after you have installed a fixed size image into a partition of the correct or larger size, all that remains is to expand the filesystem to fill the existing partition. This is not normally done by partition_setup.sh, but by a first boot script within the OS itself.

If you start using partition_setup.sh to resize partitions during installation, the remaining OS installations are likely to fail.

Ae3NerdGod, 2020-09-18:
So pinn DOES make ALL PARTITIONS before installing oses, just to be clear?

procount, 2020-09-18:
YES. (unless it is re-installing or replacing an OS, then it relies on the existing partition layout).

procount, 2020-09-18:
When you install a Raspbian image normally (with Etcher or RPI Imager etc), you are installing an image of a disk that is pre-partitioned. When Raspbian starts, it therefore has to first extend the partition size, reboot, and then resize the file system within the partition.
When PINN deals with images, they are not quite the same thing. NOOBS/PINN only deal with images of the filesystem INSIDE the partitions. Therefore a Raspbian image formatted for PINN would consist of 2 images: one for the boot partition and one for the root partition. Each image is installed to an existing partition and the OS simply needs to expand the filesystem to fill that partition. It should not attempt to resize the partition because there is no free space left to resize it.

Ae3NerdGod, 2020-09-21:
That second comment I knew. Thanks for clearing the first part t up tho, it was in the naming of the description in the json by the way. I used dots for the date, it was looking for "flavor" tars that caused the error

closed 2020-11-05 by procount