procount/pinn #421: Can't backup (usb?) Installed OS's

Ae3NerdGod, 2020-09-21:
Ive double checked the instructions on the readme, I've tried ext4, fat32, and ntfs (in that order), double checked that there is an /os folder, none of that has enabled the backup button in the maintenance menu when I check an os (or pinn) 

The drive doesn't even show up at the bottom in the maint. Menu, (yet does in main menu and it's for sure MBR.

procount, 2020-09-21:
You seem to have 2 separate issues:
* Backup button is not enabled.
* USB drive does not show up in maintenance mode.

Which OSes are you trying to backup?
Are they standard PINN OSes, or OSes you have created yourself?

lurch, 2020-09-21:
Adding more details (size/type/usb-or-SD/etc.) about which drive(s) the OSes are installed to, and which drive(s) you're trying to back them up to, would probably be helpful. And are you booting PINN itself from SD or USB?

Marxon, 2020-10-10:
Hi,
i have the same problem and can not backup my os.
Some details:
-installed os: RetroPie4, LibreElec, Raspberry OS full
-PINN (3.3.4.2) is booting from SD
-backup devices USB 3 HDD, SD card with usb adapter, ntfs formated and os folder on root 
I can only select repair, reinstall and replace, no matter which os is selected. Backup is allways greyed out.
The HDD/SD card is only shown in main menu as destination. In all other menus there is an empty destination drop down.
Thanks for any advice.

Best regards
Marxon

Ae3NerdGod, 2020-10-10:
@procount, sorry I thought I replied a bit ago... Must have not hit the button when I did..

It also affects sd card installed OS,

To troubleshoot I tried a 16gb sdcard with one os installed (raspberry desktop) backup button still didn't enable, so that should eliminate whether or not it has something to do with an OS I made.

Ps I have tried lots of usb drives most smaller than the ssd/sd card I was trying to backup. But using a 16bg sd with one os and a 64gb usb did not solve the problem

procount, 2020-10-11:
Ok. I will try to investigate what's going on here.
Probably not setting up the target USB drive correctly as it is a bit particular.

Ae3NerdGod, 2020-10-11:
I followed the readme, but just to make sure I'm understanding it the only requirements it said were
- fat32, ntfs, or ext4 
- empty folder named "os"

procount, 2020-10-11:
That sound right, but I'll check soon.

procount, 2020-10-12:
I've just tried this with a FAT32, EXT4 and NTFS formatted USB stick and it worked fine testing it with Raspberry Pi OS (Full).
Here are the full steps I took so you can test it out identically.
1. v3.3.4.4 version of PINN was used to install Raspios (FULL) 32-bit from USB to the SD card.
2. On starting PINN with no additional USB sticks attached, press 'M' (or click on the More button) twice to go to the Maintenance menu. Note that the "Destination Drive" dropdown at the bottom of the dialog is empty.
3. Tick the box next to the installed Raspios FULL OS. Note that the Fix, Reinstall and Replace menu buttons become enabled, but the backup button does not.
4. Take a blank USB stick and format it as NTFS, EXT4 or FAT32.
5. Make a folder called 'os' in the root of the USB stick.
6. Insert the USB stick into the RPi.
7. After at most 3 secs, the Destination Drive box should be populated with the drive device and name, e.g. "sda: Flash Disk"
8. If you still have the Raspios ticked, untick it.
9. Tick the Raspios selection. The Backup icon should become enabled.
10. Click on the Backup button.
11. Accept the confirmation dialog box.
12. If you have no internet connection, you may need to set the time in the following dialog. 
13. Change the backup ID and description as required.
14. continue to backup the OS.

Please try the above procedure and report back if and where it fails. I am creating a test version of PINN with additional debug information that will hopefully pin-point any problem areas if this is still failing for you and will upload it if necessary.

procount, 2020-10-12:
I have uploaded https://sourceforge.net/projects/pinn/files/testing/pinn-3344a.zip/download which provides additional logging (prefixed with "POLL:") to help identify these backup problems. It also includes a new "Show Debug Log" window under the Maintenance/Fix (Select an OS first to enable it) which allows the debug log to be viewed in a window and refreshed as required.

Marxon, 2020-10-16:
Are you kiddin me!?
If i do it step by step like you described, it works!!!
Dont know how many times i tried it before...
Thank you a lot :)

Best regards
Marxon

procount, 2020-10-16:
It always helps to follow the instructions :smile: 
If you can identify what you did wrong before and can suggest ways to improve the instructions to prevent this happening for other people, please let me know and I will update them.

I didn't think there was a problem with retropie. I'm sure I've done it myself. What error are you getting?
For some OSes, you must be connected to the internet for the backup to succeed as it needs the latest scripts.

Marxon, 2020-10-16:
Worked again! 
Internet was nesscary for backup.

closed 2020-11-05 by procount