procount/pinn #425: [OS Request] Add Recalbox 7.0 Os for Rpi4

Juanitow3, 2020-10-14:
Hello.

I want to request to add recalbox 7.0 for rpi4.
The URL for install is
https://download.recalbox.com/

Thank you.

procount, 2020-10-14:
The Recalbox noobs image that PINN uses is managed by recalbox themselves. I suggest you ask the maintainers of recalbox if they can update their NOOBS image, and then PINN will automatically use it.

closed 2020-11-02 by procount

ninjaw, 2020-11-28:
Version 7.1-Reloaded Recalbox is distributed on Noobs (again) and Raspberry Pi Imager

But I cant see anything on pinn or even pi manager

procount, 2020-11-29:
If it's available on NOOBS, it should be available on PINN.
Do you have a Pi4 or Pi400? Because I don't see a version available on NOOBS for those models.

procount, 2020-11-29:
You should find Recalbox under the "Games" tab

ninjaw, 2020-12-09:
Hello, strangely this day Recalbox is available on Imager and Pinn lite for PI4, but under the general tab

procount, 2020-12-09:
It's an externally generated OS image, so I will have to override the category, It will move to games shortly.

procount, 2020-12-10:
It should have moved to Games now.

procount, 2020-12-26:
I have created an alternative set of meta files to allow PINN to install Recalbox.
Please use the versions called "recalbox-rpiX".
Also use the v3.5.4 of PINN which additionally supports the exfat file system required for recalbox 7 backups.