procount/pinn #426: PINN application

minecraftchest1, 2020-10-28:
Something that would be nice for headless PI's would be an application that could be used to modify the config file for the PINN install on the pi. I do know that there is an option to VNC into PINN, but that may not always be easy to do, or practical to configure, for certain setups. I can take a look at writing such an application, but I imagine you would be able to do a better job.

-Minecraftchest1

procount, 2020-10-28:
Which config file? - config.txt, recovery.cmdline? Both of these can be changed within PINN using the Edit Config option form the Maintenance menu. There is even a nice gui to change all the cmdline options. It even has a virtual keyboard so you only need a mouse, or joystick, or tv remote.

Or do you mean something else?

minecraftchest1, 2020-10-28:
What I mean is porting the Edit Config menu to be able to be used without having to reboot to PINN. Rather than reinventing the wheel, I'm thinking that the UI that already is used in PINN be able to be used standalone. Hopefully, I didn't over-explain what I men. :)

minecraftchest1, 2020-10-28:
It might help if I explain what I am picturing. A raspberry pi user needs to make a modification to a commandline.txt or config.txt for an installed os. Instead of having to reboot into PINN, possibly having to find a monitor, keyboard, and or mouse to connect to the pi to make a minor change, the user instead just launches the PINN config app, specifies the change, and possibly reboots,

procount, 2020-10-28:
Sorry, but I can't see the point of this. 
You don't need a special program to do this when you can just do `sudo nano /boot/config.txt` or run `sudo raspi-config`
The point of the Edit config in PINN is to fix things in an OS that you otherwise can't get into, or can't see the display etc.

lurch, 2020-10-29:
@minecraftchest1 I think what you're asking for would be a lot of work, for only very marginal benefit?

minecraftchest1, 2020-10-29:
I agree that it isn't 100% useful, but I disagree that it would take a lot of work. I believe the existing UI  code can be ported farily easily (I haven't looked at the code only because I don't know what does what yet), and the rest is either reading the file at boot up and calling the correct functions. The only thing that really needs changing afak is changing the functions called by the buttons to write the requested changes to a file, and discovering the file path (which should be mostly static across install).

But as I said, I haven't really looked at the code yet or I would have put something together already. I could also be wrong on what needs to be done for this.

Thanks for taking the time to look at this.

closed 2020-11-05 by procount