procount/pinn #428: Is Windows IoT still supported?

theycallmevirgo, 2020-11-04:


procount, 2020-11-04:
AFAIK, Microsoft haven't updated the NOOBS version of Windows IoT for some time.
The last version issued is only available on the PI2 and Pi3, so if you have a different model like the PI4, it will not be shown as available.
I don't know what the current state of Win IoT is like, but if they have subsequently supported the PI4, you will have to install it without NOOBS or PINN.

theycallmevirgo, 2020-11-04:
Ah, I see. I am in fact on Pi 4 (I need USB3) so I will try BerryBoot.  

Thanks so much for the info

Joe

procount, 2020-11-04:
Good luck, but I think you might have the same issue with Win IoT on Berryboot too.

closed 2020-11-04 by procount

theycallmevirgo, 2020-11-04:
They support local install from ISO, which I prefer anyhow.

theycallmevirgo, 2020-11-04:
Do you get alerts from messages here? How?

lurch, 2020-11-04:
> you might have the same issue with Win IoT on Berryboot too.

Indeed: https://github.com/maxnet/berryboot/issues/269

Also, after a bit of searching I found https://www.iottechtrends.com/install-windows10-iot-core-raspberry-pi/ which claims that Win 10 IoT _can_ be installed on the Pi4, but the commenters on that article said that they couldn't get it working.
https://www.techrepublic.com/article/windows-10-iot-core-on-the-raspberry-pi-4-will-microsoft-support-the-35-computer/ also says that Win 10 IoT doesn't run on the Pi4.

> Do you get alerts from messages here? How?

https://docs.github.com/en/free-pro-team@latest/github/managing-subscriptions-and-notifications-on-github/about-notifications