procount/pinn #429: PINN won't boot

priggle69, 2020-11-04:
I got an update message the other night. Clicked update and it proceeded to reboot into a screen that just has a cursor arrow with a small green box around it. Rebooted several times, same thing.  Not that I have lost my entire SSD, multiple OS, setup, I resorted to starting over.  After trying several flash drives with the PINN OS, it still goes to the same cursor arrow.   Other SD Cards with images on them boot normal.  Raspberry Pi 4 (8G).

procount, 2020-11-04:
Have you already started over? Because your OSes would not have been affected and could be restored.

priggle69, 2020-11-04:
they are untouched on my SSD.  I was booting PINN from a SD.

procount, 2020-11-04:
That's good. so they should be safe. What did you mean by you "resorted to starting over", then? What did you do?

I had tested most pi models, but not the Pi4 8GB. But I have now and it still works fine. so I'm a bit puzzled what went wrong in your case.

priggle69, 2020-11-04:
Not sure.  I reflashed my micro SD and was going to reinstall everything.  I just can't get PINN to boot at all.  It goes from the multicolor boot box to a screen with a cursor arrow and does nothing.

procount, 2020-11-04:
So, you had PINN on your SD card, and installed various OS with it onto your USB attached SSD. It was working fine,
Then PINN updated and you couldn't get past the cursor with the green box.
You tried several times, then gave up and reflashed your SD card with PINN, which repartitioned the SD card and reformatted it.
Now you still can't get past the cursor with the green box.
Is that a correct summary?

If that is the case, then you did actually wipe the boot partitions of all your OSes, unfortunately, since they are still stored on the SD card. But all is not totally lost, since the boot partitions are not normally modified after installation, so they can be replaced. Your rootfs partitions will be on the SSD and should be safe, but you have to be careful if you want to keep them that way.

So, your choice - you can try and recover your previous OSes, or you can just start over completely and re-install your OSes. Let me know and I can try to help you. Also what timezone are you in?

procount, 2020-11-04:
Also, please describe exactly what you mean by "Reflashed my micro SD card", since PINN is normally installed just by copying it to the SD card. Or did you use Etcher or DD or RPi Imaging Utility to "flash" my etcher version of PINN?

priggle69, 2020-11-04:
All of that is correct.  I tried both ways. Copying files to SD card and downloading the image off of your link and flashing it to the SD. I just now tried the SD card in my 4g model and it booted into PINN fine.  I took the same card back out, put it in my 8g and back to the cursor arrow.

Eastern.

procount, 2020-11-04:
Had you just copied PINN across, your boot partitions would still be there, but because you flashed the etcher image, I'm afraid they probably went.

This is quite strange that it only failed in your 8G model. However, my 8G model is pre-production so there may be some differences. 🤷‍♂️ 

So what do you want to do -  Try and recovery, or start again?

priggle69, 2020-11-04:
Staring again wouldn't be bad.

procount, 2020-11-04:
OK, well that's easy (even if a bit painful).
Go to https://sourceforge.net/projects/pinn/files/archive/ where I store all previous versions of PINN.
Download pinn-3344.zip, which was the previous version and unzip/copy the contents to your SD card as usual.
On the SD card you will find recovery.cmdline. I would edit it and remove the "runinstaller" option just to save some time, since you have already reflashed the SD card which repartitioned and reformatted it, so you don't need to do it again.
Then you should find this version will run on your 8g.

After that, you could backup your SSD partitions to another SSD/HDD or laptop if you wish. Or if starting again, just re-install your OSes.

When you have it running again, I would love to know why your 8G didn't work so maybe we can do some investigation if you're up for it?

priggle69, 2020-11-04:
Downloaded 3344.  It booted and I entered my wifi password.  It showed it had an update and I chose to update. The program showed downloading, rebooted and now Im back to the cursor arrow.

procount, 2020-11-04:
Sorry. I should have told you not to update. In fact you should select ignore to stop it asking to update.

procount, 2020-11-04:
Ignore will prevent PINN asking to upgrade to v3.5, but when I release another version, it will prompt to upgrade again. Unless I can fix this issue before then, the upgrade will not work on your 8g and you will be back at the green cursor again.

priggle69, 2020-11-04:
Thanks. Re-installing right now.  I see you added Twister OS.  Nice.

procount, 2020-11-04:
Do you have any other spare SD cards? Perhaps you could try NOOBS 3.5 on another card in your Pi4 8GB and check that works. PINN and NOOBS share the same base architecture (and currently the same kernel & firmware) so if that fails it narrows down the cause somewhat.

priggle69, 2020-11-04:
I will try

procount, 2020-11-04:
Great.
It would also be useful if you could post the output of /proc/cpuinfo to see exactly what model you have and if it differs from mine.

procount, 2020-11-05:
It seems the version I tested on my 8GB version and worked was not quite the same as the one that is released.
When I downloaded the release version to verify, I was able to reproduce the problem, but only on an 8GB PI. On other models it is ok.
So I hope to have a fix shortly. Meanwhile, keep with the p3.3.4.4 version.

procount, 2020-11-05:
I have found the problem. A new version 3.5.1 will be uploaded shortly.

closed 2020-11-05 by procount