procount/pinn #430: Rotate PINN screen 180

iaflipped, 2020-11-04:
Hi,

I am using an official raspberry 7" display but PINN recovery is upside-down. I can't physically rotate the screen as it is fixed. Can I somehow edit a configuration file such as config.dsi and add 'lcd_rotate=2' for example?

tnx...

procount, 2020-11-04:
Yes. Let me find out the info.
Which model of pi?
Which version of PINN?
Are you using screenswitch.zip/pinn_init.sh?

procount, 2020-11-04:
In fact, you can do this within PINN, if you can deal with the screen being upside down.
After it has booted into PINN, press M twice to go to the maintenance menu. With PINN highlighted, press E to edit the configuration. In the first tab for config.txt, just add 'lcd_rotate=2'. Click ok, then reboot.

If you are using screenswitch/pinn_init, you will have to do it to the config.dsi file as well.

iaflipped, 2020-11-05:
Raspberry Pi 4B with PINN v3.5. In fact all I had to do was make the addition to the config.txt. Thanks!

closed 2020-11-05 by procount