procount/pinn #431: PINN will not boot on rpi4b - only shows black screen with centered mouse pointer over chroma-ish green background

ihomelabit, 2020-11-05:
PINN v3.5 seems to fail to properly boot for a new install on a rpi 4 b. Instead, it:
- Passes the rainbow screen
- Seems stuck on a black screen with only the mouse pointer at the center of the green on a squared green background
- No activity on either mouse or keyboard seems to lead to any action (screen mouse movement, change of tty, etc.)

This behavior:
- Persisted across 2 SDXC (64 GB) cards from 2 different brands, Kingston and Sandisk.
- Appears using both sourceforge's .zip files and github's .img device image for v3.5.

The device boots properly and correctly when using other distributions, so it seems tied to PINN.

The test cards were:
1. Erased via `blkdiscard`
2. Either:
    - Imaged from the .img file via `dd if=pinn-lite.img of=/dev/$device bs=512 conv=notrunc,noerror,sync`, then tested on the Raspberry, or;
    - Formatted with `fdisk` on a new DOS partition table with a single partition starting at sector 8192 (as post-relocated NOOBS cards present) using the whole card, with type `b - W95 FAT32`.
3. If not imaged, fs was created via `mkfs.fat`, and files copied/synced/umounted.

Cards were subjected to a `blkdiscard` round between tests.

A prior search seems it is a reproducible issue as per #429, but it seemed fitting to open a new one as it is no longer tied to any upgrade procedure.

ihomelabit, 2020-11-05:
Further info:
- Testing the same cards with the same PINN version on them at a Raspberry Zero W leads to complete bootup, so it seems hardware related somehow.
- The way the mouse pointer is presented on faulty booting seems to be like the pointer image file is loaded and drawn but somehow the graphical initialization fails to complete before it is tied to the GUI as the actual mouse pointer - the green background reaches just some pixels around the actual pointer.

procount, 2020-11-05:
#429 was restricted to a PI4 8GB. 
What model of Pi are you using?
Please can you provide the contents of /proc/cpuinfo from your pi,

procount, 2020-11-05:
Please can you also try NOOBS 3.5 and see if that works on your PI.

procount, 2020-11-05:
Please forget testing NOOBS 3.5 - it is not necessary.
Are you using a PI4 8GB model?

ihomelabit, 2020-11-05:
Mine is also a  rpi 4 8GB, model B though.
Just finished testing NOOBS 3.5, works fine, and seems to run the bundled Raspbian OS without issue.

cpuinfo follows:
```
$ cat /proc/cpuinfo 
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: d03114
Serial		: 10000000f94c47f8
Model		: Raspberry Pi 4 Model B Rev 1.4
```

procount, 2020-11-05:
I have found the problem. I new version 3.5.1 will be uploaded shortly.

closed 2020-11-05 by procount