procount/pinn #432: GPT

eu41, 2020-11-07:
I used a GPT partition scheme on my SSD and copied PINN on the first FAT32 partition. PINN booted successfully from the USB connected SSD, but it recreated on the disk its own MSDOS partitioning scheme for installing the OS.

Now that Raspberry Pi 4 is capable of booting from a USB connected SSD, would it be too difficult to switch to GPT instead of MSDOS partitionning? This would allow creating a many partitions on a SSD for a multitude of OSes, including BSD varieties. Those OSes could be easily booted through GRUB, as openSUSE does.

lurch, 2020-11-07:
Duplicate of #413 ?

eu41, 2020-11-07:
Yes, it looks similar to #413.
I see from that that thread that it is actually more difficult than I thought to convert to GPT.
I will try to another way, without using PINN.

closed 2020-11-07 by eu41