procount/pinn #435: Static IP setting

etnalta, 2020-11-17:
I need to set up a static IP for PINN
As described, I put a new dhcpcd.conf in the root partition, after reboot it has been renamed as dhcpcd.conf.bak, but no change in the IP.
I added to the original one
#static IP address
interface wlan0
static ip_address=192.168.1.29
static routers=192.168.1.1
static domain_name_servers=192.168.1.1

I checked /etc/dhcpcd.conf, and there is no change

Any idea/clue?

Thanks

procount, 2020-11-17:
/etc/dhcpcd.conf is readonly.
PINN should be using /settings/dhcpcd.conf

etnalta, 2020-11-18:
/settings/dhcpcd.conf is ok, with the new lines for static ip, anyway it doesn't work, still have a dhcp IP.

What am I missing?

procount, 2020-11-18:
> What am I missing?

/24

#static IP address
interface wlan0
static ip_address=192.168.1.29/24
static routers=192.168.1.1
static domain_name_servers=192.168.1.1

etnalta, 2020-11-19:
Sorry, I did try with /24, same issue.

etnalta, 2020-11-19:
I found another way to solve this using static ip assignment from dhcp server.

Pinn has its own IP address at boot, then I can use rebootp (ssh) to boot with a specific OS which has its own IP too.
Seems to work.
Thanks for your help.

procount, 2020-11-19:
Well, it worked for me. Maybe you had a syntax error somewhere 🤷‍♂️ 
I was going to suggest setting a static IP from your router instead, based on mac address. This will persist across all OSes installed.
Remember that PINN and all installed OSes are independent, so you would have to set a static IP address in each OS individually doing it with dhcpcd.conf.

etnalta, 2020-11-19:
This is exactly what I did. 
I'm using volumio and libreelec, for each I change parameters to static IP, and after rebootp x (x = boot partition number of each OS), selected OS starts with its own IP.
Now I have to setup an automated OS switch thru ssh from Jeedom when needed.....

I don't know where there is a mistake with PINN in my case for dhcp.conf, but as it works this way I won't look further.

Thanks again.

closed 2020-11-29 by procount