procount/pinn #436: Rpi 4 display not working

zamalek89, 2020-11-18:
Hi i just got my new raspberry pi 4 4gb. Im a total noob.
I extracted pinn on fat32 memory card but if i boot up i dont get display.any help please. If i unplug display and boot up wait a minite and then plug in i get a low resolution display.

procount, 2020-11-18:
This is probably a display resolution problem as the Pi4 has a problem with 1366x768(?) resolution.
Try editing the config.txt file by inserting the following 2 lines at the top of the file.

```
hdmi_group=2
hdmi_mode=28
```

If that does not work immediately, try connecting to the other HDMI port, reboot and try again.

zamalek89, 2020-11-18:
Thanx will try when i get home and post again!

lurch, 2020-11-18:
Does PINN support the "HDMI safe mode" that NOOBS does? IIRC you can press one of the numbers on the keyboard and it'll force the HDMI display into 640x480 ?

> the Pi4 has a problem with 1366x768(?) resolution.

More details here: https://www.raspberrypi.org/documentation/configuration/config-txt/pi4-hdmi.md

procount, 2020-11-18:
Yes, of course, so you can press 1 or 2 to switch HDMI modes, although it's not as useful on the Pi4 for switching to RCA output now.
But PINN doesn't look so good at 64x480, which is why I chose the other mode. Most monitors should cope with that if it is the 1366x768 issue but I agree just pressing 1 or 2 will quickly confirm the problem. If it is the cause, change the HDMI settings in config.txt to some suitable resolution values for your monitor.

zamalek89, 2020-11-21:
> This is probably a display resolution problem as the Pi4 has a problem with 1366x768(?) resolution.
> Try editing the config.txt file by inserting the following 2 lines at the top of the file.
>
> ``` [...]

Thanx that worked but whatever os i install i have to edit the config first is this normal?

procount, 2020-11-21:
Each OS (and PINN) is completely independent. Selecting a different OS is like inserting a new SD card. So each one has to be configured in its own right.

Compared to previous RPi models, the new VC6 in the Pi4 uses "pixel doubling" in its display output, which means there must be an even number of pixels in all aspects of the display timing.
The default resolution of your monitor seems to be 1366x768, but unfortunately this has an odd number of pixels (somewhere in its timing), so it is not compatible with the pi4. Therefore you need to force the PI4 to use a different resolution so that it will work with your monitor. You could also pick other resolutions than the 1280x800 resolution I chose above.

I think I remember reading somewhere that if you use a custom timing mode (group 2 mode 87 with the parameters in hdmi_timings?)  that it forces the PI4 not to use pixel doubling, so this might be a way to continue to use the 1366x768 resolution. I think there are also similar resolutions you could use instead: 1360x768 for example. See https://www.raspberrypi.org/documentation/configuration/config-txt/video.md for more information, or search the posts by 6by9 in the forum.

zamalek89, 2020-11-21:
> Each OS (and PINN) is completely independent. Selecting a different OS is like inserting a new SD card. So each one has to be configured in its own right.
>
> Compared to previous RPi models, the new [...]

Thanx for the imfo

lurch, 2020-11-21:
> Yes, of course, so you can press 1 or 2 to switch HDMI modes, although it's not as useful on the Pi4 for switching to RCA output now.

IIRC the `enable_tvout` is off by default (on Pi4) because it reduces performance slightly. But given that PINN is quite simple (when compared to a full OS) and so doesn't need high performance, I wonder if it might be worth enabling `enable_tvout` by default in PINN builds, to increase out-of-the-box compatibility? :shrug:  (or do you think it might reduce SD-card write speeds?)

procount, 2020-11-21:
It maybe worth an experiment

procount, 2020-11-29:
@Lurch - When you add `enable_tvout=1`, it not only enables the TV output, but sets it as the active display, I haven't tried forcing HDMI when tvout is enabled, though,

lurch, 2020-12-01:
Oh, didn't realise that, sorry. You can tell I've not experimented with that feature myself :wink:

closed 2021-02-17 by procount