procount/pinn #438: Problem With Raspberry Pi 3 B+

32Dexter, 2020-11-22:
Hi, I have a problem with Raspberry Pi 3 B+.
I have try to install from your site: https://pinn.mjh.nz/ but not work.
I have download pinn-lite.zip from SourceForge but not work.
I have formatted the SD card with "SD Card Formatter" and extract with 7zip the "pinn-lite.zip" in the SD and after put it in my Raspberry PI 3 B+. But it not start. Why? Can you help me Please? Thanks

lurch, 2020-11-22:
How big is your SD card? If it's 64GB or greater you may be running into this problem: https://www.raspberrypi.org/documentation/installation/sdxc_formatting.md

procount, 2020-11-22:
If you have the problem Lurch mentioned, than an alternative is to download the PINN [image file](https://sourceforge.net/projects/pinn/files/Etcher/pinn-lite.img.zip/download) and flash it to your SD card using the new RPi Imaging Utility (which can be found through the link Lurch provided above).

Also see #436 in case it is booting, but you just can't see it.

lurch, 2020-11-22:
@procount Isn't the issue in #436 specific to the Pi4, and so wouldn't affect the Pi3+ ?

procount, 2020-11-22:
Oops. Sorry. Not paying attention!

32Dexter, 2020-11-23:
I solved it, it was my mistake because I have a 64Gb SD card and the "SD Card Formatter" program formatted in exFAT and not in FAT32.
Now another problem:
If I download "pinn-lite.zip" from Sourceforge, I see all the images and download what I need, But if I download preconfigured PINN from https://pinn.mjh.nz/, (example: with image Ubuntu Mate and LibreELEC) when I get to the page where I should see the list of operating systems (example: Ubuntu Mate and LibreELEC chosen previously), the list is completely empty. Why? Thanks

procount, 2020-11-23:
Please post the recovery.cmdline that you created from pinn.mjh.nz.

When you chose the pinn.mjh.nz route, you are replacing the full list of OSes with your webpage selection.

32Dexter, 2020-11-23:
This is the "recovery.cmdline" created from pinn.mjh.nz.
`runinstaller quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline no_group no_default_source select="allnetwork" disablesdimages alt_image_source=https://pinn.mjh.nz/os_list.json?ubuntumate2010_64=28750&libreelec_rpi2=28750 loglevel=2 sdhci.debug_quirks2=4`
I assumed that the list was replaced, but I have some doubts that the paths generated by the new file are perhaps not updated with the latest directories.

procount, 2020-11-23:
Something seems to be broken somewhere. Will check.

procount, 2020-11-23:
Please use this for recovery.cmdline

`runinstaller quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline no_group no_default_source select="allnetwork" disablesdimages alt_image_source=http://pinn.mjh.nz/os_list.json?ubuntumate2010_64=28750&libreelec_rpi2=28750 loglevel=2 sdhci.debug_quirks2=4`

procount, 2020-11-23:
You seem to have allocated equal amounts of storage space to UbuntuMate and LibreElec, which would be the default case anyway. So I'm not sure why you chose this route. It might have been better just to stick with selecting the OSes through PINN itself.

32Dexter, 2020-11-23:
In fact, then I did exactly as you say as I was experiencing the problem that I did not see the images in the selection list of operating systems. As soon as I can, I try this new "recovery.cmdline"

procount, 2020-11-23:
> In fact, then I did exactly as you say as I was experiencing the problem that I did not see the images in the selection list of operating systems

Sorry, I don't know exacty what you mean by this. Can you elaborate, please?

If you want to go back to the normal selection in PINN, you will have to restore your recovery.cmdline to what it was before. To do this you can copy recovery.cmdline.new to recovery.cmdline.

32Dexter, 2020-11-23:
Excuse me what I meant is that since through the site: https://pinn.mjh.nz/, it didn't work (I didn't see the images).
I continued downloading "pinn-lite.zip" from Sourceforge and from the list I selected the systems operating I wanted and then installed.
Basically I didn't use pre-configured PINN from the site: https://pinn.mjh.nz/.
As soon as I can, I'll try PINN pre-configured with the new "recovery.cmdline" you just wrote to me.
I hope I have explained myself better.

procount, 2020-11-23:
OK So your issue is solved for you then.
I have contacted Matt to investigate the issue with his webpage and hopefully it will be fixed soon. Thanks for bringing it to our attention.

32Dexter, 2020-11-23:
Thanks to you for all the immense work you do!
Thanks

32Dexter, 2020-11-23:
> runinstaller quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline no_group no_default_source select="allnetwork" disablesdimages alt_image_source=http://pinn.mjh.nz/os_list.json?ubuntumate2010_64=28750&libreelec_rpi2=28750 loglevel=2 sdhci.debug_quirks2=4

Works Perfectly! Thanks a Lot

matthuisman, 2020-11-23:
have updated pinn.mjh.nz to now output http:// for the cmdline which fixes the issue.

closed 2020-11-29 by procount