procount/pinn #439: RPI4 does not boot when a TV is connected but the TV is off

EctoOne, 2020-11-25:
Hi, I'm not sure if this a Problem with PINN or a general PI4 thing.

I recently got a PI4 and I'm quite happy so far. The only issue I have is that the PI won't boot if I have my TV connected but the TV is off. If I turn on the TV or unplug the HDMI cable, the PI boots fine.

It is not that big of a problem since it is mostly a headless setup anyway. But having a TV connection would be a nice thing to have.

I tried several options in the config.txt but no success. And I know that it won't even load PINN because I tried forcetrigger and vncinstall in the commandline and my vnc client wasn't able to connect.

I'm not sure but I don't think I had that issue with a PI3b+, at least I never had to unplug the HDMI cable.

If it is not a general PI4 thing, the only other thing I can think of that can be the cause of the problem is that I'm using an Argon One M.2 case. Which has an extra adapter to move both HDMI ports and the audio jack to the back of the case. And to convert the mini HDMI ports to normal HDMI ports. The case also uses the GPIO ports for the fan, power button, IR receiver and maybe other stuff. Sadly I don't own a mini HDMI cable so I can't test if the case is causing the problem. And to be honest, I don't want to buy one just for this.

lurch, 2020-11-25:
> I'm not sure if this a Problem with PINN or a general PI4 thing.

Does Raspberry Pi OS _without_ PINN (e.g. installed with [Raspberry Pi Imager](https://www.raspberrypi.org/software/) ) boot on your Pi4 when your TV is connected but switched off?

EctoOne, 2020-11-25:
That was something I didn't test before. But yes it works. I just flashed another card, did the initial setup and enabled vnc. Then I've shut down the Pi and turned off my TV. I waited a bit and turned the Pi back on and waited a bit more before I started the vnc client. And it showed up instantly. So I guess it is PINN related?

procount, 2020-11-25:
Another thing you could do is test whether NOOBS also has this problem, in order to narrow it down further.

EctoOne, 2020-11-25:
Uhm, so NOOBS does not boot at all. All I'm getting is the rainbow screen.

procount, 2020-11-25:
Try the other HDMI socket, especially if the top left corner of the rainbow square is magenta
If PINN boots, NOOBS should boot.
I tried PINN with 2 HDMI monitors, both turned off, and PINN booted fine. Perhaps it is something special about your TV, DOes it support CEC?

procount, 2020-11-25:
You could try adding `hdmi_ignore_cec=1` to config.txt in PINN 🤷‍♂️

EctoOne, 2020-11-25:
OK, after changing the HDMI port, NOOBS booted. I did the setup again and enabled vnc. But when I turned of my TV it didn't boot.

My TV should support CEC, but I never really used it. I only noticed it that the TV turned on with various Distros. Since then I usually have hdmi_ignore_cec=1 in my config.txt. But that wasn't the case with the testing setups. So maybe the adapter from the Argon One case does something. Maybe I need to buy a mini HDMI cable to investigate further. But like I said, it is not that important.

procount, 2020-11-25:
I use the plain Argon One case (without the M2 SSD) and the HDMI is just a pass through and doesn't make any difference for me.

procount, 2020-11-25:
Another thing you could test is to add `hdmi_force_hotplug=1` to config.txt

EctoOne, 2020-11-25:
hdmi_force_hotplug was one of the first things I tried when I was looking for solutions.

I mean since flashing the OS directly to card worked, I could clone the SD to my SSD, but since the M2 version of the case has no access to the SD I would prefer to keep PINN.

The only other thing I can test later is a different HDMI cable. It is not the same I used before with my PI3. Although it is the one that came with the Nintendo Switch so I guess it's not particularly a bad cable.

lurch, 2020-11-25:
Yeah, unfortunately tracking down edge-cases like this tends to boil down to swapping lots of different things back and forth, until you manage to narrow down the one thing that causes it to go wrong :confused: (which I know from experience can be very tedious and frustrating!)

> But like I said, it is not that important.

Fair enough :wink: 
At least @procount now knows that it affects NOOBS as well, so isn't isolated to just PINN.

EctoOne, 2020-11-25:
I mean I was expecting even more problems with the PI4 from what I've read beforehand and based on the fact that it is relativ new hardward/software. That's why I didn't replace my main setup first. In that case it would be a bigger issue for me. But I have to say that the case is totally worth it and if PINN would not exist I probably would have way less fun experimenting with the various Distros.

Since procount only has the normal Argon One there still might be a slight chance that the adapter of the M.2 version is somehow causing the problem. They are a little bit different, but who knows.

I even thought that the thing with the different HDMI ports was fixed and that it doesn't matter which one is used. I was really surprised that changing the port caused NOOBS to boot.

Thanks for the help and constant support!

procount, 2020-11-25:
The HDMI issue is only solved in PINN. I have issued a PR for it to be included in NOOBS, so it might get included at some point.

EctoOne, 2020-11-26:
Oh, I must have misrememembered that. I thought it was fixed with an hardware update on the newer revisions of the Pi 4 and not just by software.

Anyway, I tried a different HDMI cable and some other things but nothing helped. I even used `vcgencmd get_config int` and `vcgencmd get_config str` on my OS and per SSH on PINN to see if there was something very different between the two config.txt files. But I couldn't spot anything particular.

I also tried `safemodedisabled` in PINN and `hdmi_drive=2` in the config.txt. I'm out of ideas now.

Is there a specific reason that NOOBS doesn't boot when the wrong HDMI port is used? It almost seems to me that this is the cause of the problem. Like it tries to check for something and my TV causes this check to fail for whatever reason.

procount, 2020-11-26:
Your HDMI issue aside, NOOBS should boot even if you plug into the wrong HDMI port, it's just that you can't see it booting. The 2nd monitor (with the magenta colour in the top left corner) will remain in that state throughout the boot process.

One difference between NOOBS/PINN and normal Raspios is that NOOBS/PINN still uses the legacy graphics driver, not the KMS drivers that Raspios does. The KMS drivers output the same display to both HDMI ports, so it doesn't matter which you plug into.

The latest version of PINN will try to detect which HDMI port has a monitor plugged into it and will try to use that as the active display.

EctoOne, 2020-11-26:
I see but shouldn't I see the desktop once the boot process from NOOBS is done and RaspiOS is loaded or won't it switch the port when RaspiOS is started?

Another question, probably not related to the problem but who knows, since you have the normal Argon One case, do you have to push the power button to turn the Pi on or will it start when you plug in the PSU?

procount, 2020-11-26:
If NOOBS is configured to boot into your OS, then yes, you should see the desktop when RaspiOS boots, since there is a reboot between NOOBS and RaspiOS so it is effectively starting from scratch.

I'm fairly sure the Argon M.2 case is the same as the Argon One, except for a modified deeper base and the USB connector.
Yes, it is necessary to press the button to start it up.

EctoOne, 2020-11-26:
Ah sorry my train of thought was way ahead. I thought I already had installed RaspiOS when I noticed that NOOBS wasn't booting. That's why I was confused. My bad.

And no, there are some differences between the two cases besides the obvious M.2 expansion. The M.2 case has a jumper which let's you select if you need to press the power button or not. And it has an build in IR receiver and the, already mentioned, full size HDMI ports.

Especially the jumper option was one reason why I decided to try the case. My Pis are plugged into a remote controlled power outlet, so I don't have to physically disconnect the power if needed.

EctoOne, 2021-09-07:
It's a little late but why was this closed? I still have that problem that my Pi won't boot when it's connected.

Additionally, I bought a new Pi and a normal Argon One V2 case yesterday and today I had the same issue with TwisterOS which I flashed directly. And I was able to fix this by adding the following to the config.txt:

```
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=82
```

But adding those to the config.txt from Pinn does not help. So I guess that some of the default lines breaks something. Sadly I'm not sure if it would be safe to remove any of them without breaking my whole setup.

procount, 2021-09-07:
I closed it because there was no activity since Feb. I have re-opened it now.
I don't think this is a PINN issue, but rather a firmware issue The firmware isn't booting PINN because of the display issue.
I'm not sure why the solution for TwisterOS does not work with PINN, but I think the solution lies in the correct config.txt options.
You could try using:
```
hdmi_ignore_edid=0xa5000080
```
but I would have thought force_hotplug would have fixed it. Other than that I am out of ideas.

EctoOne, 2021-09-08:
I'm sorry. But it actually seems to be fixed in one of the updates. I just updated and that created a clean config.txt which looked to me that it was at least missing one line with something like vkms or something like that in it. Can't remember if there was anything else in there before.
I probably should have done that before asking again. Sorry again, and thanks.

closed 2021-09-08 by EctoOne

procount, 2021-09-08:
Ah yes, a recent update to the firmware/kernel in PINN required additional config.txt settings. If you used an old file without them it may not have booted, but I didn't think that was related.