procount/pinn #440: Unable to install Rasplex:  debug log says ERROR 503: Service Unavailable.

LeChatNoir666, 2020-11-25:
Hi,
I wanted to install Rasplex but I cant.
Heres what the debug log says:
Executing: "sh -o pipefail -c "wget  --retry-connrefused  --read-timeout=120 --tries=inf --no-verbose -O- http://downloads.sourceforge.net/project/pinn-matt$
"https://master.dl.sourceforge.net/project/pinn-matthuisman/os/RasPlex_RPi2/RaSystem.tar.xz?viasf=1:
2020-11-25 12:24:51 ERROR 503: Service Unavailable.

I will try to get it now by hand and put it in the "os" folder to install it.

Btw: Unfortunately my installed and configured XBian was selected as I selected Rasplex and now my XBian is overwritten.
Luckily I made a Backup yesterday on my Notebook but I need to do some settings again...

Could you please add a warning, that OS xyz will be overwritten?

Oh and:
As I prepare a MMC with PINN for someone who lives some hundred KM far away from me, how can I access her PINN via internet? I tried VNC and SSH on mine with my external IP adress but wasnt able to connect to PINN...

And as I love to experiment with different OS's, how can I delete an OS I dont need anymore like in BerryBoot?

procount, 2020-11-25:
I just clicked on "https://master.dl.sourceforge.net/project/pinn-matthuisman/os/RasPlex_RPi2/RaSystem.tar.xz?viasf=1 from your post and it started downloading. Maybe it was a temporary server problem?

> Btw: Unfortunately my installed and configured XBian was selected as I selected Rasplex and now my XBian is overwritten. 

Which function were you using to do this? Install, Re-install or replace? There is already a warning message for each of these that it will either overwrite your whole SD card, or the partitions of the OS, although it does not individually list the OSes that will be overwritten. By its nature, the install function will wipe all installed OSes before installing another one (regardless of whether the installed one was selected or not). PINN cannot ADD OSes to an existing installation (unlike Berryboot - see later)

There was a similar request to block the operation of the Install function altogether if you already have OSes installed to prevent this sort of thing happening. I have something similar on my roadmap, so hopefully it will be harder to accidentally delete things in the future.

> how can I access her PINN via internet?

If you want to access PINN over a network, you first need to enable ssh and vnc. These should then be accessible on the local Lan. If you want to access it remotely via an external IP address, then you are going to have to configure your router to make those ports accessible to the internet. HOWEVER, as the username/password is well known, this will expose your PI (and through that your WHOLE NETWORK) to the internet. I'm not sure how long it will take for your Pi to be compromised, but I guess not long. So I don't advise this at all. A better course of action is to provide a VPN or another Pi running SSH preferably with keys rather than passwords, to allow you access to her local network. Once you log on to her network through that, you can then access PINN as if it were local.

> how can I delete an OS I dont need anymore like in BerryBoot?

Berryboot works differently with all OSes sharing a partition, whereas NOOBs/PINN allocates partitions to each OS. So you can't delete OSes from PINN without deleting all of the OSes and starting again. However, PINN added the "Replace OS" function so you can replace one OS with another, _provided_ it can use the same number and size of partitions.

LeChatNoir666, 2020-11-25:
Oohhh...
My fault because I misunderstood the use of PINN, sorry sorry sorry...

Well, I would change the port and password  for SSH and VNC if its possible or just telling the person to go to the PINN -> Edit configuration to enable and disable it after use. (Thanks that it have a GUI :) )

And many thanks for your time and explanation! :)

procount, 2020-11-25:
No worries.
Unfortunately, the ports and password are fixed (something else to go on my roadmap).
Coincidentally, I am getting "503 service unavailable errors" today, so I think sourceforge must be having a few problems. Hopefully it will sort itself out.

LeChatNoir666, 2020-11-27:
Hello again... :)
I had the idea to make backups of RasPlex and DietPi which worked fine and I was able to select them for install on the backup menu.
In addition i wanted to add Raspberry Pi OS (32-bit), openelec_rpi2 and XBian (btw: for why XBian cant be backed up?).

The installation finished with errors but by mistake I hit the OK button with my TV remote (btw: the mouse emulation is a great feature!!! :) ), so I cant read the error log...

Now i have only RasPlex, Xbian and DietPi in the boot menu...
But in the Setup Menu its shows, that openelec is installed.
Where is it in the boot menu? :)

As I cant add OS's I have to do the whole install again and hoping, that Pi Os (32-bit) will not fail, or?

procount, 2020-11-27:
> (btw: for why XBian cant be backed up?).

Because it uses BTRFS and I need to add specific code to backup such a file system.

> so I cant read the error log...

You can now read the debug log by going to the maintenance menu, checking any OS (e.g. PINN) select the FIX menu option, then select the Show Debug log. But it probably will not show any errors concerning the partition_setup script (I need to add that).

> But in the Setup Menu its shows, that openelec is installed.
Where is it in the boot menu? :)

If there is an error during installation, it typically deletes the OS. Sounds like a little bug. Maybe it will show as not installed when you next reboot.

> As I cant add OS's I have to do the whole install again and hoping, that Pi Os (32-bit) will not fail, or?

It sounds like it was openelec that failed, but that stopped the installation so Raspberry Pi OS did not get installed.
I will try to make it more robust so that it continues even after failure.
Another method (if all your OSes have the standard 2 partition layout, is to just install several ProjectSpaces. Then replace each project space with whichever OS you really want to install. that would be a bit more forgiving in case of failure.

> Can I somehow import it to PINN?
My idea: Making a fresh install of the same OS with PINN and then somehow overwriting it... :)

It sounds feasible. Once you have installed your clean OS with PINN, delete the contents of its root partition. Then you need to boot your original OS with Berryboot and mount your PINN drive in there. Copy your original root partition across to the correct PINN partition. Unmount your PINN drive and then boot it. (You have to do it this way due to the overlayfs Berryboot uses).  In PINN go to the Maintenance menu, select your newly installed OS, click the FIx button and select Re-run partition-setup.sh to make sure it runs properly under PINN. Then it will hopefully boot.

LeChatNoir666, 2020-11-27:
Big big thanks for your fast reply! :)

> If there is an error during installation, it typically deletes the OS. Sounds like a little bug. Maybe it will show as not installed when you next reboot.
After a reboot its not in the boot menu but in the setup...
Strange...hihihi
Btw: How about to add an "Edit" button in the boot menu so to enter the setup dont needs everytime a reboot with shift-key? :) (I have only one keyboard for 3 Pi's and need to replug then if needed ;) )

> Another method (if all your OSes have the standard 2 partition layout, is to just install several ProjectSpaces. Then replace each project space with whichever OS you really want to install. that would be a bit more forgiving in case of failure.

This sounds like a nice option! 
I will try it later! :)

About the migration of the BerryBoot OS.
Sounds good and as I love experimenting I will give it a try! :)

As BerryBoot dont support 32Bit anymore (a little bit too fast to change to the 64Bit beta stuff in my opinion) and my plans to buy a Pi4 are not on my priority list I want to switch from BB to PINN. :)

procount, 2020-11-27:
> Btw: How about to add an "Edit" button in the boot menu so to enter the setup dont needs everytime a reboot with shift-key? 

This is not feasible due to the way it works. From the boot menu, you must either boot an OS, or press ESC / cancel to reboot back into PINN.
As you have a TV remote. Just press any key on that instead of the shift key. Or if you have a mouse, just left-click.

You can also experiment with the remote_timeout option which will go to the recovery menu at boot but only for a period of time before booting the default OS.

LeChatNoir666, 2020-11-27:
Oh.... too easy to find out... ;)

I will work later on it but...
Instead of copying, can I use dd maybe to copy the root from the working Pi OS?

procount, 2020-11-27:
I would recommend to copy because what you see on Berry boot is a union of 2 files systems, a readonly base layer, with a changed layer on top. You need to copy the combination, not one or the other.

procount, 2020-11-27:
> After a reboot its not in the boot menu but in the setup...

I remember that for some errors I keep the OS, but make it non-bootable (because it is in error) so it won't appear in the boot menu. But if it is listed as installed, but not actually on the SD card, then that would be something I need to improve on!

LeChatNoir666, 2020-11-27:
If you need a beta tester, im free most times (atm im working on my new arrived NAS and beside of that, to make a MMedia MMC Card for the person I mentioned...)
Also I have a 2nd Pi 3b+ just for testing (as I said: I looove to experiment! :) )
And I plan to buy a Pi 4 (4 or 8GB) next week or so.
But even BerryBoot concentrate now more on Pi and 64Bit I will not use it anymore until 64Bit is more stable and not a beta anymore...(Except for my curiosity and experiments.. ;) )

Anyway... PINN is really cool and you did a great job!!! 👍 👍 👍

lurch, 2020-11-27:
Quick note with regards to "replacing" one root filesystem with a different one:

When the Linux kernel boots (e.g. `kernel7.img` on the boot partition) it mounts the root filesystem, and loads in a bunch of kernel modules from the `/lib/modules/kernel_version/` directory on the root filesystem. So if you boot e.g. kernel version 5.4.79-v7+ and you _don't_ have a `/lib/modules/5.4.79-v7+/` directory on your root filesystem, then the kernel won't be able to load the modules it needs.
The typical symptom of this (in my experience) on a Raspberry Pi is the USB not working, which means you can't use your keyboard or mouse :wink:

procount, 2020-11-29:
@LeChatNoir666 - As you mentioned beta test, please see https://www.raspberrypi.org/forums/viewtopic.php?p=1769704#p1769704

@Lurch is quite right about copying the rootfs partition. It was rather remiss of me not to mention it as I had assumed you would be using the same kernel version on the boot partition as on the root partition, but this may not be the case! Please ensure they all match up.

closed 2021-02-17 by procount