procount/pinn #442: Problems installing PINN

eeyatahn, 2020-12-02:
Aloha Procount this is denavin from the Raspberry Pi forums.
I am still trying... I am following your instruction on how to add BSDTAR to Ubuntu but have run into an error. 

First off some info for you Libarchive-3.3.1 is now Libarchive-3.3.2 so following the instructions a written causes an error but I was able to discover this so not an issues there but you may want to update your directions to reflect this.

I was able to get everything done up to the point of ./configure where I ran into this error;

**~/libarchive-3.3.1$ sudo ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... no
checking whether make supports nested variables... no
checking whether to enable maintainer-specific portions of Makefiles... yes
checking whether make supports nested variables... (cached) no
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for gcc... no
checking for cc... no
checking for cl.exe... no
configure: error: in `/home/plg/libarchive-3.3.1':
configure: error: no acceptable C compiler found in $PATH**

I am unable to continue from this point.... Apparently I do not have a C compiler installed so I download and install one from GitHub from https://github.com/rui314/8cc.git with no success, I still get the same error. My lack of knowledge of Linux is my main failing point here. Can you please help me. Thanks

P.S I have not given up Still Trying to master this.

lurch, 2020-12-02:
> Apparently I do not have a C compiler installed

Running `sudo apt install build-essential` is typically the easiest way of fixing that :)

procount, 2020-12-02:
Hi eeyatahn/Denavin  & welcome to Github! Let's see if we can sort out your issues here. I'm rather overloaded atm so apologies if my responses are sometimes delayed, especially with the time difference. Lurch is also an expert collaborator, so he is a good person to listen to. (@lurch - see https://www.raspberrypi.org/forums/viewtopic.php?p=1770766#p1770766 for the background).

Let's start from the basics, so I can follow along and identify any differences. Please be precise in your answers so that I can do exactly the same thing when necessary.

What model of Raspberry Pi are you using?
What type of power supply are you using to power your Pi?
How do you connect your Pi to the internet (wifi or ethernet)?
What network speed do you typically get?
What make, model & size of SD card or other storage are you trying to install PINN onto?
What VM are you using to host Ubuntu 20.10 on your PC? (I m currently using either Ubuntu 14.04 or Ubuntu 20.04 (64 bit) )

What is the main problem you are tying to solve? Is it just to get PINN installed with multiple OSes, or is it to import an existing OS into a PINN formatted drive? Because I thought you said you had unfortunately lost the data that you wanted to import, so that wasn't so much an issue now, yet you are back to trying to install bsdtar (I'm asking just so I know how best to help/advise)

eeyatahn, 2020-12-03:
Aloha Procount, I hope this answers all your questions sufficiently. 

**What model of Raspberry Pi are you using?** <<< Raspberry Pi 4 Model B 4gig
 **What type of power supply are you using to power your Pi?** <<<< The one that came with the Labis kit I purchased from Amazon 5V 3A output
 **How do you connect your Pi to the internet (WiFi or Ethernet)?** <<<< Ethernet hard wire via Netgear Nighthawk Model XR500 Router
**What network speed do you typically get?** <<<< I use Spectrum @ 70Mb per second down, 6 Mb up no Data cap.
**What make, model & size of SD card or other storage are you trying to install PINN onto?** <<<< San Disk Ultra 128 GB XC1 Also have a San Disk Ultra Plus 128 GB SSD via USB 3 but that is mainly for storage and Date transfer from my PC only
**What VM are you using to host Ubuntu 20.10 on your PC? (I m currently using either Ubuntu 14.04 or Ubuntu 20.04 (64 bit) )** <<< Virtual Box v.6.1 with Ubuntu 20.10 64bit. I also have Ubuntu 20.04.1 LTS 64 bit Desktop installed as well

**What is the main problem you are tying to solve? Is it just to get PINN installed with multiple OSes, or is it to import an existing OS into a PINN formatted drive? Because I thought you said you had unfortunately lost the data that you wanted to import, so that wasn't so much an issue now, yet you are back to trying to install bsdtar (I'm asking just so I know how best to help/advise)**

Yes, I did loose all the data on that SD card but I have 4 other SD cards loaded with, Lineage OS, LebreELEC / Kodi, Ubuntu 20.10 and Lakka all with data that I want to preserve and transfer as well. I am doing my experimentation with the ONE SD card I lost the data on. I have reloaded Raspbian on it, if I can get that archived and transferred to PINN successfully then I will transfer over my other SD cards to that SD card rather than take the chance of loosing more data. I tried to use TAR as you suggested but it failed, it may be the option flags are wrong. But since I don't know what the flags you set for BDSTAR mean I need to get BSDTAR working so I can do the archive of the SD card. Once I get my SD cards archived I will then transfer them to PINN rather than doing the Card Switch every time I want to change OSes.

My main goal beside learning something new is, I work in the Audio / Visual industry and work with LED Video Walls. I would like to see if there is a way I can use the Raspberry Pi in my work for Video and on the LED Wall. But before I can do that I need to learn as much as I can about Linux and the Raspberry Pi. Till now my main experience has been with Windows and Mac OS and building PC's from scratch. The last time I did any real coding was back in the days of BASIC v1.0 on a Commodore 64, so I am a little rusty. Though Linux is very much like DOS in syntax and commands it is also very different. Working with getting PINN and archiving the OSes has been very enlightening and helpful in my endeavor to learn Linux and the Pi.

I'm sorry to be a pain, though I have worked with computers for the last 40+ years I have never done anything with Linux which is why I purchased the Raspberry Pi. If you don't have the time or desire, no worry's I don't want to cause you any issues or problems. It's not that important. But I will gladly accept any help or assistance given.

One other question. I noticed that PINN makes the partitions for each OS 512 MB. That seems kind of small for data storage. Does PINN setup data storage for each of the OSes installed? Will it eventually use the remainder of the 128 GB card or must these partitions be expanded first manually?

As for the time difference I understand that very well. I live in Hawaii which puts me out of sync with everywhere, don't worry about it.

Thank You for All your Help and Understanding,

Paul <aka Denavin / Eeyatahn

procount, 2020-12-03:
Paul, 
Thanks for your comprehensive answers. They provide a good starting point. I realised you were in Hawaii - a place I'd love to visit some day, maybe on my retirement world tour! I also grew up on DOS and VIC-20/CBM-64/128 Basic, so you are in good company. :smile:

As it is 1am in the UK, I will keep it brief for now. I don't suppose it matters which version of Ubuntu you use for your host environment, but as we both have 20.04.1, I would suggest you use that in preference so we can be the same.

You should follow Lurch's advice above to install build-essential first. Then you should try to build bsdtar3.3.1 again following the instructions I pointed you at before. (Remember you will need to use `sudo make install` for the last step). To check it is installed, just type `bsdtar -h` and it should report `bsdtar 3.3.1 - libarchive 3.3.1 zlib/1.2.8`

Also it would be a good idea to download/install the Raspberry Pi Imaging Utility into Windows. It's currently the easiest way to install images & even PINN onto an SD card, especially considering the size of your devices.

OK. Let me know how you get on with those tasks before we move on.

I'm not sure how/why you think there is only 512MB allocated for each OS.
Each OS requires a number of partitions: normally one small FAT32 partition of about 256MB in size for the boot files, firmware and kernel, and one EXT4 partition which holds the rest of the root file system, plus space for any of your user data. 

The EXT4 partition is normally flexible in size and will have a minimum size of the rootfs size+ 400MB. This is a bare minimum and is normally insufficient. So once you have decided which OSes are to be installed, any of the remaining space on the drive is divided equally amongst any of the flexible partitions. So you can see that if you fill up the SD card with as many OSes as you can, each one will have the barest of space allocated to it. You want to install 4 OSes, so on a 128GB card, they will each have approximately 32GB each to use. Lineage/Android is atypical as it includes 4 partitions - 3 fixed in size and one variable.

As I may have said before, importing an OS into a PINN multi-boot system is not an easy task - about as hard as it is for me to convert images into NOOBS format in the first place. It's something that NOOBS/PINN provide for using a "Custom OS" if you read the docs, but it's an area that I want to improve upon because more and more people are requesting it. Most of the constituent pieces are there to do it within PINN now, so I just have to bring some things together in a future version. But for now we have to do it manually.

Now, there are 2 ways we can attempt to do this: 
1) create a custom OS out of each of your existing OSes, and install them with PINN like you would a plain standard OS. 
2) Install new versions of each OS that you want, and then replace them with your customised versions.

2) Might be the quickest way, but it may not work for Lineage (do you have v16 or v17?) so we may end up doing a bit of both.
I think I may prefer to start with 2 for speed, but I'm also willing to do 1) if you want to do it that way, as it is a documented method in my wiki.

I'll have a think overnight, but let me know your thoughts if you have a preference.

lurch, 2020-12-03:
>  I work in the Audio / Visual industry and work with LED Video Walls. I would like to see if there is a way I can use the Raspberry Pi in my work for Video and on the LED Wall.

Lots of [video wall](https://www.google.com/search?q=raspberry+pi+video+wall) projects :slightly_smiling_face: 
LED walls: https://www.raspberrypi.org/blog/giant-led-wall/  https://www.raspberrypi.org/blog/as-we-are-face-mapping/

>  But before I can do that I need to learn as much as I can about Linux and the Raspberry Pi.

As @procount notes, trying to adapt multiple standalone OSes to all be installed by PINN is quite an advanced subject. Perhaps reading https://github.com/raspberrypi/noobs/wiki/Standalone-partitioning-explained and https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained will help explain how a NOOBS/PINN OS-image is different to a standalone OS image.

So while I can appreciate the convenience of having all your OSes on a single SD card, with your admission of your lack of Linux and Pi knowledge I think it _might_ be easier for you to stick with having each OS on its own SD card, at least for now? Once you've gained some general Linux and Raspberry Pi knowledge (there's lots of good knowledge at https://www.raspberrypi.org/documentation/ if you've not already seen it, but be aware that it only covers Raspberry Pi OS rather than any 3rd-party OSes), you might then be in a better place to attempt advanced PINN tasks?

eeyatahn, 2020-12-03:
@lurch 
Thank you for the links. The information will help much.

I admit that I am a glutton for punishment, though I know it would be much better to take my time, I am one of those people that jumps right into a project with both feet. I know that I am trying to climb a mountain without a rope or equipment but I am determined to learn this one way or another. 

To give you an idea, 3 years ago I had never touched an LED Video wall and one month after starting to work with them I was setting and running large complex walls. I won't say it was easy or that I did not pull what is left of my hair out but like everything else I stuck with it and by doing so I have learned much. 

I started when computer used Mag Tape, Magnetic Core Memory, IBM Punch Cards and Hard Drives were the size of a Washing Machine. About a month ago I purchased my first Raspberry Pi. A computer is a computer, all I needed to figure out was how it differed from everything else I have used through the years. Well a Pi is very different but in the amount of time I have had it I have learned so much. I know that I am taking this the hard way but I am determined to do it no matter how many times I may fail but we all learn from out mistakes. 

If you guys are willing to put up with my questions I will continue with my projects not matter how hard they may be. In just these few days I have learned much and must once again thank you for all your help. 

I also learned how to code the hard way. I learned BASIC by buying books, typing in the code then trying to figure out why the program did not run. It was the same with DOS, trial by fire. Though it has been many years since I have done any coding, I have the basic knowledge to understand the concept all I need to learn is the code itself which I am doing again by trial and error.

I will figure this out. Trust me I am determined. Though I may get frustrated I never give up.

Paul

eeyatahn, 2020-12-03:
@procount 
When I was removing the partitions on my failed attempt, Gparted showed me the size of the partitions that PINN had created. Other than the Fat32 Boot partitions, the partitions reserved for the OS were all 512 MB. It may have been because I never got it completely finished. That is why I asked.

procount, 2020-12-03:
Either something went wrong or you misinterpreted the output. Off the top of my head, each OS you mention requires more than 512MB, most need a few GBs. You will see as we progress.
How far have you got?

lurch, 2020-12-03:
> I will figure this out. Trust me I am determined. Though I may get frustrated I never give up

That's a great attitude to have :smiley: 

> Other than the Fat32 Boot partitions, the partitions reserved for the OS were all 512 MB.

That sounds very strange. If it happens again, could you take a screenshot?

eeyatahn, 2020-12-03:
@procount working on it now. Got BSDTAR compiled and installed. Started archiving the Raspbian OS. Moving my virtual file to a larger drive the one I put it on was too small to hold everything will need to make it biger and try again after I make the virtual drive bigger. But going good sofar.

procount, 2020-12-03:
Oh ok, so you're ahead of me!. And you're following the instructions in my wiki ->https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN#backup-raspbian ?

eeyatahn, 2020-12-03:
Minor delay had to make a new virtual disk that means I will have to start again. Could not expand the size so making a new one. Should not have any problems will make a couple of archives than try my hand as setting up the multi OS in PINN. 

Had a couple of errors to start but was able to figure them out as simply having to change a name here or there. 

Yes following your instructions and making changes where necessary to directory names or drive names.

procount, 2020-12-03:
When you have created your archives, I suggest you put your original SD cards away in a safe place :wink:
(You may have a problem restoring Lineage with this technique, but we'll cross that bridge later).
Then I suggest you install PINN onto your SD card as follows:
1. download and install Raspberry Pi Imaging Utility to your Windows PC.
2. Let it create a desktop icon (I can't remember if it asks or not)
3. Make a copy of the desktop icon and rename it to "PINN install"
4. RIght-click on "PINN Install" icon and select properties.
5. In the "target" box, add to the end of the line: ` " --repo https://raw.githubusercontent.com/procount/pinn-os/master/iu/os_list_iu_pinn.json` (So my complete target is `"C:\Program Files (x86)\Raspberry Pi Imager\rpi-imager.exe" --repo https://raw.githubusercontent.com/procount/pinn-os/master/iu/os_list_iu_pinn.json`)
6. Click OK to save.

Now when you want to install PINN to an SD card:
1. Click on the "PINN install" icon. Allow it to make changes to your device.
2. Click on CHOOSE OS and select "PINN on Github"
3. Click on CHOOSE SD card.
4. Insert your SD card in USB reader and it should appear in the dialog box. Select it.
5. Click on Write.
6. When complete, you can remove the SD card and boot it in your RPi.

This technique (hopefully) avoids all problems of having to format large SD cards as FAT32 or removing existing partitions. It is a destructive operation and will wipe everything that is/was on your SD card.

lurch, 2020-12-03:
> Minor delay had to make a new virtual disk that means I will have to start again. Could not expand the size so making a new one.

A workaround for that would have been to create an additional virtual disk, attach it to your existing VM and mount it as a new drive :wink:

eeyatahn, 2020-12-03:
No worrys already created a new Virtual disk and installed Ubuntu to it am now recreating the Raspbian archive.

eeyatahn, 2020-12-03:
Well all was going good so I left it do work on the archive of Raspbian at 1am last night. This morning I checked is and found this error. Not sure what happened. Looks like something to do with the network socket. 

> a media/plg/rootfs/tmp/.XIM-unix
> a media/plg/rootfs/tmp/dhcpcd-pi
> a media/plg/rootfs/tmp/dhcpcd-pi/libdhcpcd-wpa-784.1bsdtar: media/plg/rootfs/tmp/dhcpcd-pi/libdhcpcd-wpa-784.1: tar format cannot archive socket 
> 
> a media/plg/rootfs/tmp/dhcpcd-pi/libdhcpcd-wpa-784.0bsdtar: media/plg/rootfs/tmp/dhcpcd-pi/libdhcpcd-wpa-784.0: tar format cannot archive socket
> 
> bsdtar: Error exit delayed from previous errors.

Included are the last few before the error. Don't know if that will help but either way it look like the Archive did not complete. Archive is for the Raspbian Rootfs partition.

procount, 2020-12-03:
What exact cmdline did you use to archive it?

procount, 2020-12-03:
I have noticed that some of the lines were not displaying correctly on my wiki page so I have added some additional reformatting to make it clearer. Note that the bsdtar parameters are different when archiving the rootfs compared to the boot partition.

eeyatahn, 2020-12-03:
The one directly from your instructions.....

sudo bsdtar --numeric-owner --format gnutar -cpvf ~/os/Raspbian/boot.tar . $ cd /media/plg/rootfs

The only changes were the user and the Root name.

procount, 2020-12-03:
Please review my wiki page now I have reformatted it better.
The stuff after the $ should have been on a separate line.
And you need to use an additional parameter when archiving the rootfs (to prevent the error you just got).
Take care you execute the bsdtar command from the correct directory too.

eeyatahn, 2020-12-03:
Ok I see the change.. Will try again... Thank You

eeyatahn, 2020-12-04:
Still running into an error...

> plg@plg-VirtualBox:/media/plg/rootfs$ sudo bsdtar --numeric-owner --format gnutar --one-file-system -cpf ~/os/Raspbian/rootfs.tar .
> 
> : tar format cannot archive socket: tar format cannot archive socket

It works for an hour and this all I get back.

lurch, 2020-12-04:
[Googling for that error message](https://www.google.com/search?q=tar+format+cannot+archive+socket) turns up quite a few results, and it looks like it _might_ only be a warning, rather than an error? Or perhaps it's only a warning in GNU tar, but an error in bsdtar?

@procount It looks like https://serverfault.com/a/525812 is one of the few results I found that suggests a workaround for this error, but I dunno if the syntax used there also works with bsdtar. Although given that (AFAIK) there's no real reason to backup the contents of the `/tmp` directory (which is where these offending socket files seem to be) perhaps adding some `--exclude` flags would be the better workaround? :shrug: 

@eeyatahn Do you want to try again, but this time add --exclude=tmp to the bsdtar options? i.e.
```
sudo bsdtar --numeric-owner --format gnutar --one-file-system --exclude=tmp -cpf ~/os/Raspbian/rootfs.tar .
```

(@procount I've just done some local testing, and there doesn't seem to be a way to tell bsdtar to exclude the _contents_ of the `tmp` directory, but still create an empty `tmp` directory in the archive? So perhaps some later part of the instructions would need to be updated to recreate the `tmp` directory?)

procount, 2020-12-04:
Considering all the OSes I convert, I've not come across this one.
I remember when we used plain tar, that we excluded the "special" files (device nodes etc) by excluding /proc /dev etc, but I thought generally the --one-file-system parameter was an alternative way to accomplish that. Strange that it is not working on socket files.
I have no idea why such socket files would be in the /tmp folder, but as it is the /tmp folder, I guess they are transient and could be deleted prior to archiving? (but @eeyatahn, - I wouldn't do that unless you had a complete backup of your original SD card - you don't want to lose anything! Lurch's suggestion above does the same thing in a non-destructive way). So try Lurch's suggestion. It's always possible to copy over any files that you need from the /tmp folder later when the archive has been restored by PINN.

procount, 2020-12-04:
@lurch - It seems to be a bug in bsdtar/libarchive. See https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=233006 and there is a patch for it. Can you see which later version of bsdtar it was included in, because I can't follow it. However, I seem to recall Maxnet saying there was a problem with later versions of bsdtar (or was that something else?) 🤷‍♂️  I suppose I will have to look up the various threads on the subject.

procount, 2020-12-04:
Seems like it should have been fixed in the next version 3.3.2 😮 
https://github.com/libarchive/libarchive/releases/tag/v3.3.2
So, @eeyatahn -  maybe you could try building/installing/trying that version instead?

eeyatahn, 2020-12-04:
Ok I'll try version 3.3.2 and see how it goes.... Thanks

lurch, 2020-12-04:
> It seems to be a bug in bsdtar/libarchive. See https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=233006

That bug report is about ENOENT and is unrelated to the issue here about socket files. ENOENT might happen if you're tarring up a big directory full of different files, and one of the files gets deleted during the creation of the tar archive. Which obviously won't happen in our scenario, because we're not backing up a "live" filesystem.

> but I thought generally the --one-file-system parameter was an alternative way to accomplish that.

Again that's more of a "live filesystem" thing. If you had partitionA containing:
```
subdir1
subdir1/file1
subdir1/subdir2
```
and partitionB containing:
```
subdir3
subdir3/file2
subdir3/subdir4
```
then you could mount partitionA at `/mnt` on your running filesystem, and mount partitionB at `/mnt/subdir1/subdir2` on your running filesystem, and you'd then see:
```
/mnt/subdir1
/mnt/subdir1/file1
/mnt/subdir1/subdir2
/mnt/subdir1/subdir2/subdir3
/mnt/subdir1/subdir2/subdir3/file2
/mnt/subdir1/subdir2/subdir3/subdir4
```

If you then asked tar to archive up the `/mnt/subdir1` directory, you'd get a tarfile containing:
```
subdir1
subdir1/file1
subdir1/subdir2
subdir1/subdir2/subdir3
subdir1/subdir2/subdir3/file2
subdir1/subdir2/subdir3/subdir4
```
but if you instead passed the `--one-file-system` parameter to tar, you'd instead get
```
subdir1
subdir1/file1
subdir1/subdir2
```

Nothing at all to do with socket-files :wink: 

Of course on _some_ distros, `/tmp` is itself mounted as a tmpfs filesystem, which means it only exists in RAM, at which point a) it wouldn't exist in the "offline filesystem" anyway and b) even in a running filesystem, the `--one-file-system` parameter would exclude it. But there's no reason that `/tmp` can't just be a regular directory on the root partition.

> Considering all the OSes I convert, I've not come across this one.

As a socket file can't be archived by tar, it's unlikely to ever be in an install-image, and will probably only be in a filesystem that's actually been used at runtime.

eeyatahn, 2020-12-04:
Quick question.... How do I uninstall the 3.3.1 version of BSDTAR or will 3.3.2 over write it.

eeyatahn, 2020-12-04:
Hope this was the correct way... I just deleted the folder for 3.3.1 the compiled 3.3.2.

eeyatahn, 2020-12-04:
Trying again using 3.3.2 will see how it goes...

eeyatahn, 2020-12-04:
oops....

procount, 2020-12-04:
I suggest you first try Lurch's suggestion of excluding the tmp folder altogether.

Following Lurch's comments on v3.3.2, I'm not sure it will make any difference. But if you want to try it, I expect it will just overwrite the previous version.

eeyatahn, 2020-12-04:
Success!! Thank you @Lurch your suggestion worked... @procount  you should add that to your Wiki.

eeyatahn, 2020-12-04:
Ok well my success was short lived came up with a new error on the next part of the process.

> ~/os/Raspbian$ xz -9 -e boot.tar
> xz: boot.tar.xz: Cannot set the file group: Operation not permitted

eeyatahn, 2020-12-04:
Nope just tried that did not work either got a "No such file or directory" error

eeyatahn, 2020-12-04:
Ok, I am confused. I got an error when trying to do "~/os/Raspbian$ xz -9 -e boot.tar" but if I look in the /os/Raspbian folder there is a "boot.tar.xz" file there. Is the file corrupt and unusable? Pluse is the Archive "boot.tar" now lost and will have to be remade?

procount, 2020-12-04:
xz either compresses or decompresses a file, deleting the original.

Try `xz -d -k boot.tar.xz`
That should decompress your xz file but also keep the xz file and not delete it.

eeyatahn, 2020-12-04:
But will the error I got cause any problems...
And yes I just was looking at the commands and am trying it on the Rootfs.tar using the -k flag to see if I get an error on that one too

procount, 2020-12-04:
It should be ok. Setting the file group is not important.

eeyatahn, 2020-12-05:
Found a typo in the wiki;

> $ umount /media/\<username\>boot   
> $ umount /media/\<username\>root  
Missing the "/" after  "\<username\>

procount, 2020-12-05:
Thanks. Should be fixed now, along with some other corrections as a result of changing the formatting earlier.

lurch, 2020-12-05:
> Cannot set the file group: Operation not permitted

That's nothing to worry about - for these backup files that you're creating, the group-ownership isn't important. For more information, see https://www.google.com/search?q=user+and+group+ownership+linux and https://linux.die.net/man/1/chgrp
(providing links to extra resources, because you said you're keen to "learn more Linux stuff")

I believe the "cannot set the file group" error probably means that after creating the `boot.tar.xz` file, it wasn't able to change the group-ownership of this new file. This typically happens if you're creating a file on a filesystem that doesn't support Unix permissions e.g. FAT32.  Just ignore it as a harmless warning :slightly_smiling_face:

eeyatahn, 2020-12-05:
@procount  @lurch In the instructions on building a multi OS system for PINN you have Raspbian and RetroPI as examples. If I wanted to add other OSes to PINN I would need the .json files to do that as per your instructions. Is there someplace I can look to find out where, how or if those files can be downloaded. 

Though I will eventually want to add RetroPi to a PINN setup right now I am looking for were I can get the .json files I would need for LibreELEC / Kodi. Is there a repository of information on were these files can be found. So that I can look for them myself, so I don't have to bother you all the time. I do not want to become any more of a pest than I already am.

lurch, 2020-12-05:
I believe https://downloads.raspberrypi.org/os_list_v3.json and https://github.com/procount/pinn-os/blob/master/os/os_all.json are the "starting points" that you're looking for?

procount, 2020-12-05:
To expand on Lurch's reply, the better starting point is  https://github.com/procount/pinn-os/blob/master/os/repo_list.json which lists all the different repositories I use.

But if you can spare the download time, I suggest getting a FAT32 formatted USB drive, create a /os folder on it, and insert it into your RPi that has PINN on it. Select the OSes you want to download, then click Download on the Archive menu. This will download everything for each selected OS and would be a good starting point for modifying them into becoming your own custom OS installation.

Of course, downloading just the metafiles will result in less data downloaded so it may be quicker. But contrast that with having to search for the location and generally having to download each file manually and individually. Might be better to set PINN downloading the ones you want just before bedtime. :wink:

Also, just to remind you that this is all only necessary when you want to import an existing customised OS. Sometimes it is just easier to start by installing a new fresh OS and then copy your customisations across from your previous OS.

Also note that many OSes can be converted to NOOBS/PINN format using the method detailed in the Wiki. But not all OSes are the same. Some need very complicated conversions. For example, although I previously had Lineage16 converted, Lineage17 took me weeks to work out how to convert it and make it multibootable with some input from Konsta, so don't expect to convert that in the same way as Raspbian.

Whilst I have plans to include this functionality in PINN, the variations in OSes means it will be limited to those OSes I know how to convert and may take a while to implement.

lurch, 2020-12-05:
> the better starting point is https://github.com/procount/pinn-os/blob/master/os/repo_list.json which lists all the different repositories I use.

I just wrote a fun little python script to download every single `os.json` :grin:  https://gist.github.com/lurch/6ecc3c48c3989fa61b26721ee9c0d958
89 in total!

@procount You may be interested to hear that it detects these "errors":
!!! Couldn't download https://releases.screenlyapp.com/NOOBS/os.json
!!! Couldn't download http://le.builds.lakka.tv/noobs/Lakka_RPi4/os.json
!!! Couldn't download http://le.builds.lakka.tv/noobs/Lakka_RPi2/os.json
!!! Couldn't download http://le.builds.lakka.tv/noobs/Lakka_RPi/os.json
!!! Found duplicate os_name TLXOS 
(which I've not investigated, so they might be problems in my script)

procount, 2020-12-05:
TLXOS uses odd names. I raised it with XECDesign before, but he said they insisted.

procount, 2020-12-05:
I don't have a problem downloading those os.json files.
TLXOS uses strange OS names. I raised it with XECDesign before, but he said they insisted.

lurch, 2020-12-05:
>> !!! Couldn't download https://releases.screenlyapp.com/NOOBS/os.json
!!! Couldn't download http://le.builds.lakka.tv/noobs/Lakka_RPi4/os.json
!!! Couldn't download http://le.builds.lakka.tv/noobs/Lakka_RPi2/os.json
!!! Couldn't download http://le.builds.lakka.tv/noobs/Lakka_RPi/os.json

> I don't have a problem downloading those os.json files.

Arghh, that was annoying to debug... viewing the URLs in the browser or downloading them with `wget` always worked, but trying to download them with python always failed. I eventually managed to track down that for some reason those servers were replying with a "HTTP Error 403: Forbidden" response if the User-Agent header wasn't set in the request! [My gist](https://gist.github.com/lurch/6ecc3c48c3989fa61b26721ee9c0d958) has been updated and those `os.json` files now download correctly (so it now downloads 93 `os.json` files in total).

>> !!! Found duplicate os_name TLXOS

> TLXOS uses strange OS names. I raised it with XECDesign before, but he said they insisted.

Ahhh, they have the same `os_name` for their Pi0/Pi1 and Pi2/Pi3/Pi4 OSes (even though the `os_info` files for each are stored in separate subdirectories). Yeah I won't be updating my gist to deal with _that_ corner-case :wink: 
```
        {
            "os_info": "http://rpi.thinlinx.com/NOOBS/TLX_RPi/os.json",
            "os_name": "TLXOS ",
            "supported_models": [
                "Pi 2",
                "Pi 3",
                "Pi Compute Module 3",
                "Pi 4"
            ],
        },
        {
            "os_info": "http://rpi.thinlinx.com/NOOBS/TLX_RPi_IoT/os.json",
            "os_name": "TLXOS ",
            "supported_models": [
                "Pi Model",
                "Pi Zero",
                "Pi Compute Module Rev"
            ],
        },
```
@procount Does this mean that it's impossible to install both the Pi0/Pi1 and Pi2/Pi3/Pi4 versions of TLXOS onto the same SD card at the same time? (and is it something that NOOBS and/or PINN checks for?)

procount, 2020-12-05:
Difficult, but not impossible. I guess you would have to use ProjectSpaces and install one TLXOS from a Pi0/1 and the other from a Pi2/3/4 because the Show_all option would only show one of them, not both. Can't remember why they insisted on a trailing space in the name either.

Ha! User agents - they took me a while to sort out for PINN as well so that I could download from sourceforge & github. :smile:

lurch, 2020-12-05:
It's an extreme edge-case, but if you _were_ using `show_all`, would the same version of TLXOS get installed regardless of whether you booted PINN on a Pi1 or a Pi3?  (It's not something worth worrying about, I'm only asking out of curiosity :smiling_imp: )

> Can't remember why they insisted on a trailing space in the name either.

Oh yeah, I hadn't even noticed that :laughing: Super-weird!

procount, 2020-12-05:
Not sure, but I think it would always show the same one. I would guess the PI3 version since they would be read in sequence and I guess it would overwrite the first 🤷‍♂️ 

Just reading the Sourceforge support site and there are several open tickets about slow downloads, interrupted downloads and 503 errors over at least the last 5 days. They seem to be working on it, but it looks like it will take a while.

lurch, 2020-12-06:
>  I eventually managed to track down that for some reason those servers were replying with a "HTTP Error 403: Forbidden" response if the User-Agent header wasn't set in the request!

It had been bugging me that https://docs.python.org/3/library/urllib.request.html#examples says "OpenerDirector automatically adds a User-Agent header to every Request."
So I did even more digging and found that `urllib` _does_ set a default User-Agent header (i.e. the python docs *are* correct), but for some reason cloudfare (which is CDN-hosting the https://releases.screenlyapp.com/ and http://le.builds.lakka.tv/ domains) explicitly blocks requests with Python's default User-Agent header, but allows through requests with any other User-Agent!!
```python
>>> import urllib.request
>>> req1 = urllib.request.Request('https://releases.screenlyapp.com/NOOBS/os.json')
>>> req1.add_header('User-Agent', 'Python-urllib/3.6') # explictly set the default User-Agent value
>>> resp1 = urllib.request.urlopen(req1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.6/urllib/request.py", line 223, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.6/urllib/request.py", line 532, in open
    response = meth(req, response)
  File "/usr/lib/python3.6/urllib/request.py", line 642, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.6/urllib/request.py", line 570, in error
    return self._call_chain(*args)
  File "/usr/lib/python3.6/urllib/request.py", line 504, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.6/urllib/request.py", line 650, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 403: Forbidden
>>> req2 = urllib.request.Request('https://releases.screenlyapp.com/NOOBS/os.json')
>>> req2.add_header('User-Agent', 'python-urllib/3.6') # all we've done is change a single character
>>> resp2 = urllib.request.urlopen(req2)
>>> resp2.getcode()
200
```
Presumably this is an easy way for cloudflare to block the majority of "scripted access", while still allowing access to regular web-browsers (and people technical enough to set a custom User-Agent).

Detective Lurch solves another mystery :sunglasses:

procount, 2020-12-06:
Conversely, sourceforge takes web-browsers to a HTML download page, so I had to set my user-agent to something more "scripty" to get it to download the file directly.

eeyatahn, 2020-12-06:
Well still running into problems getting PINN to install. 
Just tried and can not get PI to start. No bios, no boot, does not even turn on video. Used the Unzip and copy method for PINN-Lite.
It's not the SD card this is a different SD card from the last time I tried to get PINN to work. 

Used SDcard.org SD Card Formater v5.0.1 to format card. Downloaded PINN-Lite from Sourceforge Unzipped and copied all files to the SD card. Put it in my PI and NOTHING.... again. Don't know what is wrong but I keep having this issue.

lurch, 2020-12-06:
You need to avoid using SD Card Formater if you're using "big" SD cards, because they get formatted to exFAT, which the RPi can't boot from. More info here: https://www.raspberrypi.org/documentation/installation/sdxc_formatting.md

eeyatahn, 2020-12-06:
Well even Raspberry Pi Imager will format a 64Gb card with exFAT retuardless of that it says. I select "Erase Format card as FAT32" and it still formats it as exFAT. Guess I will just use Format from windows. At least I can force that to FAT32 from a CLI window.

procount, 2020-12-06:
See my earlier post on how to use RPI Imaging Utility. It is by far the easiest way to program PINN on any Sd card without any of these issues.

eeyatahn, 2020-12-06:
But isn't FAT32 only going to be able to access 4.3 GB? All the files for PINN equal out to 50.1GB. There is no way windows will copy that 50.1 GB to a FAT32 drive as far as I know.

[135 more replies]

closed 2021-02-17 by procount