procount/pinn #444: Lineage OS 17.1 appears corrupt

Heclalava, 2020-12-03:
I am trying to install Lineage OS 17.1 as it is much better than 16. However it reaches a point during the download where it throws up an error message and says no response from host, and then an error message about an archive being corrupt.  I have been waiting patiently for Lineage OS 17.1 and was very excited to see it now available, however after multiple attempts to download it and install, it keeps failing. Retropie and RasPiOS 64 both download and install without problems, it is just this Lineage OS 17.1, maybe the files are corrupt on Source Forge?

procount, 2020-12-03:
I just tried a fresh install myself on a 4B v1.2 4gb with SanDisk ultra 128gb SD card. It downloaded and installed fine.
I know I had to make a small correction to lineage yesterday, but that was a slightly different problem about file format types.
It sounds more like a network or sourceforge glitch. I suggest you keep trying. Or download the whole lineage folder from sourceforge and put it in a /os/ folder of a USB stick and install from there instead.

rroblik, 2020-12-03:
Hello @procount 

How can we easily do that ?

>  download the whole lineage folder from sourceforge and put it in a /os/ folder of a USB stick and install from there instead.

I need it for my Pi3 and as I can't find in in my fresh PINN interface I try to load it ontu usb stick from another computer.

Thanks

procount, 2020-12-03:
> I need it for my Pi3 and as I can't find in in my fresh PINN interface I try to load it ontu usb stick from another computer.

If it is missing from your list of available OSes in PINN then there normally is a reason.
Are you trying to install it to USB? 
If so, it will be hidden, because it is not installable in that way.

When the next version of PINN comes out, you should be able to install it to USB, but with some restrictions that I have allowed to be relaxed.  In fact I have already released v3.5.3 as a beta if you read my thread on the forum, so you could try it with that. But please read the update notice to be aware of the limitations.

rroblik, 2020-12-03:
I've booted from usb and yes, try to install on usb ...

I'm doing it that way because on not enought space in `/os` using `pinn-lite.img`

Don't understand how to do another way ... Looking after this beta version maybe I can try

procount, 2020-12-03:
> I'm doing it that way because on not enought space in /os using pinn-lite.img

Hmmm. You're definitely confusing me now. 😕 
The /os folder is where some OS installation files may be stored. It is where the Raspbian and LibreELEC installation files are stored in the FULL version of PINN. You do not install OSes into this folder - they are installed into other partitions on the rest of your drive that may be hidden from you if you don't know how to view them.

> Don't understand how to do another way ...

Have you read the manual? Please take a look at https://github.com/procount/pinn/blob/master/README_PINN.md as I am sure it will answer many of the questions you have about how to use PINN properly.

> Looking after this beta version maybe I can try (edit : but where is this beta ?  here)

That's one location, but please go to https://www.raspberrypi.org/forums/viewtopic.php?p=1769704#p1769704 for the full details about what has changed in this version.

rroblik, 2020-12-03:
Yes I read a lot of but a little bit confusing ...
I'm trying to build from the 3.5.3 archive ...

procount, 2020-12-03:
Why are you trying to build it? It will take you hours... and to no benefit.

Just install it.
Go to https://github.com/procount/pinn/blob/master/README.md#quick-start and it will explain some ways to install it.

procount, 2020-12-03:
The link to download the beta version can be found in the post at https://www.raspberrypi.org/forums/viewtopic.php?p=1769704#p1769704

rroblik, 2020-12-03:
Damn I missed the link !
Thanks I'm trying

procount, 2020-12-03:
> You should host it elswhere !

That is "easier said than done" (to quote an English expression)

rroblik, 2020-12-03:
Juste for the smile : 
[image]

rroblik, 2020-12-03:
Thanks @procount ; installing both Pi OS an lineage OS 17 🙏🏽

@Heclalava sorry for thé little off topic. 

I also get many sourceforge glitches 
[image]

procount, 2020-12-03:
> How can we easily do that ?
>>download the whole lineage folder from sourceforge and put it in a /os/ folder of a USB stick and install from there instead.

For example, LineageOS17_rpi3 can be found at https://sourceforge.net/projects/pinn/files/os/lineage17-rpi3/

On a USB stick, create a local folder called /os/lineage17-rpi3 and download all files from the above sourceforge URL to you local folder, including the slides_vga folder and its contents.  This manual method might be more successful when sourceforge is unreliable, especially if your download supports continuation.

Alternatively, If you insert a USB_stick with a /os/ folder on it to PINN, there is a menu button in the Archive menu called Download which will download the selected OSes onto the USB stick in a similar fashion. It may still suffer similar problems, but I think the download has better resume capabilities, so you might be able to retry.

Once you have the complete list of files downloaded, you can reboot PINN and you can install your OS from the USB stick offline, instead of downloading it.

The other OSes are in parallel folders on sourceforge.

I don't know why it is being so problematic at the moment. Maybe it has a DoS attack at the moment or something. 

If you know of another free file hosting site with direct downloads and statistics that can host 100GB's of files, I would be willing to try it.

Heclalava, 2020-12-04:
OK so I am trying to downloadthe OS files manually, and create the os folder on a USB stick to install from USB, and I tried to download Home / os / lineage17-rpi4 / slides_vga / 1.jpg and immediately got a error from Sourceforge:

503 Service Unavailable
No server is available to handle this request. 

So the issue seems to be Sourceforge causing the problems. 

Any other way to distribute the OS files? Does github limit maximum file size to upload?

rroblik, 2020-12-04:
@procount 

FYI : I successfully installed Pi os x32 + Lineage 17.1 on my Pi b3 ; but I'm not able to run Lineage.

After boot OS choice menu, I have a long black screen (screen on but no image) then after about 2 minutes I have the Lineage boot animation screen ; stuck since 3 hours.

How can we try to solve this ? Another issue here ? 

Thanks

Heclalava, 2020-12-04:
@rroblik  So I managed to get RaspiOS 64, Lineage OS 17.1 and RetroPie setup on my Pi 4 by USB install. I've tested only RaspiOS so far as I've setup that first to make managing config files in the other 2 OSs easier. I'll test Lineage to see if I have any issues there when I do the first boot. ...

procount, 2020-12-04:
Are you installing to SD card or USB? LineageOS 17.1 will/should not install to USB with PINN v3.5.2 or earlier. But it should work fine on SD

Heclalava, 2020-12-04:
@procount I'm installing from a USB to a SD card. I've tested both RaspiOS 64 and RetroPie and both installed fine and run fine from the SD card. I'm in the latest version of PINN. I've previously installed Lineage OS 16 on the SD card with RaspiOS and RetroPie and had no problems with that installation. It's just Lineage OS 17.1 that's giving me hassles. I'd love to get this working instead of 16 as I've tested 17.1 on its own and it's a much more stable and snappier OS.

procount, 2020-12-04:
It was a pain to get working. Definitely worked for me locally. I'll just check downloading it again.

Heclalava, 2020-12-04:
Yeah I'm considering downloading all the files again a doing a reinstall of Lineage 17.1 to see if that works. I'll wait first to see if you find anything after a download. With all the source forge issues today to try get everything downloaded, I'm uncertain of something was corrupted during the download. I'll redo the download and reinstall and let you know how it goes.

procount, 2020-12-04:
I just managed to install LineageOS 17.1_rpi4 using PINN and downloading from sourceforge. It took a while, but it booted fine.
Then I realised I was using the latest v3.5.3 beta, so I've downgraded to v3.5.2 and I'm trying again.
I will try to note the size downloaded, but it's not unusual for it to be different - it's not exact.

procount, 2020-12-04:
Sorry, I missed the final downloaded size again, but it downloaded in about 30 mins and installed fine again.
What default resolution is your monitor? I wonder if the black screen is a symptom of the 1366x768 resolution not working on a PI4. I was going to suggest trying the other HDMI port, but I just checked it myself and LIneage works fine on either port.

Heclalava, 2020-12-04:
My resolution is 1366 x 768. However I'm running RaspiOS and RetroPie on that resolution without problems. When installing Lineage 17.1 only direct from Konstakang's site I've managed to also run it on that resolution. What problem are you referring to with this specific resolution?

procount, 2020-12-04:
Hurrah!
PI4's new graphics uses pixel doubling, which means that all aspects of the video signal needs to have even numbers.
1366x768 has an odd number somewhere so it does not work. There are workarounds, but just setting a different resolution is the easiest. Have a search on the forum, there are many posts about it.

Heclalava, 2020-12-04:
Ok interesting. I've noticed that when set to this resolution that audio doesn't work and I need to add this line to the config.txt file to get audio working:

snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_compat_alsa=0

But Lineage is now running on 1366 x 768 as HDMI group 39 is specifically for this resolution. 

I'll build on this now and let you know if I experienced any other issues. Thanks for the support.

procount, 2020-12-04:
You're welcome.

rroblik, 2020-12-05:
@Heclalava Great!

@procount I’ve maybe the same issue? But in my side I made usb to usb install under pi 3 and my native resolution is 1920x1080 ... even that I’ve noticed the Lineage animation screen ran under lower resolution (I don’t remember which)

I suspect a corrupt download from SF but I didn’t see error during PINN download (I was not front of my Pi during this long time step but I see the final PINN success message.

procount, 2020-12-05:
@rroblik - I'm afraid I have little understanding of how Lineage works. My only aim is to convert as many useful OSes as I can to be installable by PINN in multiboot fashion. Once installed, they should be the same as a direct image installation, in all but the few changes I had to make for it to be multibootable, at which point I refer you to the OS maintainer, especially for Hifiberry support etc. 

So you are using an RPi3 and installing LineageOS17.1_rpi3 onto USB? 
That's subtley different to @Heclalava's situation, so the 1366x768 resolution should not apply to you.

I'm not sure if this is correct but I seem to recall Lineage 16 used a consistent 1280x720 resolution. Lineage 17 seems to start off in that resolution, but will later change to a higher resolution according to the EDID output of your monitor, hence why it can use the 1920x1080 resolution, for example. But the initial startup still seems to be in 1280x720.

rroblik, 2020-12-05:
Yes. 

I just tried to a native lineage image from usb stick (without PINN) and there is a crash after time/time zone settings. Unable to successfully run it...

I will ask on Lineage side. Thanks for your help and your time !

oijkn, 2021-01-06:
During my first attempts I also had errors during the installation, so I followed your advice by putting the OS on a USB key in a directory named `os`. When I tried to install locally via the USB key everything went perfectly.

On the other hand I notice big problems of slowness (my old Nvidia Shield is much faster) yet I have the 8G version of pi 4. For example the installation of the Molotov application takes a lot of time and when I want to watch a TV channel, it doesn't work.

Even overclocking CPU at 2000 does not improve the situation.

Any advice?

@procount Maybe can you make LineageOS 16 atv available for testing please?

procount, 2021-01-06:
PINN is just an installer for OSes. 
After installation, the OS will behave exactly the same as if it were installed as an image. So if you have any issues with the OS, you should contact the OS maintainer. In this case Konsta has a thread on the Raspberry PI forum -> https://www.raspberrypi.org/forums/viewtopic.php?f=73&t=281187

I may take a look at 16-atv when I get some time. Do you think it will be faster?

oijkn, 2021-01-06:
Yes I'm agree with you, PINN is only an installer but perhaps some people have the same problem and can share their experience. 

> I may take a look at 16-atv when I get some time. Do you think it will be faster?

I really don't know if it's faster or not, so if you make it compatible with your installer then I can try it 😉

Heclalava, 2021-01-06:
Yes Lineage OS is slower, remember that it's Android adapted for non Android hardware. It won't have recently the same functionality as an Android device like a tablet or TV box. I had to tweak the hell out of the build prop to improve speed and a lot of system wide tweaks and extra third party apps to improve functionality. Even then it's still not performing as well as my Android TV box. However that fact that it can multiboot alongside other operating systems like RetroPie, RaspiOS etc makes it way cooler. And installing from USB definitely makes it installation far easier and quicker.

closed 2021-02-17 by procount