procount/pinn #445: RetroPie 4.6 multiboot from USB Hard drive

RandomSamurai, 2020-12-04:
I am currently trying to get a multi boot to work on a Pi4-4gb in a Nespi4 case. 
I have Pinn installed on a small partition on an SSD (pi has already been set up to run from usb when no SD is installed), which boots without an issue. I select the three OSes I want to use.
Rasbian, OpenElec and Retropie. 

On finishing the first 2 work without an issue. However, RetroPie throws up errors.
    start4.elf: is not compatible
    USB-MSD boot requires newer software

The interesting thing is that if I start again and install just RetroPie, then no issues.

I will try setting it up with SDcard as the boot device, but still pointing to the SSD for the main installs.  While I can't 100% remember, but I think in that case all 3 OSes worked. If it does, I suppose I could run it that way, but really wanted to run just from the SSD.

procount, 2020-12-04:
I'll run some checks on Retropie. It may just need the firmware updating as you say.

achessoux, 2020-12-26:
I have the same issue with 2020-09-03 firmware installed. It is the latest critical (the stable version of December does not work - bug is raised in firmware github).
Raspbian, recalbox, librelec work fine, but retropie... Starts.elf is not compatible...
If anyone has an idea....

ur7x, 2021-01-27:
I'm trying to boot from a x872 with a Samsung 970 Pro NVMe SSD.  Requires a plug/unplug for the RPi4 to see the SSD.
I know that Geekworm does not recommend Samsung Drives due to "EMF interference problems"  Does anyone have a list of tested SSDs that work with the x872 adapter?

achessoux, 2021-01-27:
It should solve your issue : 
https://github.com/raspberrypi/firmware/issues/1515#issuecomment-761245134

ur7x, 2021-01-27:
Thanks, tried that.. the usb plug unplug problem remains.

DerangedPineapple, 2021-04-03:
Same thing happens to me with Retropie 4.6.

I have usb ssd 128gb and installed LibreELEC 9.2.6 (updated to latest beta) and Retropie 4.6 and I get the same message USB-MSD boot requires newer software when trying to boot Retropie.

Rspi bootloader is 18.3.2021 1b43d5b6

LibreELEC boot and works just fine. Maybe more up to date Retropie would work like 4.7.1? 

Any guide how to update Retropie within PINN?