procount/pinn #446: [Feature Request]

Heclalava, 2020-12-07:
I know you're probably very busy, but it would be great if Kali Linux 64 bit for RasPi 4B were available in PINN. I have installed on a separate SD card, and it works great. I would love to be able to multi boot with Lineage 17 and Retropie.

procount, 2020-12-07:
You're not wrong there!
See #382. The 64bit version is working, but the 32bit isn't.
I'll try and find some time to release the 64bit version.

Dimensional, 2020-12-17:
I noticed that in the newer versions of Kali Arm, 2020.4, they use the ext3 FS instead of ext4. That may be the problem for the 32bit version you are trying to run, which is running the 2020.3 update, but I'm unable to check. I'm working on getting 2020.4 update converted to Pinn format as we speak.

Heclalava, 2020-12-17:
@Dimensional I'm actually running the 64 bit version from that separate SD card. That I downloaded directly from offensive security website.

Dimensional, 2020-12-17:
Yes. I've done the same. I'm trying to convert them into formats for Pinn for multibooting. And I downloaded them from the offensive security website as well. Wrote them to a separate SD, booted to make sure they worked on their own, and then shut off and dumped following the instructions found in the wiki.

procount, 2020-12-28:
Kali 2020.04 now available

closed 2020-12-28 by procount