procount/pinn #448: Raspberry Pi OS + Recalbox installation fails

lmutt, 2020-12-12:
I tried to install "Raspberry Pi OS Lite (32-bit)" + "RECALBOX" with latest PINN (from https://sourceforge.net/projects/pinn/files/latest/download), but in the end of installation appears a dialog with the message "OS(es) Installed with error. See debug log for details." and an "OK" button. Clicking in this OK button, the hardware reboots and boots in "Raspberry Pi OS Lite (32-bit)" successfully. Rebooting again from this OS, in the PINN menu, Recalbox does not appears as an option (the menu flashes so fast that there is no time to read). With shift pressed, in "recovery mode" I was able to confirm that the Recalbox was not listed as installed. After this, inserting the SD card in my PC, I've checked that the partition called "RECALBOX" is empty.

What could be the problem? I've tryed the same setup with NOOBS and both OS works.

Please see the debug file collected from /tmp/debug, attached.
[debug.txt](https://github.com/procount/pinn/files/5682286/debug.txt)

PS: This is my first trial on PINN after discovering that NOOBS does not support HDMI CEC or Joystick input (there were some issues with CEC and Joystick with PINN, but this is a minor issue that I will try to resolve later). My hardware is an Raspberry Pi 1B, with an 16 GB SD card.

procount, 2020-12-12:
Recalbox is a new version that has just been released for NOOBS by it's maintainers. I've not even tried it myself yet.
Thanks for including debug.txt. it indicates the setup script failed, which causes PINN to delete the installation.
It's odd that it works in NOOBS though.
It probably needs some changes to it's meta files.
I'll investigate the issue.
Thanks for reporting it. Sorry you found this on your first use of PINN.

annaclets, 2020-12-12:
Recalbox's partition image is web based. It may have simply failed to download.

The solution is simply just try again.

lmutt, 2020-12-12:
> Recalbox's partition image is web based. It may have simply failed to download.
> 
> The solution is simply just try again.

I've tried 2 times.

procount, 2020-12-12:
There are some issues with their metadata, like the OS name for PI0 and PI1 has a '/' in it, which doesn't play well
I had offered to review their files, but I haven't had time to yet.
It will require some changes, so please be patient for a while.

lmutt, 2020-12-12:
> There are some issues with their metadata, like the OS name for PI0 and PI1 has a '/' in it, which doesn't play well
> I had offered to review their files, but I haven't had time to yet.
> It will require some changes, so please be patient for a while.

Thank you very much. I will wait.

lmutt, 2020-12-16:
Wouldn't it be simpler to adapt PINN to be more robust with OS metadata than trying to convince an OS to correct its metadata? I also remember that it works in NOOBS (that I abandoned because it requires keyboard/mouse to select the OS).

procount, 2020-12-16:
I will update PINN anyway. It's only the pi0/1 version that doesn't work. The pi2 pi3 and pi4 should all download ok.
But none of the extra PINN features will work with recalbox atm, like download or backup

tommyrydel, 2020-12-16:
I have the exact same issue mentioned. "OS(es) Installed with error. See debug log for details." and the Recalbox was not installed and empty when I open it on my PC.
I'm running Raspberry Pi 3B+ and a 32 GB SD card. 

I'm new to this world so maybe I'm doing something wrong, but exactly the same issue.

procount, 2020-12-16:
Recalbox is not correct at the moment. I will work with the maintainers to fix this. Please be patient in the meantime.

tommyrydel, 2020-12-16:
I also tried the Retropie through PINN and it installs v 4.6, but it does not load any Roms. I can see all of them in my list, but they don't load. I have tried to install the latest version 4.7.1 of Retropie on a separate SD card and it works fine. Any ideas?

procount, 2020-12-17:
No idea. It worked last time I tried it. I will put it in the queue (which is unusually very long at the moment....)

ernesto-barroso, 2020-12-18:
the same here: i've just tried it and get the same error: «OS(es) Installed with error».

i'm also running a RaspBerry Pi 3 and it also worked with NOOBS (a dual installation, LibreELEC + Recalbox).

is there any other way to make it work?

anyway, i would like to thank you your work in PINN: [i can use my gamepad in the boot menu!](https://cuaderno.poderna.com/linux/noobs-o-pinn-libreelec-recalbox/#reconfiguracion_del_mando_de_juego_usb)

procount, 2020-12-18:
> the same here: 

You mean Recalbox? - same answer as above, or LibreELEC?

ernesto-barroso, 2020-12-18:
recalbox, sorry. i mean i've tried to install libre elec + recalbox with PINN and recalbox failed to install (it does work with NOOBS to me).

hyperbrute, 2020-12-21:
Hi,
Same error «OS(es) Installed with error» for me with raspberryos+libreelec+recalbox with PINN on RaspBerry Pi 3 on 64GB SD card.
If it could help :
When I look at recovery.cmdline :
`runinstaller quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline no_group no_default_source select="allnetwork" disablesdimages alt_image_source=http://pinn.mjh.nz/os_list.json?raspberrypiosfull32-bit=14375&libreelec_rpi2=14375&recalbox-pi3=14375&datapartition=14375 loglevel=2 sdhci.debug_quirks2=4`
and i look at os_list.json :
https://pinn.mjh.nz/os_list.json?raspberrypiosfull32-bit=14375&libreelec_rpi2=14375&recalbox-pi3=14375&datapartition=14375

procount, 2020-12-26:
I have created an alternative set of meta files to allow PINN to install Recalbox.
Please use the versions called "recalbox-rpiX".
Also use the v3.5.4 of PINN which additionally supports the exfat file system required for recalbox 7 backups.

hyperbrute, 2021-01-01:
Hi @procount 
I just tried with the new version of PINN.
Exactly the same error than before at the end of the installation.
(i don't understand what you say "Please use the versions called "recalbox-rpiX"",
is it special versions of PINN ?!)

Happy new year anyway and thank you for your work ;)

procount, 2021-01-01:
You should now see TWO versions of Recalbox in PINN under the Games category. If you are using a PI3, you will see the original recalbox called "Recalbox - Pi3" and a new one called "recalbox-rpi3". If you use the original one, you will still get the same error. So use the new "recalbox-rpi3" version with the shorter description.

hyperbrute, 2021-01-01:
Ok. 
Yes i used the new one when i choose it on https://pinn.mjh.nz/.
I understand my error : I download the new recovery.cmdline BUT used the old one instead... 
I'll try the new one

Thank you !

procount, 2021-01-01:
I've not tried it via Matt's webpage. It usually takes 24 hours to update with any changes. Plus you need to re-download your recovery.cmdline

redm123, 2021-02-14:
The issue is already closed, but some thoughts nonetheless, as I also struggled with this one:

> You should now see TWO versions of Recalbox in PINN under the Games category. If you are using a PI3, you will see the original recalbox called "Recalbox - Pi3" and a new one called "recalbox-rpi3". [...]

That's pretty confusing. Entirely unclear why there are two, and what's the difference. I went for "Recalbox - Pi3", because it looked ... better. First option would be to support the original Recalbox version. Second best, give the other one a better description, that it is a PINN-fixed version which should be used instead.

And the error message says I should check the log. I can only click OK, and *poof* it's gone and reboots. After re-entering PINN the debug log does not show anything useful (I suppose I'd have to login through SSH or whatever). It would be great if it would allow me to actually view the log (like a button Show Log, which opens the log window). And it might tell me which system had the error (I can only guess)...

procount, 2021-02-15:
The one that doesn't work comes direct from the recalbox maintainers for NOOBS. It is not 100% correct for PINN. I have told them how to change it but they haven't done it yet, so I added one that does work. Maybe I need to make the description clearer.

procount, 2022-02-14:
Recalbox has been updated by its maintainers and I have little control over what they do. Their metafiles are quite different to other OSes, but they should work with PINN. Let me have a look and I'll try and replace them with better versions like I did before.

ChristopherLKL, 2022-02-15:
Many thanks.

Envoyé le :lundi 14 février 2022 21:32
À : ***@***.***>
Cc : ***@***.***>; ***@***.***>

procount, 2022-02-16:
I have done some quick changes to allow Recalbox to be installed without an error message. 
Please reboot PINN TWICE and it should then install properly.
However, there are still some issues with it:

1. Backup seems to be enabled, but I haven't checked it.
2. Pi Zero 2 is not selectable. A workaround is to add `showall` to recovery.cmdline (or use the Edit Options under the Edit Config dialog) and then install the version for the Pi3. On their website, they include a separate version for the Pi Zero 2, but there is no corresponding NOOBS version. I don't know what the difference is yet.

I will try and create a better set of metafiles for Recalbox to avoid the duplicate sets, but it should be installable now.

simmtek, 2023-08-06:
I have just got this error, it worked fine a couple of days ago, then they did an update and now I get this error on Pi4-8gb