procount/pinn #449: Multi Boot with Android (OmniROM)

Adam90a, 2020-12-15:
I am attempting to dual-boot Raspberry Pi OS lite with OmniRom (Android 11)

Several problems:

1) I have a Raspberry Pi 4, I have used the OTP code to boot from USB, so unfortunately now I cant use SD for something like BerryBoot. I can only boot from USB.
 
2) OmniROM is a large image file, around 8GB and will need to be on an ext3 or ext4 usb drive seperate from PINN. 
(FAT32 has a maximum individual file size of 4GB)

3) OmniROM is not an image offered by PINN through the normal menu.

I would love help getting this project moving, I am lost in a few ways so anything would help and I can update if I find a solution to the whole project. I understand that problem #1 is permanent. Please let me know if more information is needed or would be helpful.

procount, 2020-12-15:
1. Option 1 was never permanent, even on a pi3. In a Pi4 even less so because you can change the boot order as you wish - it is not OTP. Only the flash bit in pi3 was otp to enable USB boot. It doesn't prevent SD card boot. Just put an sd card in on its own. On a pi4 you select the order of boot devices to look at, so it could be USB then SD or SD then USB, for example. It is documented on the rpf website.

2. Use an ext4 formatted USB drive instead.

3. Very true atm. 

I converted rtandroid and Lineage, so hopefully OmniROM will also be successful. If you want to have a go, look at my wiki about converting OSes for PINN. Also read README_PINN.md in its entirety. Then look at partition_setup.sh for Lineage17 for inspiration. 🙂

Adam90a, 2020-12-16:
Thank you! I have been working on it. A little out of my level of knowledge but I am learning so that's good. Will update here if I can figure anything out.

Adam90a, 2020-12-16:
I think I have cobbled OmniROM together incorrectly. The install starts then says, 
"An incorrect file checksum has been detected in /tmp/media/sda1/os/OmniROM/boot.tar.xz"
I will try restarting unless this is an indication of something I am doing wrong, any thoughts?

procount, 2020-12-16:
Did you calculate a new checksum for that file?
Did you put the new checksum in partitions.json?
If you are having problems just delete the checksum lines from partitions.json and it will install the file without checking.

There is also another checksum  in os.json for partition_setup.sh