procount/pinn #451: Use SD as storage when using USB installation?

EctoOne, 2020-12-17:
Hi, I am not very familiar with Linux so I apologize if this is a simple question.

Is it possible to use the spare space on the SD card as storage for any/some of the Distros which can be installed to a USB drive?

In my case, I only have RaspiOS lite installed on a SSD. And I'm using a different USB thumb drive to store/transfer some files. Including some files which I barely need to access but need to keep around.

So, I was wondering if it would be possible to use the unused space of my SD card as extra storage which would remain even if I chose to install something new on the SSD.

The space doesn't even need to be automatically mounted (which I would actually prefer to keep my file manager clean) if It would be possible to create a simple script to mount it when needed.

EctoOne, 2020-12-18:
Well, I used gparted to create a new partition on the SD. After that I had to reinstall my system because I the boot process stopped when it tried to mount the partitions. Or something like that, can't remember what exactly the line was but it was something with `/dev/disk/by-uuid`. But since I had to do a clean installation anyway, that was fine.

Now, the new partition shows up in my file browser, but I have not accessed it yet. Because I'm still not sure if this is a good idea and if a PINN update might mess up this setup. Since even installing a new system didn't mess with the SD, I assume an PINN update would be save as well.

Any information would be very helpful.

procount, 2020-12-18:
A simple reboot might have solved it rather than a reinstall so that the kernel knows about the new partition.

Is PINN on your SSD drive, or is it on your SD card, with the boot partition on SD card and the root partition on SDD?

If PINN and all of Raspios LITE is on your SSD, then you have no problem as PINN will not be using your SD card.

If you are booting PINN and RaspiosLITE from your SD card (with rootfs on SSD), with your additional data partition on the SD card, then it will be ok UNLESS you decide to INSTALL other OSes. Such action will erase the SD card along with your data partition. 
REINSTALL or REPLACE operations should be ok though.

PINN self-update only affects PINN and none of the other partitions.

EctoOne, 2020-12-18:
I tried rebooting twice and the issue didn't get resolved. But like I said, that was no real issue.

I'm not entirely sure about where which partition is. I have copied the PINN files onto the SD, and from there I installed RaspiOS on the SSD.

I used REINSTALL to fix my boot issue and if I understand you correctly, that's why I still have access to the data partition.

But would it be possible for example to install PINN on an SD, and then install the available 500mb data partition (or whatever size that thing has) on top there SD as well. Which I assume is just what it says it is, a singular empty partition.

And in a second step install any other OS onto an USB drive and still have access to the data partition or would that be wiped?

procount, 2020-12-18:
> And in a second step install any other OS onto an USB drive and still have access to the data partition or would that be wiped?

There is no (current) way to install an additional OS at a later time. All OSes have to be installed at the same time, since PINN shares all of the drive space between all the OSes, so there is no room left. The only way is to install some ProjectSpaces in your first step, then replace one of them with a different OS later.

FYI - most OSes comprise 2 partitions, a small FAT boot partition and a larger EXT4 root partition. When PINN is installed on the SD card and you install OSes to an SSD, the boot partitions get stored on the SD card, and the root partitions get stored on the SSD.

Some RPi models support USB boot. If this is enabled, you can install PINN to the SSD. When you install OSes in this way, both boot and root partitions will be stored on the SSD. You can then use the SD card freely for data transfer.

EctoOne, 2020-12-19:
It was not my ideas to install an additional os later. It was more of assigning an empty partition in the first step on the SD and then install the OS on a USB drive.

I was thinking about an feature where, when you select any USB drive as installation device, an extra option would show up where you can choose if the remaining space on the SD card should be used as shared store device. But since the root partitions are still on the SD, that partition wouldn't survive any additional changes.

Thanks for the info though, I knew that an OS could have multiple partitions but I don't know that the boot partition was installed on the SD. That actually explains why I had some issues with Timeshift when I tried it a few days ago.

So, to summarize. My setup should be fine as long as I use REINSTALL if I need to reinstall my system.

closed 2020-12-19 by EctoOne