procount/pinn #453: Request: apply change to every os you install

Ae3NerdGod, 2020-12-24:
I don't think noobsconfig is what I want, but if there's a way to do it with noobsconfig I'll try it

I have a device with a pi3 compute board in it. I have to manually add a few things to config.txt for the display to work. I'd like a way to append these lines to bottom of config.txt no matter what OS I install, is there a currently available way to do that?

procount, 2020-12-24:
At the moment, the "noobsconfig" feature is quite basic involving just copying or replacing files, but not the alteration of existing files. Therefore you'd need a separate "flavour" that installed a modified config.txt file for each OS you want to use.

But I can see this would be a useful feature. One way is to allow a script to be used as part of "noobsconfig", but I do worry about the security implications of this. Another way would be to add a method of adding or altering lines in such files, or allowing things like `sed` expressions.

lurch, 2020-12-24:
@procount As a simpler "solution", I guess you could add some logic so that if there's an `extra_config.txt` in the PINN partition, it's automatically appended onto the `config.txt` in the boot (FAT32) partition of any OS that PINN installs? Bit of a "blunt" approach but probably easier than messing about with scripts or `sed` expressions?

Ae3NerdGod, 2020-12-24:
I was gonna mention something like what @lurch said. Maybe config_persist.txt and it could be copied to the boot partition and "include config_persist.txt" appended to original config? Least amount of modification to the original os