procount/pinn #454: Using PiDrive 375 GB USB MSD / migrating from berryboot

mar1ged, 2020-12-25:
I want to stop using BerryBoot and switch to PINN exclusively. During this I will reformat all storage media, so converting is no issue and I will simply start from scratch.

In one of my Raspberries (3B+) I use a USB MSD PiDrive with 375GB and Berryboot was able to boot from that one directly. As far as I understood PINN should be able to do the same: https://github.com/procount/pinn/blob/master/README_PINN.md#option-3-boot-and-rootfs-on-usb

From what I understand I need to install PINN on the PiDrive but this is where I have a problem in understanding. In order to put PINN on the drive I see two approaches:

1. connect the 375GB PiDrive to a PC, format it with FAT32
2. boot PINN from MicroSD in ssh mode, login from a PC and format the PiDrive / download and unpack the PINN lite zip

But in both scenarios I would have a harddisk with a huge FAT family file system. Is this the way to go or is there a third alternative ?

Dimensional, 2020-12-25:
You don't have to worry. When you run PINN off the fat32 formatted partition, it'll copy all of it's files into memory and then reformat and repartition the drive, shrinking the PINN fat32 partition down to the bare minimum and opening up space for all the other systems you plan on installing.

procount, 2020-12-25:
Have a look at this post in my sticky thread -> https://www.raspberrypi.org/forums/viewtopic.php?p=1638508#p1638508
It saves formatting the whole SD card as FAT32.

mar1ged, 2020-12-26:
Just for your information and future reference: I used an ugly hack by simply booting into recovery and `dd if=... of=... bs=4M`ing the microSD to the MSD drive. I was then able to boot from USB and install the desired operating systems. The partitions coming from the 32GB microSD card have been automatically resized to properly fit the 375GB drive.

As mentioned: ugly, but functional ;-)

closed 2020-12-26 by mar1ged