procount/pinn #455: Download OS images without PINN

mar1ged, 2020-12-26:
Perhaps my understanding is incorrect, because of this I wanted to ask this question.

I learnt that when I start PINN I can download OS images to an attached USB device. But how can I download them for example on my Windows or Ubuntu machine ?
I have a rough understanding of how the repo and various JSON files - especially those for the partitions and tarballs - work and I think that I even found the [sourcecode](https://github.com/procount/pinn/blob/master/recovery/multiimagedownloadthread.cpp) where PINN does the downloading.
But could I somehow differently download the files ? I would like to store them on MicroSD or a small USB stick when I setup a MicroSD for later install

procount, 2020-12-26:
Most of my OSes are stored in a folder on sourceforge, so just download the complete contents of whatever OS folder you want. Put the folders under a /os/ folder on your USB stick for them to be recognised.

lurch, 2020-12-27:
If you download the "big" `pinn.zip` from https://sourceforge.net/projects/pinn/files/ you should see how the files & folders need to be arranged.

closed 2021-02-17 by procount