procount/pinn #456: Boot the first OS in list

TurboGraphxBeige, 2020-12-27:
We can change the order of the boot menu and by default PINN boots the last booted OS.

Can we make it the default behavior to boot the first OS in list instead of the last one? Something more like GRUB.

procount, 2020-12-27:
Use the "Sticky Default" checkbox to select the one you want to boot. It doesn't have to be the first in the list. (List ordering is just for aesthetics).
To temporarily boot another OS, press SHIFT when PINN boots, then ESC to get to the boot selection window again. Then either choose another OS to boot for this one time, after which it will revert to the sticky default, or change the sticky default to boot another one each time, or remove the sticky default to boot the last OS booted.

TurboGraphxBeige, 2020-12-28:
I think what I would need is the Sticky Default option but with the boot menu still visible without pressing something. Maybe there's already a way to do that and I am missing it?

procount, 2021-02-17:
The sticky default is a way to boot as quickly as if there was only 1 OS installed.
You can use forcetrigger + remotetimeout for similar functionality although it will pause on the main menu for a short time first.
I guess you want a sticky default with a time delay.

voggur, 2021-05-06:
sticky do not work for my as the hdmi cec dos not work after running OS that do not support CEC and there for I get stuck in that OS.
If a option to disable the update of "default_partition_to_boot" to static would solve this for me.
(Also I find this behavior surprising as the main reason to reboot would be to run other OS and not the same again)

lurch, 2021-05-06:
> (Also I find this behavior surprising as the main reason to reboot would be to run other OS and not the same again)

...unless you've installed some system updates, and you need to reboot the OS for the updates to take full effect? :wink:

k2helix, 2024-08-05:
It's been a while since this issue was created. However, this is indeed an enhancement for PINN that should be considered as it has many use cases.

Consider the case where the user wants to run a specific OS by default without the need to turn on the screen or connect via VNC (e.g. headless raspbian), but has a secondary OS where the user boots and needs the screen (e.g. Android TV).

Or simply two OSes that the user boots indifferently but are intended for distinct meanings.

This would make PINN work like Grub in the sense that not only is there a default OS to boot but also a different OS can be selected in case it's needed.

Although the forcetrigger + remotetimeout workaround makes the trick, it is way more tedious as the user is presented with PINN's main screen, next if they want to boot another OS exit the menu and then they can choose the OS.

But I can't tell if it's worth the time to implement this feature or leave the behaviour as is.

procount, 2024-08-05:
I think the Sticky Default satisfies the requirement to boot a preferred OS each time it is turned on. But the selection of an alternative OS to boot occasionally is the bit in question.
Perhaps all you are wanting is to swap around the order of the Boot menu and the Recovery menu? So if you don't want the Sticky Default OS to boot, you can go straight to the Boot menu and select an alternative OS without going through the Recovery Menu? (although it is only a single press of the 'Esc' key to dismiss the Recovery menu).
This would require an architectural change, since currently it is not possible to go from the Boot Menu to the Recovery Menu without rebooting.

k2helix, 2024-08-05:
Maybe it would be easier to add a new setting 'forcebootselection' or 'stickyskipsbootselection' 

I mean, that way it wouldn't require an architectural change. If PINN is able to know which OS was booted last time and select it in the boot selection screen, it should also be possible for it to select the sticky OS, right?

So the option would be either true or false, and depending on that PINN would boot directly to the sticky OS or not. For instance, if Raspbian is the sticky OS and forcebootselection is set to false, the behaviour remains as is right now. However, if it is set to true, then PINN shows the boot selection screen and has Raspbian already selected, as if it was the last booted OS.