procount/pinn #457: I can't run Lanka or Debian64 installed with PINN

cpcbegin, 2020-12-30:
I install some systems in my new raspberry Pi using PINN but when I select to run Lakka and Debian 64 show the enclosed error. 
[image]

procount, 2020-12-30:
Are you trying to boot these from USB? Even if not, both of these were created prior to the Pi400 coming out so need new firmware to boot on a Pi400 (as the error message indicates). Even then, it may not be sufficient.
Try updating their start.elf and fixup.dat files from the latest Raspbian image and they _may_ then work.

Note that these are marked as suitable for the Pi4, but that this also includes the Pi400, which is not always appropriate. If updating the firmware does not work, you will have to wait until the maintainers update their image to include the additional drivers for the Pi400.

closed 2021-02-17 by procount

cpcbegin, 2021-02-21:
No, I try to boot from microSD.