procount/pinn #458: Can't boot with a second USB attached

ultraviolet74, 2021-01-06:
I've installed PINN without issue and it boot regularly every os I've installed (RaspbianOS 64bit, LibreElec, Retropie, Twister) and I'm using an ArgonOne m2 case which contain a SSD WD Blue 500MB. In RaspbianOS 64 which is the one I use most, I've installed OMV so I need to plug another USB drive as storage. If I plugthe USB drive (a Maxtor M3 2TB) after OS started, it will be recognized and it works without problem, if I reboot OS with maxtor attached, RaspbianOs won't start and after PINN it show black screen and only red light on raspberry RPi 4 8GB with latest update and latest firmware and latest bootloader.

procount, 2021-01-06:
So you're booting PINN & your OSes from the SSD WD Blue on USB.
PINN boots fine, even with your Maxtor plugged in, but Raspbian OS 64 will not boot from PINN when selected if the Maxtor is plugged in?

ultraviolet74, 2021-01-06:
exactly, but at the moment I'm not tried it with the others OSes installed

procount, 2021-01-06:
Please post the contents of your /settings/installed_os.json file.

ultraviolet74, 2021-01-08:
Really sorry for my delay in answer, here are the content of the files you requested:
installed_os.json:
{
  "bootable" : true,
  "description" : "LibreELEC is a fast and user-friendly Kodi Entertainment Center distribution.",
  "folder" : "/settings/os/LibreELEC_RPi4",
  "group" : "Media",
  "icon" : "/settings/os/LibreELEC_RPi4/icon.png",
  "name" : "LibreELEC_RPi4",
  "partitions" : [
"PARTUUID=dd07b8ce-08",
"PARTUUID=dd07b8ce-09"
  ],
  "password" : "libreelec",
  "release_date" : "2020-10-24",
  "supported_models" : [
"Pi 4"
  ],
  "supports_backup" : true,
  "url" : "https://libreelec.tv/",
  "username" : "root"
 },
 {
  "bootable" : true,
  "description" : "A 64-bit BETA port of Debian with desktop and recommended applications",
  "folder" : "/settings/os/raspios64",
  "group" : "Testing",
  "icon" : "/settings/os/raspios64/icon.png",
  "name" : "raspios64",
  "partitions" : [
"PARTUUID=dd07b8ce-0a",
"PARTUUID=dd07b8ce-0b"
  ],
  "password" : "raspberry",
  "release_date" : "2020-05-27",
  "supported_models" : [
"Pi 3",
"Pi Compute Module 3",
"Pi 4"
  ],
  "supports_backup" : true,
  "url" : "http://www.raspbian.org/",
  "username" : "pi"
 },
 {
  "bootable" : true,
  "description" : "Retropie4 v4.6 gaming distro using Emulation Station for RPi2/3B+",
  "folder" : "/settings/os/Retropie4",
  "group" : "Games",
  "icon" : "/settings/os/Retropie4/icon.png",
  "name" : "Retropie4",
  "partitions" : [
"PARTUUID=dd07b8ce-06",
"PARTUUID=dd07b8ce-07"
  ],
  "password" : "raspberry",
  "release_date" : "2020-04-27",
  "supported_models" : [
"Pi 4"
  ],
  "supports_backup" : true,
  "url" : "https://retropie.org.uk/",
  "username" : "pi"
 }
]

and here cmdline.txt:

console=serial0,115200 console=tty1 root=PARTUUID=dd07b8ce-0b rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles

I did some other test and with the second hdd attached every so doesn't start. In retropie it starts but it give an error sdb (I didn't sign exactly, I should retry and tell you...)

lurch, 2021-01-09:
Are there any files (e.g. start.elf) on the second USB drive that the bootloader might try and boot from? I think you can use `USB_MSD_EXCLUDE_VID_PID` to tell the bootloader to explicitly ignore your second USB drive? https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md
(although the bootloader runs _before_ PINN, so if you always see PINN then this isn't the problem)

procount, 2021-01-09:
I was wondering if the second HDD has somehow got the same PARTUUID references as the SSD (e.g. by cloning) so the RaspiOS boot loader is sat waiting for the wrong rootfs partition to be ready as it won't have the correct files on it 🤷‍♂️

ultraviolet74, 2021-01-09:
I've tried with two different external autopowered HD one maxtor 2TB and another 1TB on sabrent box. I always see PINN but when SO should start it don't!

lurch, 2021-01-10:
@ultraviolet74 Can you boot RaspiOS (from PINN) with just one USB drive connected, run `lsblk && blkid`, plug in your second USB drive, run `lsblk && blkid` again, and then plug in your 3rd USB drive and run `lsblk && blkid` one more time? And then copy'n'paste the results here :slightly_smiling_face: 
(reason for doing it one at a time, rather than only at the last step, is so that we can see what changes each time, rather than having to guess at which output corresponds to which drive)

lurch, 2021-01-10:
Hmmm, when you say "autopowered" do you mean they're powered from just the Pi's USB ports, or do the HDs have their own external PSU?

ultraviolet74, 2021-01-11:
here's are the three lsblk && blkid:
pi@raspberrypi:~ $ lsblk && blkid
NAME    MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda       8:0    0 465.8G  0 disk
├─sda1    8:1    0    63M  0 part
├─sda2    8:2    0     1K  0 part
├─sda5    8:5    0    32M  0 part /media/pi/ee58078a-d1d4-4f78-b93c-6bebce850066
├─sda6    8:6    0   256M  0 part /media/pi/boot
├─sda7    8:7    0 155.2G  0 part /media/pi/retropie
├─sda8    8:8    0   512M  0 part /media/pi/prjboot
├─sda9    8:9    0 153.4G  0 part /media/pi/prjroot
├─sda10   8:10   0   253M  0 part /boot
└─sda11   8:11   0   156G  0 part /
/dev/sda1: SEC_TYPE="msdos" LABEL_FATBOOT="RECOVERY" LABEL="RECOVERY" UUID="0B03-66CB" TYPE="vfat" PARTUUID="dd07b8ce-01"
/dev/sda5: UUID="ee58078a-d1d4-4f78-b93c-6bebce850066" TYPE="ext4" PARTUUID="dd07b8ce-05"
/dev/sda6: SEC_TYPE="msdos" LABEL_FATBOOT="boot" LABEL="boot" UUID="E0C7-8D4F" TYPE="vfat" PARTUUID="dd07b8ce-06"
/dev/sda7: LABEL="retropie" UUID="2c2ec3fa-953d-451d-b37d-269238f1eec9" TYPE="ext4" PARTUUID="dd07b8ce-07"
/dev/sda8: SEC_TYPE="msdos" LABEL_FATBOOT="prjboot" LABEL="prjboot" UUID="78F5-A3D5" TYPE="vfat" PARTUUID="dd07b8ce-08"
/dev/sda9: LABEL="prjroot" UUID="2d2564f9-2b36-4ed8-80d9-694d27997dde" TYPE="ext4" PARTUUID="dd07b8ce-09"
/dev/sda10: LABEL_FATBOOT="boot64" LABEL="boot64" UUID="8557-A844" TYPE="vfat" PARTUUID="dd07b8ce-0a"
/dev/sda11: LABEL="root64" UUID="1f71cb18-698f-499e-8541-6842ea51fb52" TYPE="ext4" PARTUUID="dd07b8ce-0b"
pi@raspberrypi:~ $ lsblk && blkid
NAME    MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda       8:0    0 465.8G  0 disk
├─sda1    8:1    0    63M  0 part
├─sda2    8:2    0     1K  0 part
├─sda5    8:5    0    32M  0 part /media/pi/ee58078a-d1d4-4f78-b93c-6bebce850066
├─sda6    8:6    0   256M  0 part /media/pi/boot
├─sda7    8:7    0 155.2G  0 part /media/pi/retropie
├─sda8    8:8    0   512M  0 part /media/pi/prjboot
├─sda9    8:9    0 153.4G  0 part /media/pi/prjroot
├─sda10   8:10   0   253M  0 part /boot
└─sda11   8:11   0   156G  0 part /
sdb       8:16   0   1.8T  0 disk
└─sdb1    8:17   0   1.8T  0 part
/dev/sda1: SEC_TYPE="msdos" LABEL_FATBOOT="RECOVERY" LABEL="RECOVERY" UUID="0B03-66CB" TYPE="vfat" PARTUUID="dd07b8ce-01"
/dev/sda5: UUID="ee58078a-d1d4-4f78-b93c-6bebce850066" TYPE="ext4" PARTUUID="dd07b8ce-05"
/dev/sda6: SEC_TYPE="msdos" LABEL_FATBOOT="boot" LABEL="boot" UUID="E0C7-8D4F" TYPE="vfat" PARTUUID="dd07b8ce-06"
/dev/sda7: LABEL="retropie" UUID="2c2ec3fa-953d-451d-b37d-269238f1eec9" TYPE="ext4" PARTUUID="dd07b8ce-07"
/dev/sda8: SEC_TYPE="msdos" LABEL_FATBOOT="prjboot" LABEL="prjboot" UUID="78F5-A3D5" TYPE="vfat" PARTUUID="dd07b8ce-08"
/dev/sda9: LABEL="prjroot" UUID="2d2564f9-2b36-4ed8-80d9-694d27997dde" TYPE="ext4" PARTUUID="dd07b8ce-09"
/dev/sda10: LABEL_FATBOOT="boot64" LABEL="boot64" UUID="8557-A844" TYPE="vfat" PARTUUID="dd07b8ce-0a"
/dev/sda11: LABEL="root64" UUID="1f71cb18-698f-499e-8541-6842ea51fb52" TYPE="ext4" PARTUUID="dd07b8ce-0b"
pi@raspberrypi:~ $ lsblk && blkid
NAME    MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda       8:0    0 465.8G  0 disk
├─sda1    8:1    0    63M  0 part
├─sda2    8:2    0     1K  0 part
├─sda5    8:5    0    32M  0 part /media/pi/ee58078a-d1d4-4f78-b93c-6bebce850066
├─sda6    8:6    0   256M  0 part /media/pi/boot
├─sda7    8:7    0 155.2G  0 part /media/pi/retropie
├─sda8    8:8    0   512M  0 part /media/pi/prjboot
├─sda9    8:9    0 153.4G  0 part /media/pi/prjroot
├─sda10   8:10   0   253M  0 part /boot
└─sda11   8:11   0   156G  0 part /
/dev/sda1: SEC_TYPE="msdos" LABEL_FATBOOT="RECOVERY" LABEL="RECOVERY" UUID="0B03-66CB" TYPE="vfat" PARTUUID="dd07b8ce-01"
/dev/sda5: UUID="ee58078a-d1d4-4f78-b93c-6bebce850066" TYPE="ext4" PARTUUID="dd07b8ce-05"
/dev/sda6: SEC_TYPE="msdos" LABEL_FATBOOT="boot" LABEL="boot" UUID="E0C7-8D4F" TYPE="vfat" PARTUUID="dd07b8ce-06"
/dev/sda7: LABEL="retropie" UUID="2c2ec3fa-953d-451d-b37d-269238f1eec9" TYPE="ext4" PARTUUID="dd07b8ce-07"
/dev/sda8: SEC_TYPE="msdos" LABEL_FATBOOT="prjboot" LABEL="prjboot" UUID="78F5-A3D5" TYPE="vfat" PARTUUID="dd07b8ce-08"
/dev/sda9: LABEL="prjroot" UUID="2d2564f9-2b36-4ed8-80d9-694d27997dde" TYPE="ext4" PARTUUID="dd07b8ce-09"
/dev/sda10: LABEL_FATBOOT="boot64" LABEL="boot64" UUID="8557-A844" TYPE="vfat" PARTUUID="dd07b8ce-0a"
/dev/sda11: LABEL="root64" UUID="1f71cb18-698f-499e-8541-6842ea51fb52" TYPE="ext4" PARTUUID="dd07b8ce-0b"

ultraviolet74, 2021-01-11:
> Hmmm, when you say "autopowered" do you mean they're powered from just the Pi's USB ports, or do the HDs have their own external PSU?

yes the first, they're powered from just the Pi's USB ports. One is inside the ArgonOne case sata m2 (but in fact powered by an USB3) and the other a Maxtor M3 powered from USB3

procount, 2021-01-11:
Sounds like a power supply problem to me. Do you notice the RPi's red LED flashing or flickering at all, or is it always a solid red?

ultraviolet74, 2021-01-12:
always solid red!

procount, 2021-01-12:
But it could still be a power problem. The Red LED means that the RPi is getting enough voltage. But there still may be insufficient power to drive your external SSDs/HDDs if they are powered from the RPi USB since this is current limited.
Your log above show an output for lsbk when you plug your 2nd (1.8TB) drive in, but there is no entry for blkid.
When you plug your 3rd drive in, the entries for both lsblk and blkid do not appear for drives 2 or 3.
Unless I am misunderstanding the logs, this suggests to me that there is not enough power to run both of them, and it is even marginal to run just the 2nd drive. It might be better to run them from an external power supply, or use a powered USB hub if you have one. What does @lurch think?

lurch, 2021-01-14:
I agree - the first lsblk shows a 465.8G `/dev/sda` (which must be your SSD), the second lsblk shows a 465.8G `/dev/sda` and a 1.8T `/dev/sdb` (which is obviously one of your HDDs), but the third lsblk _only_ shows `/dev/sda`! So this definitely sounds like the HDDs aren't getting enough power to work properly, and you should connect them via a powered USB hub.

ultraviolet74, 2021-01-14:
do you have some good power USB hub to advise me?

closed 2021-02-17 by procount