procount/pinn #459: Cannot Sign into insider while installing windows loT

Custom9409, 2021-01-07:
When I select the windows loT option and click install. It installs a few things and then prompts me with a sign in message. I try to click the button but it wont take me to the sign in page.

procount, 2021-01-07:
I thought Microsoft abandoned Windows IoT on the RPi a long time ago, especially the NOOBS version.
I suspect you will get the same problem if you try to install it with NOOBS.
If you really want to install it, you might be best to install it directly, but I've never done that.
Sorry, but I think that OS is dead and I won't be doing anything to fix the installation.

closed 2021-01-07 by procount

lurch, 2021-01-07:
@Custom9409 If you need any help with Windows IoT, I suspect you'll need to ask Microsoft...