procount/pinn #461: Allow pressing the "Boot" button via CEC

hupfdule, 2021-01-11:
I have not found a was to easily press the "Boot" button in the boot menu via CEC.

I can use "Channel +" and "Channel -" to move the selected entry in the list of OSs.
I can use the buttons "1" and "2" to press the "Up"/"Down" buttons with the same purpose.
But it seems that is not possible to directly press the "Boot" button without having to move the pointer over it. I would expect "4" to active the "Boot" button (as it is possible with "1" and "2" for "Up" and "Down"). Also "3" does not active the "Cancel" button.

Pressing the "Exit" button actually leaves the boot menu. From the list at [cec_keys.md](https://github.com/procount/pinn/blob/master/cec_keys.md) I assume this is the bound behaviour for Escape. But from the same document I would expect that "Play" would be bound to Enter and that would activate the "Boot" button. That is not the case. 

So tl;dr:

- Why to "3" and "4" do not activate the "Cancel" and "Boot" buttons?
- Is "Play" assumed to activate the boot button? As this is not working for me, is this a problem with my remote control?

procount, 2021-01-11:
By default, the Play button should be mapped to the Enter key, and hence to the boot function. The Exit key is mapped to Escape and hence to cancel. I will check it out later when I can get near the TV. If you view the debug log (see troubleshooting on the wiki) or use the log option from the fixup menu, you should be able to see CEC key presses in there. You can also create your own cec_keys.json file to map different keys if your remote is different.

hupfdule, 2021-01-11:
That's a bit strange. In the debug log I can see that the Play button doesn't produce any event in PINN. Other keys, also unmapped keys, do produce an event and get get listed in the debug log.
While booted into Kodi (from within PINN) the Play button works as expected and gets recognized.

It is especially strange, since the README states that the default mappings should work with Samsung TVs and mine is a Samsung.

I will try to create another mapping via cec_keys.json, but still wonder why the Play button doesn't work in PINN.

hupfdule, 2021-01-11:
Apart from that, is there a reason why "3" and "4" don't operate the corresponding buttons in the boot menu?

procount, 2021-01-11:
The 3 and 4 keys are not mapped by default there for some reason. Maybe I forgot to extend the cec file after I added them. You could always add your own mappings for those as well to see if that fixes it.

procount, 2021-01-12:
You can try dropping this cec_keys.json file on your PINN partition to see if it is any better.
(You'll have to rename it by removing the .txt extension, which I had to add to be able to post it here)
[cec_keys.json.txt](https://github.com/procount/pinn/files/5801296/cec_keys.json.txt)

redm123, 2021-02-14:
Play button works for me. But it's still somewhat hard to discover...
What I would intuitively expect: use my up/down direction keys to cycle through the list, and use the Ok button in the center of those to boot the selected entry. (left/right could then cycle the active button of the four at the top)
The cursor/pointer functionality wouldn't really be needed here, with the remove control use-case.

This could go along with the other options, Channel +/-, 1, 2, 3, 4, Play, Exit.

Btw.: Cancel should be labeled "Reboot", because that's what it does.

procount, 2021-02-15:
There is so much variation in CEC remotes. That's why I provide the option for you to create your own cec_keys.json file.

Yes I have wanted to change the cancel to reboot, but it's a standard button box so it requires more work than I imagined.

redm123, 2021-02-16:
But you use the up/down/left/right for cursor control... so this seems to be somewhat standardized? I was just suggesting to use those in the boot interface for direct navigation instead (**not** in the config interface). I think this is more useful than the cursor and easier to discover than Channel +/-, 1, 2, Play, Exit. After all the boot interface has just a single purpose: select and boot. And that should be as obvious and easy as possible, as it may be used by other people than me, that usually haven't read the PINN manual. My 2ct...

I'll figure out the JSON config meanwhile :)

procount, 2021-02-17:
If you figure out the JSON config file to work as you wish, you can always post it here for others to take advantage of your preferences.

hupfdule, 2021-02-26:
Sorry for my late response. 

A modified cec_keys.json does indeed work.

I am still a bit surpised that the Play button doesn't produce any event in PINN while it does in Kodi…

Nevertheless, thanks for your help.

closed 2021-02-26 by procount