procount/pinn #467: Adding a partition manager.

brunoga, 2021-01-13:
I understand this would depart from the way that NOOBS works, but this might be a good thing. The usual method of OS installation is PCs is to partition any storage the way you want and then install OSs to partitions you created yourself.

Currently the method used for RPi is terrible as it does not allow you to easily add or remove OSs without wiping data completely so I was wondering if it would not be possible to get rid of the current auto-partition method in favor of something like this:

- Create all PINN required partitions with their minimum possible size (or with some slack).
- Let the user select how they want to partition the remaining space to install OSs. To simplify, the sure would request a single size that could possibly be split between boot/os partitions (boot being as small as possible).
- When selecting the OSs to install, let the user pick where to install it (which partition created above).

This might even be fully optional and the default auto-partition method could still be used if the user did not care.

That would make installed multiple OSs (specially in SSDs) so much easier...

procount, 2021-01-14:
Have you seen http://pinn.mjh.nz ?

brunoga, 2021-01-18:
Actually, I have not. Thanks. This is better but still not ideal.

lurch, 2021-01-18:
@matthuisman I've just had a look at that website, and (although I dunno how easy / hard it would be to implement) I wonder if something like this might make the "resizing" step easier?
[image]

a) numbers to make it easier to match up the visual partitions with the name of the OS on that "partition"
b) a dark-shaded region within each slider to show the minimum size required by each OS (so that you have an idea how much "slack" there is for each OS)
c) perhaps the title could make it more obvious what to do? e.g. "Set your system sizes (by moving the sliders)"

closed 2021-02-17 by procount