procount/pinn #468: Unable to replace os after resizing partitions.

brunoga, 2021-01-14:
As a workaround for the auto-resizing of partitions in PINN, I installed some OSs, project partitions and a data partition. After that, I used gparted to resize the partitions to more useful sizes and at the same time increase the size of the data partition.

This worked ok and I was able to boot all OSs that i tried but them I tried to use the replace function in some partition to replace the installed os (or data partition) and it fails saying there is not enough space in the partition. Even if that specific partition was not resized, trying to install the same os that is already there and working results in the same error message.

Any ideas why this might be happening?

While we are at it, I thought about a different approach for the partitioning: What if one could add an arbitrarily sized data partition (instead of one with 500mb only)? The idea would be to allocate the space for the data partition and them use the remaining space for all the installed OSs.

FWIIW, I am actually running the rpi4 with a 1TB SSD drive so the auto-partitioning it does is far from ideal but if I could do what I mentioned above with the data partition, it would actually be ok.

Thanks.

procount, 2021-01-14:
Without any concrete data, I can only guess what your problem was. Which OS was it? what was your partition layout? How big did you resize the partitions to, fdisk output etc.
Did you account for the approx 400MB slack that is included in the root partition of most OSes that PINN installs?

The 500MB data partition is just an example. A Data partition is an OS without any files - so just the meta files. You can easily create another one (or several) of different sizes. Store them in the /os partition of PINN's partition, or on a UBS stick and install them from there.

If you use Matt's webpage (http://pinn.mjh.nz) then you might be able to specify the size of the data partition that way.

brunoga, 2021-01-18:
So I was more puzzled by not being able to install the OS on the same partition it was installed after resizing *OTHER* partitions, but I figured out the issue.

Ubuntu will try to grow its partition if it finds extra space just after it, but it is doping it in a way that is corrupting the partition table. I have found a way to disable the resizing and now I managed to configure the partitions the way I wanted them. Thanks and sorry for the noise.

closed 2021-01-18 by brunoga