procount/pinn #470: Lineage tv Backup restore failed

venkatmakam, 2021-01-24:
Hi i tried to restore the lineage tv backup in rpi4 8gb,and it failed with error "system.tar is not found". When i cheked the backup folder it has the file "system.img. gz" not the "system.tar.gz". The following is the content of backup folder. 

[image]

procount, 2021-01-24:
Sorry about that.
Lineage-17 was a difficult OS to convert requiring some changes to the procedure and something in the backup needed changing.
Can you post the contents of your os.json and partitions.json here, please?
I can then update PINN and provide instructions for you to fix this. 

Just for my info, do you remember which version of PINN you backed it up with and which version of PINN you are trying to restore it with?

venkatmakam, 2021-01-24:
Hi @procount, please find the files from the backup folder

**os.json:**
```
{
 "description" : "Lineage OS (KonstaKANG) 17.1 Analog TV build for RPi4",
 "download_size" : 3438864947,
 "feature_level" : 0,
 "group" : "Media",
 "kernel" : "5.4.58",
 "name" : "lineage17-rpi4-atv#20210121-162146",
 "password" : "",
 "release_date" : "2020-11-21",
 "sha512sum" : "9456253a84b8ff017794b01c2c40470ffa5ceabe2f4e6172fe802007ea763f9cb5d3da5bdb7cd856d62ced2e3815b8af6b2f2a7b033c26f9bebed9d18265fbf6",
 "supported_models" : [
"Pi 4"
 ],
 "supports_backup" : "true",
 "supports_sda_boot" : true,
 "supports_sda_root" : true,
 "supports_usb_boot" : false,
 "supports_usb_root" : false,
 "url" : "https://konstakang.com/devices/rpi4/",
 "username" : "",
 "version" : "17.1"
}

```
**partition.json**
```
{
 "partitions" : [
  {
   "filesystem_type" : "FAT",
   "label" : "boot",
   "partition_size_nominal" : 126,
   "sha512sum" : "5d66a7333e7ff7138ec5068f8f89ad86c4cf82033a17d22f527d8c47bc46b17646419f567f4c8213511f2d52b6615ec35cbf6cae1804213f81b06fc395250f1d",
   "uncompressed_tarball_size" : 26,
   "want_maximised" : false
  },
  {
   "filesystem_type" : "ext4",
   "label" : "system",
   "mkfs_options" : "-O ^huge_file",
   "partition_size_nominal" : 1445,
   "sha512sum" : "688f6c65bc7b5a959b7277763043ef4f9945892c3c61c912b8c9b5a332a89d853cff0e732e01afd66ec82e7ea340062dc98e323bf6d89c4f38483ed9f3918a76",
   "uncompressed_tarball_size" : 945,
   "want_maximised" : false
  },
  {
   "filesystem_type" : "ext4",
   "label" : "vendor",
   "mkfs_options" : "-O ^huge_file",
   "partition_size_nominal" : 549,
   "sha512sum" : "c28cf1ae6d74aef9a07c65bc257b7db82f33e1bcf5ce7e229e73aaadd44b4a9503fe37aa02db066edd4e2c17658dc7ad533900a73814eddf1dbdf19459c2810b",
   "uncompressed_tarball_size" : 49,
   "want_maximised" : false
  },
  {
   "filesystem_type" : "ext4",
   "label" : "userdata",
   "mkfs_options" : "-O ^huge_file",
   "partition_size_nominal" : 3952,
   "sha512sum" : "c05d1154b3adcbfd08a583e912fcd0efb7a0a4df9a8bb2b1961612e166175060ceffe72cd559e64cf28c15daa7d41c0e7a2f6d09ec71acb027ed4741ee150453",
   "uncompressed_tarball_size" : 3452,
   "want_maximised" : true
  }
 ]
}
```
* I used V3.5.4 version PINN
* I installed Lineage OS after 11th. Also, i installed it from external drive by downloading them from the sourceforge.

procount, 2021-01-24:
Thanks. Interesting.
Did you download it from Sourceforge after 11th Jan also?

venkatmakam, 2021-01-25:
Hi @procount, I did not remember when I downloaded , but I suspect it was before jan 11th...

commit 5e8af15b (2018-05-01): automake: port to Perl 5.22 and later (#484)

Fixes #470