procount/pinn #471: Enhancement Request:  Allow the user to specify a custom filename for a backup

jharris1993, 2021-01-27:
Greetings!

Enhancement request:
Allow the user to create backups with customized filenames instead of just the original name of the installation.

Justification:
As a part of my research into robotics using the Raspberry Pi, I may customize the image in particular ways, and wish to save a particular customization before changing it or doing something potentially stupid.

I would like to be able to - in some way or other - identify what version or customization this particular backup contains.

Did I install Blockly?  Did I enable EduBlocks?  Does this image have an update, (or downgrade), to Jupyter Labs or the base Python install?  As it is, the backup is distinguished only by the name of the installed O/S and a string of numbers and characters which makes identifying the differences between backup "A" and backup "B" impossible without careful notes - and since the full name isn't visible until after the backup is created, even that isn't a trivial task.

One possible solution would be to create a top-level metadata file that would contain all the necessary metadata to allow the restore operation to continue - the name of the OS to be restored, the revision and date codes and characters, and anything else that the name of the enclosing directory provides.

On other systems, programs like Acronis True Image allow the backup to both have a custom filename and additional annotations, (comments) created as a part of the backup process.  This kind of feature, especially backup comments, would be greatly appreciated - especially by the community of users who wish to use the Raspberry Pi - and PINN - for research and development.

Respectfully submitted.
Jim "JR"

procount, 2021-01-27:
When you are prompted to backup an OS, it defaults to adding a date and time to the OS name, which is useful for chronology of backups. You can modify this for your needs, although I'm not sure it is displayed in the OS selection dialog. Additionally, there is the description field, where you can put additional information that you suggest above. Finally, you can provide an Alias for that OS backup to give it a distinctive name, instead of everything being called Raspbian.

Are those sufficient, or is there something still missing?

jharris1993, 2021-01-27:
Absolutely true, all of it.  Early cars steered with a tiller and had a crank to start them.  However an electric starter, later on power steering and automatic transmission have made automobiles much more user friendly.

>  . . .you can provide an Alias for that OS backup to give it a distinctive name

Perhaps true, but not very obvious.  The most obvious place is in the filename and/or description.  Maybe I can't see the forest for the trees, but this did not suggest itself to me - and I don't know how much of the alias or description field is visible when viewing the backup for restore.  I suspect that it's completely invisible when looking at the backup itself within the O/S directory.

>  When you are prompted to backup an OS, it defaults to adding a date and time to the OS name, which is useful for chronology of backups. You can modify this for your needs, although I'm not sure it is displayed in the OS selection dialog.

One thing you mentioned while we were playing with backups was that the backup name and date contains specially formatted characters and such that trigger the installation scripts to add or skip steps.  Because of this, I really hesitate to mess with any of that data, even though it's clearly editable.  And like you, I don't know how much of this is visible during restore.

Assuming the date and time are immutable, (and are messed with at the user's peril), things like the description are not clearly visible by inspection while looking at the folder containing backups.  Asking the user to start PINN just to decide which backup to put on a flash drive, (or copy to cloud storage), is a bit excessive from a usability standpoint.

procount, 2021-01-27:
The filename format is:
"Name - flavour=alias#date@partition"
(Not sure the partition bit is used)
You can mess with the date as long as you don't include the delimiters. It's just for user information. I don't think PINN processes it.
So you can see most information from the backup drive. 
The description is stored in os.json and is displayed for all OS within PINN.

closed 2021-02-17 by procount