procount/pinn #472: Enhancement Request:  Allow the user to specify a custom name for an original installation/

jharris1993, 2021-01-27:
Ref:  My previous issue [471](https://github.com/procount/pinn/issues/471).

Enhancement Request:
In a manner similar to the request specified above, I would like to propose an enhancement to allow the user to specify a custom O/S or project name to the downloaded operating system during install.

Justification:
Especially in a research-and-development environment, (which is a use-case particularly suited for PINN), a particular operating system may need to be installed and modified in distinctly different ways that are absolutely incompatible.

Example:
- Raspbian, modified for the ultra-noob/younger crowd with basic blockly programming and simple programming based games installed.  In this version, most of the more advanced features, (like sudo) would be hidden behind an administrative "firewall" needing a special login/password to access.
- The same version of Raspbian, modified for a more advanced class of students with Python, Java, JavaScript, Jupyter Labs, and other advanced programming tools and features.  More of the base installation's features would be available, but (possibly) sudo would be restricted.
- The same version of Raspbian, configured for the very advanced programmer/hobbyist which gives the fullest range of tools and authority to the user so that they can do whatever they want.

Development on these versions might be done in parallel - to make sure similar functions work similarly in all versions of the software.

Being able to give a particular installation a distinctive name at install time - which would be preserved during any backup process, would be invaluable.

jharris1993, 2021-01-27:
>  I had seen PINN as a mostly user-centric tool, but this also makes it more of an OS developer tool.

Users, like people, come in a variety of colors and with a broad range of needs and desires.

The most common, and simplest, use-case doesn't require PINN at all.  Flash an image and you're done!  The SD card copier takes care of your backups, and other utilities handle the rest.

By definition, PINN is an advanced tool designed for the more advanced use-cases.  Though it's an advanced tool, it should not require a degree in quantum mechanics to operate it.  Compare the command line tool parted with the GUI version GParted.

>  I suppose the question is, whether a customised version of an OS like Raspbian, still needs to be known as Raspbian, or is it a different OS?
>
>  If you are creating your own OS, you can of course manually edit all the metafiles as you wish. So does PINN need to include additional functionality in its GUI to do this as well?

First of all, this has nothing to do with customized "spins", (PINN installers), of any O/S's.  Customized spins are a totally different beast altogether.

What I am talking about is a stock download/install of a pre-prepared distribution like PINN's Raspbian 32-bit spin.

My original example had the user downloading four or five _**identical copies**_ of Raspbian, each intended for a different usage scenario.  These are standard installations that will be customized, perhaps heavily, _in situ_ by the user.

The user would perhaps download a single copy of Raspbian and store it locally to save download time.

He would then install the exact same package, multiple times, for multiple uses. Obviously it is a massive convenience if the user can re-name the installation at the time it's installed, (ideally carrying over to the partition names).

He does each installation once and once only, naming it as appropriate at the most appropriate time - install time - and then moving on to the next.  The user experience would be tremendously sub-optimal if he had to install, re-enter PINN, muck around with metadata or aliases, before starting the next operation.  A 45 minute operation could easily take three or more hours.  Go ahead!  Ask me how I know! ;)

Here's a potential real-world example:
An amateur radio operator, ("ham" radio), may have a Raspberry Pi that he uses for his station operation.  It might be convenient, (or even necessary), to have different operating systems installed to control different usage scenarios.
- One installation that's a plain vanilla Raspbian install.
- One installation for near-earth satellite communication.
- One installation, with its own separate call logs for participating in ham radio contests.
- Another installation for experimental microwave single-sideband video.
- And so on.

The idea here is that we, none of us, can imagine the potential uses for PINN in the hands of a variety of usage scenarios, so it needs to be flexible and easy to use.

procount, 2021-01-27:
Aliases were a request from a previous user for this same problem.