procount/pinn #474: Rename the "cancel" button on the boot selection page and add a "reboot" button to the main page.

jharris1993, 2021-01-27:
Greetings!

I have seen many requests on your thread asking for a "reboot" feature.  This feature already exists on the boot selection menu disguised as the "Cancel" button - which does a hard restart.

There are times when - having done something within the sections of the main dialog - the user may wish to reboot instead of continuing.  Right now the only way to do that if the boot menu isn't available, is to use the console.  The user has to login, provide a password, and only then can he request a reboot.

The first case is simple to implement:  Just change the name of the "Cancel" button to "Reboot" which is more closely aligned with its actual function.

The second case might not be that much more difficult:  Copy the button, with a link to the cancel/reboot functionality, and place it somewhere on the main dialog where it is available from every page.  Perhaps the lower right-hand corner?

Respectfully submitted:
Jim "JR"

procount, 2021-01-27:
Ctrl-Alt-f2 followed by Ctrl-Alt-Del is another quick way to reboot.
Case 1 can easily be done.
There are additional requests to go straight to the boot selection dialog when clicking shift at startup, so I have been experimenting in this area, but it means some major architectural changes.

jharris1993, 2021-01-27:
My idea is to create a direct functional link, not a handful of keystrokes.

Question:  Once the cancel button is renamed, isn't it still a widget, (or something like that - a control with an associated method).  Once you have this object - the cancel/reboot button - can't you take the button and put it wherever you want, in as many places as you want?

Of course, there are real-estate concerns - you cant take something the size of the "backup" tile and place it in a narrow toolbar.  But wouldn't a "cancel/reboot" button be small enough to fit there?

Or maybe I'm missing something?

procount, 2021-02-17:
The Cancel button is a standard radio button, so it would need changing to a more custom button just to change its name. Harder than I first thought.

You can actually press ESC from the Archive and Maintenance menus to go straight to the Boot Selection Dialog, just like you can from the Main Menu. Except from the Main Menu, you must have some OSes installed to enable this functionality.