procount/pinn #475: CM4 Module USB

NumberOneGit, 2021-01-30:
To enable USB on the IO Board for the CM4 the following has to be added to config.txt:
 "dtoverlay=dwc2,dr_mode=host"
This works in both RP OS and Ubuntu, but adding it to the config.txt in PINN has no effect and I am unable to use a keyboard or WiFi dongle.  I'm new to PINN, so hopefully I'm not missing something obvious.

procount, 2021-01-30:
PINN does not include the dwc2 USB kernel driver, which is why your config.txt setting has no effect.
I suspect the same is true of NOOBS too.

lurch, 2021-01-30:
See also https://github.com/raspberrypi/noobs/issues/620
(I like to keep related issues interlinked)

procount, 2021-01-30:
I'll try and include it in the next version for you to test it out.

NumberOneGit, 2021-01-30:
Amazing!  That is greatly appreciated...  the Compute Module 4's with WiFi Modules are currently on an ambiguous backorder status of anywhere from two to six months.  I'm not sure how many users will be using PINN on CM's but for those of us that are it will be a huge help.  I had intended to run everything from SSD, but in the meantime am trying to get it up and running via ethernet cable and VPN.  Thanks again for your great work.

lurch, 2021-01-30:
Huh? I'm _fairly_ sure that PINN doesn't include any VPN support...?

NumberOneGit, 2021-01-30:
Sorry, typo..  VNC

procount, 2021-01-30:
I don't have a CM4, so you'll have to be my guinea pig to test it.

NumberOneGit, 2021-01-30:
Absolutely.  Happy to do so!

procount, 2021-02-01:
I've uploaded a new version
https://sourceforge.net/projects/pinn/files/testing/pinn-354g.zip/download
Please check if it now works.

NumberOneGit, 2021-02-01:
This works perfectly.  Keyboard and mouse up and running, WiFi dongle not working but will seek list of compatible ones.  Thank you for getting this up and running so quickly!

procount, 2021-02-01:
Cool! Glad it now works.
Which wifi dongle chip are you using? I did add some new versions, but I had to remove some with the upgrade to the kernel until I can find new compatible versions.

NumberOneGit, 2021-02-01:
A bit off topic, but is this list specifically showing OS's that are recommended for the CM4 or for the Pi4B...  I was expecting to see Lineage OS and RPOS 64-bit.  I know I can use showall command but was just generally wondering if there was a potential compatibility issue.  Will send that WiFi model over in a moment, it's BlueShadow and I believe RTL8821CU but will double check that.. is there another device ID you need?  I will need to re-install some version of Linux to get that but working on that now.

procount, 2021-02-01:
I don't have a CM4, so I can't test them. If you can set showall and test any OSes that you think should work (i,e, those for the Pi4) and let me know which one's work, I'll gladly update them so they are available for the CM4 too.
RTL8821CU is sufficient. I haven't even tried to include that one.

NumberOneGit, 2021-02-01:
It is a little buggy where I have to unplug/re-plug the device in to get it to work, but once it works it works.  Just making a note for anyone else that may have this issue.  I'm not using a powered USB hub at the moment, so it's possible that that would resolve the issue.  I'll have a better idea through regular use and update if I figure out a pattern.

NumberOneGit, 2021-02-01:
No issues when using powered USB hub.

NumberOneGit, 2021-02-02:
This is a separate issue, so I'm not sure if you would rather I start a new thread but I'm trying to install version of other OS's that are more up to date, such as Lineage 18.1 and a nightly build of LibreElec..  the problem I'm facing is that the "reserve" command seems to possibly be SD Card specific..  I get an infinite "Waiting for SD card (settings partition)" loop.

procount, 2021-02-02:
How are you installing these newer OSes?
Are you using the "reserve=X" or "reserve=+X" formats?
Why are you storing your images on the SD card for installation, instead of putting them on a USB stick? The latter will save more space on your SD card to be used by your OS installations.

NumberOneGit, 2021-02-02:
My theory was that PINN was defaulting to looking for an SD card because that's the default for a normal Pi.  However, the error may just be related to trying to set "reserve=" to too high a value.  I'll try the USB method as soon as I get back home to try that out on.  That's the most sensible route and I realized that last night after reading through the README a bit more.  I assume it will work as expected.

procount, 2021-02-02:
OK, but how are you trying to planning to "install versions of other OS's that are more up to date, such as Lineage 18.1"?
Have you converted them already to .tar.xz format for use by PINN? PINN cannot install .img files.

NumberOneGit, 2021-02-03:
For LE I just put the NOOBS version of the nightly build in the OS folder and that worked with no problem, the new version fixes the USB problem.  I was going to install Lineage 18 over the Lineage 17 partitions and edit related .json files, as the sizes seemed pretty similar.  I'm still trying out different configs and that's a bit of a convoluted routine so I'm just skipping it for now.

procount, 2021-02-03:
OK. For both of them (especially lineage) you would need to run the partition_setup.sh script from the Fixup menu after overwriting the partitions.

procount, 2021-02-16:
DWC2 included in V3.5.5 Just released.

closed 2021-02-16 by procount