procount/pinn #476: both ubuntus at the same time problem (partition setup)

Ae3NerdGod, 2021-02-02:
one of the ubuntus properly changed the root in commandline.txt, i believe it was vanilla ubuntu (not mate) but i could be wrong.

-fstab on both was not modified correctly (or at all?) and used default labels (LABEL=writable)
-one commandline.txt did not change at all either
-both boot options would boot vanilla ubuntu 

simple manual edits of proper partuuid in all 4 files solved issues

procount, 2021-02-02:
Until you do an update and then it will probably fail again.
I don't know why Ubuntu insist on using labels to reference their partitions, because it makes installing more than one copy on a machine very difficult, requiring manual intervention, as you found out.

Ae3NerdGod, 2021-02-02:
Are you saying an update to like the firmware or something reverts commandline.txt?

I remember solving something like this problem on 18.04 by making an apt postinstall (with PINN partition_setup.sh) with the correct partition name/uuid at time of creation.

Ae3NerdGod, 2021-02-02:
It was to avoid uboot entirely, same concept.

This was the snippet from my partition-setup.sh

```
#keep avoiding uboot

#echo "Post-Invoke {\"/usr/bin/dd if=/boot/firmware/vmlinuz bs=1 | zcat >/boot/firmware/vmlinux\";}; " >> /etc/apt/apt.conf.d/99pinn-avoid-uboot
#echo "Post-Invoke {\"/usr/bin/sed /boot/firmware/nobtcmd.txt -i -e \"s|root=[^ ]*|root=${id2}|\";}; " >> /etc/apt/apt.conf.d/99pinn-avoid-uboot
#echo "Post-Invoke {\"/usr/bin/sed /boot/firmware/btcmd.txt -i -e \"s|root=[^ ]*|root=${id2}|\";}; " >> /etc/apt/apt.conf.d/99pinn-avoid-uboot
```

procount, 2021-02-02:
> Are you saying an update to like the firmware or something reverts commandline.txt?

Yes, I think an `apt update` might just do that if it updates the firmware. I don't think there is that much difference between 18.04 and 20.04 in that respect, but I've not tried it in a while.

Ae3NerdGod, 2021-02-02:
I'm going to assume you meant 20.10, because that's what's on PINN, but I will test a solution and report back if it works. 

Also maybe opening an issue up in launchpad(?) Would get them to check command line for edits before it replaces it like ubuntu does with other configs I've seen? Although they might just be set in their ways :/

procount, 2021-02-02:
I think it's RPi specific, and they probably did it to boot from SD or USB, but partuuid would have been better. I even mentioned it to Martin Wimpress, but it didn't get changed.

Ae3NerdGod, 2021-02-02:
Still seems like they could at least make apt aware that that's a configs file that shouldn't be replaced. 

Maybe I'll make my own bug report, outside of the context of using ubuntu with PINN, and bring up general issues with the method they're using, because replacing boot config files blindly on an OS that's supposedly supposed to be server-stable (at least in public opinion/youtube tuts) just sounds like an ape or dolphin is coding it or they've overlooked the issue for convenience to their own testing.

Ae3NerdGod, 2021-02-02:
Now that I mention it it could be a limitation of the 'diversion' process the firmware goes through

procount, 2021-02-02:
Just check it is still true before reporting, don't rely on my word for it, as I haven't tried it for a while.

Ae3NerdGod, 2021-02-02:
Will do. I try to confirm info more often than not anyway. Rule number 3; never trust no-bo-dy.

Ae3NerdGod, 2021-02-02:
Would there be a way to make PINN itself perform an action to an OS upon every boot of said os? This could allow for a partition-setup-verify.sh or the like?

Just brainstorming btw.