procount/pinn #477: having an issue with cm3 and the thing its attached to (handheld built in joysticks)

Ae3NerdGod, 2021-02-02:
first of all, none of the buttons come up as the right buttons, and some dont get detected at all. but enough buttons were workingfor me to ignore that, and just remap buttons to the right actions, which works fine when im actually IN recovery

the joystick just completely doesnt work on the boot menu screen, or "pres shift to enter recovery" screen. 

am i missing an enable joystick option? i did a semi thorough search for that in docs, and didnt see anything specifically saying the joystick needs to be enabled manually for boot selection or anything like that.

if im not missing something, what can i provide to help troubleshoot? im pretty sure the joystick identifies itself as a "sparkfun yada somethingoranother " last time i checked, but i dont wanna fill this issue with more needless info than need be, so tell me what info you need and ill do my best to provide it

procount, 2021-02-02:
Try this beta version ->https://sourceforge.net/projects/pinn/files/testing/pinn-354g.zip/download
I updated the joystick code a lot.
The format is pretty similar, but you can create your own Inputs section which also adds deadzones. See #376 for more details.

Ae3NerdGod, 2021-02-02:
I will look into this, thanks.

closed 2021-02-17 by procount