procount/pinn #478: WLAN missing, Ethernet unmanaged in Kali Linux (64-bit)

Ben1701A, 2021-02-02:
I managed to find the cause of the unmanaged brcmfmac device, but I saw no trace of the built-in WLAN.
I tried a USB stick I had lying around, and dmesg showed me weirdness:
```[ 1160.825419] usb 1-1.1: new high-speed USB device number 6 using xhci_hcd
[ 1160.928577] usb 1-1.1: New USB device found, idVendor=13b1, idProduct=0039, bcdDevice= 0.01
[ 1160.928609] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1160.928626] usb 1-1.1: Product: Linksys AE1200
[ 1160.928642] usb 1-1.1: Manufacturer: Cisco
[ 1160.928657] usb 1-1.1: SerialNumber: 000000000001
[ 1160.943254] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43236b for chip BCM43235/3                                                        
[ 1160.943764] usb 1-1.1: Direct firmware load for brcm/brcmfmac43236b.bin failed with error -2

Is it trying to load the ethernet firmware for the USB stick? If so, sounds like that could be the cause of the missing WLAN adapter.

procount, 2021-02-02:
Did you mean to post the above in the PINN forum? Or is it more a Kali question?

Ben1701A, 2021-02-02:
Hmm. Maybe I should drop this in the kali-pi repo...

Ben1701A, 2021-02-02:
> Did you mean to post the above in the PINN issues? Or is it more a Kali question?

I noticed other people were writing issues about the included OSes in PINN, so I thought I would try here.

procount, 2021-02-02:
If you are trying to use the wifi dongle in Kali, then the Kali forums should get you a better response as I don't know much about it.
I only know enough to get it converted for PINN and installed.

Ben1701A, 2021-02-02:
It's also the internal WLAN module. It's just missing, like the dongle.

closed 2021-02-17 by procount