procount/pinn #479: How to migrate PINN from SD to USB

Marxon, 2021-02-06:
Hi procount and all others,

how do i migrate PINN (with multiple OS installed) from SD card 
to a USB device in order to boot from it? 
Currently i am trying to backup the installed OSes via PINN's backup function to a second USB device, create a new clean PINN installation on the main USB device and finally restore the backups. 
If PINN automatically adjust the PARTUUID and fstab during this proccess this maybe works.
If not how i do it the right way? 

Thanks for any advice
Marxon

procount, 2021-02-06:
There are several ways, but the method you propose is probably the best one.
Most OSes will boot fine from USB but some won't, so it would be wise to check this first.
Is your USB drive bigger than your SD card? When you restore your backups, PINN will divide any additional space between the OSes. If you want to use this space for other OSes, you need to install them at the same time as your backups, or reserve some space by installing some projectspaces which can be replaced later. PINN cannot ADD OSes at a later date because it will use up all of the available drive space when it partitions it.
Restoring an OS from backup is almost like installing from scratch, so PINN will adjust hte fstab and cmdline.txt appropriately.

There is also the clone SD feature, which can be used to clone a complete SD card to another one. The last partition can (optionally) be resized to fill the remaining drive space. I'm not sure how this will work if you intend to boot the clone from IUSB. I suspect you may have to re-ruin the partition_setup.sh script on each cloned OS from the fixup menu to adjust the fstab and cmdline. It's not something I'd envisaged before.

Some recent issues have surfaced regarding backups. There are workarounds, so if some of your OSes fail to backup, fail to restore, or you can't mount USB devices properly within the restored OS, just let me know and I'll point you to the fixes.

procount, 2021-02-06:
If your USB drive is large, don't try formatting it all as FAT32 just to install PINN. Use the Etcher image version of PINN and install it like you would any normal OS using Balena Etcher or the Raspberry Pi Imaging Utility. It will be a lot easier for you!

Marxon, 2021-02-06:
Thanks for your information procount!
PINN is really great. So i don't have to deal with a all this settings and config stuff.
Yes the target USB drive is bigger than the SD. I'm going to install one or two project spaces to reserve some space and will use the Etcher image like you said.
Feedback follows later this evening.

Marxon, 2021-02-07:
Hi again
Unfortunately my backups can't be installed succsesfully. 
Using your debug log capture script also gives me thiss error:
`cat: can't open '/settings/os/ProjectSpace_-_1/partition_setup.sh': no such file or directory`.
Nevertheless it creates this log:
[log.txt](https://github.com/procount/pinn/files/5938446/log.txt)

procount, 2021-02-07:
The error from the debug log can be ignored.

>Unfortunately my backups can't be installed successfully.

Thank for you including the log.  From that it looks like there was a problem with the LibreElec storage partition only, but it is not apparent from the log what the cause was. The others should have installed ok.
Do you remember any error messages on the screen when it failed?

Marxon, 2021-02-07:
There was only a message like "Installation finished but there was an error"

Marxon, 2021-02-12:
I guess i finally did it.
First I made a backup of LibreElec via the internal backup feature, installed my RaspberryOS and Retropie PINN backups together with a fresh install of LibreElec and then restored the LibreElec backup. 
Now everything seem to be working.

closed 2021-02-17 by procount