procount/pinn #480: CrowPi2 GPIO config

Blind-Freddy, 2021-02-07:
Hi. What a great system, Thank you.

I am using Pinn to load TwisterOS or CrowPi2 OS (Elecrow) on a CrowPi2 laptop.
To initialise hardware, CrowPi2 needs a GPIO pin to be held low at boot. This is achieved with the following line in config.txt when CrowPi2 boots.

gpio=0=op,dl

However, when Pinn boots, there is no config.txt file to achieve this. Is there a way to get this into one of the config files for Pinn OS?
The work around is to use a jumper wire, but this is quite inconvenient.

Any advice would be appreciated.  Thank you

lurch, 2021-02-07:
I believe you can simply create config.txt on the PINN partition (the first FAT-formatted partition on the SD card) and the firmware will then read it, the same as it does for any other OS?

Blind-Freddy, 2021-02-07:
Thank you. Works a treat.
There is an existing config.txt which I understood from the documentation, was ignored.
I suppose I should have tried it before reaching out.

I will close the issue. Thanks for the quick response.

Blind-Freddy, 2021-02-07:
Thank you Andrew. As per issue, it works perfectly.

Very rapid response. This is a wonderful project for my purposes.

Thanks for being there.  Garry

closed 2021-02-07 by Blind-Freddy