procount/pinn #481: Windows 10 on ARM unable to boot in PINN

MysticWhiteDragon, 2021-02-07:
I tried to make a multiboot image on an SSD using (Raspberry Pi OS, Konstakang Android, and Windows 10 on ARM).  I have successfully managed to boot Raspberry Pi OS and android without any issues.  However, I can only get into the UEFI Boot Partition for Windows.

procount, 2021-02-07:
Microsoft haven't supported Windows IoT for years.

MysticWhiteDragon, 2021-02-07:
I'm not referring to Windows IOT, but Windows on ARM.
https://www.worproject.ml/

procount, 2021-02-07:
Has someone produced a noobs/PINN image for this installation then?

procount, 2021-02-07:
A quick read of that site suggests WoR requires GPT, but PINN only supports MBR (not GPT), so it would not be able to manage the installation. You might manage to fork-and-spoon it in manually with some sort of hybrid GPT/MBR partition format, but I've no idea how you would do that.

Itai-Nelken, 2021-02-07:
WoR (Windows on Raspberry) won't work with multiboot (mbr or gpt) because it uses a uefi or something.
I'm not sure how to explain why it doesn't work, but I can say 100% that it doesn't work.

Itai-Nelken, 2021-02-07:
>A quick read of that site suggests WoR requires GPT, but PINN only supports MBR (not GPT),

actually it does support MBR, and its recommended not to use GPT with it because its unstable.

MysticWhiteDragon, 2021-02-07:
> Has someone produced a noobs/PINN image for this installation then?

Not that I'm aware of.  I used one of the OS images and just modified the second drive to be NTFS instead of ext4.  It managed to successfully copy both the root and system partitions to the drive and boot into the EFI.

I was planning on putting this on an m.2 SATA drive inside of an enclosure for a portable system that I'm making.  I might look at trying out MBR on WoR.  And I'll also look to see if there is a way to do a GPT/MBR hybrid using gParted or something similar.

procount, 2021-02-07:
I hope you appreciate that just copying the partitions will not work. 
You have moved the partitions from their default location on the drive (typically /dev/mmcblk0p1 and /dev/mmcblk0p2, or /dev/sda1 & /dev/sda2 if on USB)  to another partition. PINN knows where to find the first boot partition (or UEFI), but that partition needs to be able to find the root partition (or however WoR is partitioned - I have no idea).
On Linux you typically have to fix up cmdline.txt and /etc/fstab to tell it where the new locations of its partitions are. This is the purpose of the partition_setup.sh script file for each OS.
Android is a lot more complex.
Most 64-bit OSes seem to require UBoot , so that the subsequent boot loaders can be more standardised, but UBoot only works on the first partition, so I have to strip it out.
WoR - I have no idea, especially as it requires UEFI. @Itai-Nelken suggests it is not possible. If it is possible, it will need more than just copying the partitions across, I'm afraid. He also suggest MBR is supported, so stick with that if you can.

lurch, 2021-02-09:
In summary: if you want to use "Windows 10 on ARM" with PINN, you'll have to ask the "Windows 10 on ARM" developers to specifically add support for PINN to their OS :wink:  (and with no disrespect to PINN, I suspect they have "more important" features to work on than supporting installation with PINN)

mop999x, 2021-02-09:
i wonder if its possible to make a partition on win 10 and change the boot order from UEFI and install maybe a iso/img backup of pinn and other OS  on that partition if possible it could kinda have dule boot at least that's what i was thinking about since i tried to make a dule boot tablet with win/linux os's and started hitting limitations. Also based on the chat i had on with one of the people mods or devs not sure who developed WOR they did mention UEFI and potential instability

lurch, 2021-02-09:
> install maybe a iso/img backup of pinn and other OS on that partition if possible

Not gonna work, easier (and cheaper in the long-run) to just use separate SD cards.

> it could kinda have dule boot at least that's what i was thinking about

For anyone else who thinks "this should be easy", I suggest reading https://github.com/raspberrypi/noobs/wiki/Standalone-partitioning-explained and https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained which should help you understand @procount 's [comment](https://github.com/procount/pinn/issues/481#issuecomment-774762427).

Since SD cards (that a Raspberry Pi boots from) have so much less capacity than Hard Disk drives (that PCs boot from), "dual booting" is much less common on a Raspberry Pi than it is on a PC. Combined with the fact that it's much easier to swap SD cards than it is to swap HD drives! :laughing: 
So developers of Raspberry Pi OSes tend to assume that they're the **only** OS installed onto a card, and support for NOOBS or PINN seems to be somewhat of a rarity. Whereas on a normal PC "dual boot" support in an OS is much more common.

Ae3NerdGod, 2021-02-11:
When I was experimenting with WOR, I was able to make a dual boot using PINN to boot uefi firmware as a boot option, which then I guess 'chainloaded' windows from an entirely separate ssd. It worked as well as WOR works. So barely.

Ae3NerdGod, 2021-02-11:
"Has someone produced a noobs/PINN image for this installation"

That might get one of you contacted by microsofts legal department. WOR gets away with it by making you build the image to install on your computer, you just download a script. But installed windows-- not the installer-- as a PINN image maybe you could get away with?

Problems when I tried it
-cant tar ntfs
-windows sucks
-uefi was still wonkey and didn't support esp partition yet (does now)

Might look into it again as a proof of concept, but too much stuff doesn't work for it to be a priority

Ae3NerdGod, 2021-02-11:
In theory, you could make PINN create a uefi firmware partition and an esp partition and fill it with grub or clover or something that could boot windows from a project space. Didn't think to try it till now

Ae3NerdGod, 2021-02-12:
[Proof of concept](https://github.com/Ae3NerdGod/pinn-uefi/raw/main/WOPINN%20proof%20of%20concept.mp4)

UEFI on PINN starter kit: 

- REPO: add `repo=http://github.com/Ae3NerdGod/pinn-uefi/raw/main/uefi_list.json` to recovery.cmdline after PINN repo
- RELEASE: https://github.com/Ae3NerdGod/pinn-uefi/releases/tag/1.0

MysticWhiteDragon, 2021-02-13:
I was able to make an entry for PINN to boot the UEFI partition.  I can't get it to boot the Windows partition from the PINN installation drive. But if I put use another USB drive, the UEFI partition will boot and detect the UEFI partition on the second USB drive and boot into WIndows fine.  I would like for it to read the partition from the PINN drive, but I guess at least it does work this way.  I have an entry in PINN and m able to boot.

I'll make an entry on github for my method sometime soon.

Ae3NerdGod, 2021-02-13:
> I would like for it to read the partition from the PINN drive, but I guess at least it does work this way.  I have an entry in PINN and m able to boot.

I'm working on this, I'm having trouble figuring out how to get PINN to install an .img file

> I'll make an entry on github for my method sometime soon.

Please do, I'm starting with linux but I'm having the same type issue, single drive setup might take a specific process. If/when I get it working I'll make a short guide if necessary

Ae3NerdGod, 2021-02-13:
@procount this is the error I get when I try to install an .img using pinn

[image]

Do I need to change partition.json to indicate it's a raw img? 

Either that, or do you know of a way to set the ESP and boot flag, using `mkfs.fat`

lurch, 2021-02-13:
I've never even looked at Windows on ARM (and have no plans for doing so), but just replying to:

> Do I need to change partition.json to indicate it's a raw img?

Yes, you can specify a 'raw' partition type in `partitions.json` (which I believe is what e.g. [RiscOS does](http://packages.riscosopen.org/noobs/partitions.json), but note that this is just a raw _partition_, whereas an `.img` file is typically a raw _disk_ image, so you'll need to extract the partitions from the disk-image out into separate partition-images, if you want to write them with PINN. It's been a long time since I looked at all this stuff, but IIRC `kpartx` was useful for doing this kind of thing?

Ae3NerdGod, 2021-02-13:
@lurch i literally just remembered i had this discussion already https://github.com/procount/pinn/issues/420  and was researching riscos like you said last time, thank you.

i used dd and pulled a blank fat32 with ESP flag set from gparted. I made sure to use /dev/sdX**1** and zeroed the drive, made 2 ext4 partitons the same size as the ESP, and dd'd back onto the drive.  one became fat32 so im pretty sure i got JUST the partiton and not the table. and the ESP flag stayed set

so sounds like i just need to follow riscos's example and xzip it, remove the `.img` and update the json and ill be good to go

lurch, 2021-02-13:
I had to google what ESP was and I found https://en.wikipedia.org/wiki/EFI_system_partition
Remember that PINN doesn't support GPT, so the bit that'll be relevant for you is "its ID in the master boot record (MBR) partition-table scheme is 0xEF". I've got no idea what the "ESP flag" you mention is, but if it's this partition-ID, then note that this is set in the MBR (e.g. by `parted`) rather than in the partition-data itself. And before you ask, no I don't know how you'd do that in PINN, that'll be a question for @procount :wink:

procount, 2021-02-13:
I think there is a partition_type entry you can add to partitions.json. see my wiki.

Ae3NerdGod, 2021-02-13:
@lurch I didn't know the "Id inside the boot record" part. explains why grub-install can't find the esp or a valid mbr , but tianocore UEFi doesn't seem to care as long as there's an EFI folder

Other than the id in the MBR, ESP is just a fatXX partition with it's hex type changed to 0xEF from what I understand. 

I'll have to look into downloading a standalone copy `parted` to set the "flag" using partition setup or something, see if that works.

Ae3NerdGod, 2021-02-13:
@procount 
I see you have a json option for setting a specific partition number, and setting the partition active, and an offset. i wonder if it would be too much trouble for you to make one that sets the partition outside of the extended partition, and/or an option to put it at the end of the drive?
That's how most ESP partitions are placed on disk by automatic linux practitioners from what I've noticed, and it would be great for an ESP partition.
 
 PS, I noticed you're not using the `requires_label ` on ubuntu mate 20.10 at least (and i suspect vanilla ubuntu) available for pi 4 which you mention specifically in the json option description, and in a previous issue, that ubuntu requires lol. just pointing it out didn't think it merits an entire new issue

procount, 2021-02-14:
They were original NOOBS options. I think the partition number was for WindowsIoT, and the offset was either for the same, or most likely RiscOS. Partitions 3&4 are already outside the extended partition, but locating one at the end of the disk would require some thought (to check there are no implications)

It could be that I forgot to add that option. Thanks for pointing it out.

MysticWhiteDragon, 2021-02-14:
Using a second USB drive, I was able to boot Windows 10 on ARM.  My setup is linked [here](https://github.com/MysticWhiteDragon/pinn-winonpi).  Just copy the Windows10 folder to the os folder of PINN and copy the WIndows 10 on ARM "boot" partition to the partition that the script created on the PINN drive.

Ae3NerdGod, 2021-02-14:
@MysticWhiteDragon ive updated the repo with a new strategy in mind--shouldnt affect your method, but my way tricks pinn into leaving free space after its OS's extended partition, which ive had partial success installing debian to via the usb installer. windows might be trickier, but if youre willing to give it a try heres a basic guideline

 step1 UEFI PREP

- install uefi mode and as many OS stubs as you want -- these will be for PINN os's NOT UEFI OS's 
- when pinn completes the uefi mode setup, get back into pinn by hitting CTRL+ALT+F2 and then CTRL_ALT_DEL to reboot, then press shift to open pinn setup back up.

step 2 PINN OS INSTALLATION

- check the checkboxes for the OS's you want but DONT INSTALL. INSTEAD press m or click more at the top right of windows TWICE to get to **_maintenence mode_**
- check the checkboxes for the stubs you want to install the previously selected OS's to. then select replace. a menu will pop up, match the OS's to the stubs via the drop down boxes

step 3 - UEFI OS INSTALLATION
this is where it will get tricky for windows. you need to make a **primary** ESP partition, and a **extended** partiton for OS install(s)
i did this via the debian installer for linux, for windows id suggest booting one of the PINN os's you installed and using gparted to make the partitions. once youve made an ESP partition boot the windows installer and try to install it in a partition you make in the extended partition ( if it will let you) 

you MIGHT get away with two primary partitons if windows crys about an extended partition, but windows might also moan about not having its precious recovery partiton that cant fix anything except a damaged boot sector on legacy drives-- which it WOULD NOT BE ABLE TO BOOT FROM  _** THANKS MICROSOFT**_ _SO USEFUL. MUCH RECOVERY. VERY BOOT._

Ae3NerdGod, 2021-02-14:
@procount if you dont think its worth doing in PINN no biggie, i didnt realize parted could be used non-interactively until recently. ill learn how to do it in partition_setup.sh if it cant be done via PINN easily. just suggesting as a cleaner solution.

Ae3NerdGod, 2021-02-14:
also, hypothetically,  thought about making an issue for the pi4 eeprom asking them to allow a boot a start.elf off an ESP partiton. that way pinn could just be installed on the ESP. if so would you be willing to look into making PINN's recovery partition not freak out about it being type 0xEF fat (if it even does freak out, was unable to test.)

procount, 2021-02-14:
It sounds like it is more of a firmware boot feature rather than PINN.
I think I have some researching to do......

Ae3NerdGod, 2021-02-14:
If you mean booting off an esp partition, I agree, all PINN would have to do is ignore partition type and just assume it's partition is fat32. Which it might already do. it will mount correctly as fat32 is all I mean.

Ae3NerdGod, 2021-02-14:
@MysticWhiteDragon forgot to mention I changed the repo name (won't change again)

New repo url:

`repo=http://github.com/Ae3NerdGod/pinn-UEFI-Mode/raw/main/uefi_list.json `

No second release yet.

lurch, 2021-02-14:
> i didnt realize parted could be used non-interactively until recently. ill learn how to do it in partition_setup.sh if it cant be done via PINN easily.

I'm not quite following what you're trying to do, but note that (despite the name) `partition_setup.sh` is intended for modifying the _files_ on the filesystems for the respective OS, to tell the OS where its boot and root partitions are. It's not really intended for creating or modifying the raw partitions themselves.

Ae3NerdGod, 2021-02-15:
> I'm not quite following what you're trying to do, but note that (despite the name) partition_setup.sh is intended for modifying the files on the filesystems for the respective OS, to tell the OS where its boot and root partitions are.

im trying to create a uefi 'plugin' for PINN for lack of a better word. more accurately im trying to workaround PINN's (rightful) apathy in supporting a GPT via pinn ( only one device that pinn works on officially supports gpt) 
if i can get PINN to boot an EFI os FROM pinns boot menu, using a single sd/disk, that will be a complete success of this endeavor.

>It's not really intended for creating or modifying the raw partitions themselves.

and hopefully i wont have to end up using it that way. im in the experimenting stage, seeing what works, whats possible, whats not. i think if we could get JSON options to 1) create (outside of the extended partition) or boot from a **primary** ESP and 2) leave free space in the extended partition then id be able to do what ive set out to do. 

it seems one of the biggest issues is i cant get grub to boot from an ESP inside an extended partition, it just gets really confused and spits out tons of errors that contradict eachother

lurch, 2021-02-15:
I know very little about EFI, but I _think_ in a typical setup there'd only be a single ESP; and given that the main usage of EFI is with GPT-partitioned disks (which don't have extended / logical partitions), it wouldn't surprise me if the EFI System Partition on an MBR-partitioned disk "works best" when it's one of the 4 primary partitions. (and I guess it's also _probably_ a rarity to not have the ESP as the first partition of the disk? :wink: )
@procount Didn't Windows IOT have something to force one of it's partitions to be a primary partition, or am I misremembering and/or getting it confused with something else?

> leave free space in the extended partition

Could you use a ProjectSpace for that?

Ae3NerdGod, 2021-02-15:
ok, let me unpack the issue thoroughly, sorry if its long-winded:

> in a typical setup there'd only be a single ESP

i believe you are correct, I'm not trying to make multiple, just one outside the extended partition.

> and given that the main usage of EFI is with GPT-partitioned disks (which don't have extended / logical partitions), it wouldn't surprise me if the EFI System Partition on an MBR-partitioned disk "works best" when it's one of the 4 primary partitions.

so it seems unfortunately

>  (and I guess it's also probably a rarity to not have the ESP as the first partition of the disk? 😉 )

seemingly incorrect from my experiences, perhaps that was the standard at first, but every UEFI os I've installed recently puts it at the END of the disk now, i think the way GPT works its still partition 'number 1' but from my setups it doesn't seem to matter at all WHERE the ESP is on the disk, just that its there, has an EFI folder, and inside that folder the proper boot stuffs. i even noticed the latest windows 10 install disk put it at the end of the disk as well, unless i make specific partition changes in the installer.

TL;DR; with this UEFI environment (and most current ones it seems) neither placement, nor type (logical/primary) seems to matter to the FIRMWARE. its grub-install getting confused (confirmed via installer log), so i doubt windows would handle it well, also. 

 

> Could you use a ProjectSpace for that?

that's what i initially did, made uefi project spaces.

the problem is that the UEFI 'firmware' seems to be to some extent just chainloading a UEFI environment the same way qemu would, (i think from uboot in this case) that means if you're in EFI mode at all, the partition it resides in, and therefore the extended partition, is mounted for the EFI variables (boot entires, settings, custom drivers, the extensible stuff in **E**xtensible **F**irmware **I**nterface

due to some limitation of extended partitions, according to gparted, none of the logical partitions in a extended partition can be modiifed (size changed, deleted) if any ONE logical partition is inside the extended partition is mounted. pinn works around this by moving everything in the SETTINGS partition into ram and replacing it afterwards i believe, and doing most things from a ramdisk.

they can however, be created, it seems, which is why i specifically need empty space and not partitioned space.

Ae3NerdGod, 2021-02-15:
all that being said, my plan now that im realizing in that eeprom issue that the eeprom _can_ boot from ESP, and that i assumed wrong when it failed to boot an _extended_ ESP, then im going to see if 
1) installing pinn to a an empty ESP, changes it from esp during initial pinn setup
2) if i can change pinns fat32 to an ESP using a partition_setup.sh

@procount maybe itd be easier to make another recovery.cmdline option like it currentlyhas one for reserving empty space in the `RECOVERY` partition but instead reserve unpartitioned space in the extended partition? maybe `installreserve=(+)XXX `

Ae3NerdGod, 2021-02-15:
@procount 
the version of parted included seems to be missing the esp flag
[image]

https://www.gnu.org/software/parted/manual/html_node/set.html

any chance it could be updated?

procount, 2021-02-15:
> Ae3NerdGod:- make one that sets the partition outside of the extended partition, and/or an option to put it at the end of the drive?

You can already assign 2 out of 3 of the primary partitions (2,3, or 4) (PINN will move the extended partition accordingly between 2 and 4, I believe). But to create it at the end of the drive would need an additional option.

> Ae3NerdGod:- ( only one device that pinn works on officially supports gpt)

You mean the Pi4?

> Lurch:-  Didn't Windows IOT have something to force one of it's partitions to be a primary partition, or am I misremembering and/or getting it confused with something else?

Yes, Windows IoT required partitions 2 and 3, I believe.

> Ae3NerdGod:- pinn works around this by moving everything in the SETTINGS partition into ram and replacing it afterwards i believe, and doing most things from a ramdisk

Not quite. When PINN does an initial format, it removes every partition (extended, logical and primary) except partition 1. Depending on its size, this partition is then either resized to a minimal size, OR everything on it is copied to ramdisk and it is recreated from scratch before copying everything back. So this is not related to the settings partition, nor modifying the extended partition.

> Ae3NerdGod:- maybe it'd be easier to make another recovery.cmdline option like it currently has one for reserving empty space in the RECOVERY partition but instead reserve unpartitioned space in the extended partition? maybe installreserve=(+)XXX

Possible. Is this in addition to the empty space/partition requested to be outside the extended partition space?

> Ae3NerdGod:- the version of parted included seems to be missing the esp flag. any chance it could be updated?

Doubtful, as I think it will break a lot of PINN's functionality. Later versions of parted miss out some other crucial features, AFAIK.

Ae3NerdGod, 2021-02-15:
> You can already assign 2 out of 3 of the primary partitions (2,3, or 4) (PINN will move the extended partition accordingly between 2 and 4, I believe). But to create it at the end of the drive would need an additional option.

i was not aware, making it partiton 2 might suffice, its windows IOT that does this? ill look into it

> You mean the Pi4?

AFAIK, yes.

> Possible. Is this in addition to the empty space/partition requested to be outside the extended partition space?

i was able to make the `recovery` partition the ESP form the looks of it, so the extra partition outside the extended isnt necessary.

But reserving space inside the extended partition would help i think, im in the middle of actually testing its usefulness but im fairly certain it would. 

> Doubtful, as I think it will break a lot of PINN's functionality. Later versions of parted miss out some other crucial features, AFAIK.

thats a drag, however i was able to pipe commands into fdisk and achieve the result i needed, so no worries

procount, 2021-02-15:
It is Windows IoT that REQUIRES it, but it informs PINN to do it via partitions.json.

Adding a ProjectSpace is not sufficient to reserve space in the extended partition? I guess this is a location issue and need to be empoty-space? So that's the only thing you need at the moment?

lurch, 2021-02-15:
Yeah IIRC some of the later versions of parted remove various partition-resizing functions (although as mentioned before, it's a **long** time since I've looked at all of this!)

Ae3NerdGod, 2021-02-15:
> Adding a ProjectSpace is not sufficient to reserve space in the extended partition?

Not for windows, you can tell linux not to format anything but the windows installer insists on doing unnecessary stuff, idk what.

>I guess this is a location issue 

I'm guessing as well. When I try to use the windows installer on a pre-partitioned space, or in empty space after the extended partition (when I trick PINN into leaving some) the Install fails and gives a vague partition error.

>and need to be empty-space? So that's the only thing you need at the moment?

yes, although I found a workaround for now, its not very clean and involves making a partition just to delete it, unmounting everything and deleting, then remounting. Using partition setup.sh

rusagent, 2021-07-21:
> yes, although I found a workaround for now, its not very clean and involves making a partition just to delete it, unmounting everything and deleting, then remounting. Using partition setup.sh

Could you write out detailed how you did that please? I really would like to have win10 alongside of all the other unix systems on my ssd^^

Ae3NerdGod, 2021-07-28:
> Could you write out detailed how you did that please? I really would like to have win10 alongside of all the other unix systems on my ssd^^

Ive been pretty bogged down with life and work lately, havent had time for this project. however its worth mentioning it is very possible, and i had a working setup with a rpi native pinn-supplied os, a uefi linux install, and a windows install all on PINN's bootloader menu and they all worked.

Ae3NerdGod, 2021-07-28:
very basic unclean way IIRC:

-Install any pinn OS you want and some project spaces
-put the sd card/usb stick in a working linux pc and shrink the extended partition (and partitions inside it) enough for an ESP uefi partition 
-put a copy of the Tianocore boot stuffs in each project space's boot partition and split each root partiton into a small fat32 for grub/bootmgr(windows) and the os root partition
--this is where it gets weird
--make the project space bootable, boot one and install an UEFI os
--configure that specific Tianocore Bios to boot from the correct grub/bootmgr partiton-- there was a trick to this but i dont remember it. good luck
--repeat for each uefi OS
--???
--Profit.

thats as much as i remember, and basically any management utility in pinn will probably mess this setup up, so id avoid it.

Niehztog, 2024-05-12:
As with the new Raspberry Pi 5 and its NVMe capabilities, having a multi boot setup including windows will become a priority for quite some people, I'd like to point out that "Windows 10/11 on ARM" definetely works fine with a MBR partition scheme (the UEFI partition needs to have the "active" flag though).
As stated in earlier comments there have been attempts to add Windows on ARM to PINN, for example here: https://github.com/Marcinoo97/pinn-uefi
I havn't tried it yet, but I would appreciate if someone with more dedication and knowledge than me would provide step by step instructions on how to set it up.