procount/pinn #482: How to flash so its possible to Create a Multi Boot SD

mop999x, 2021-02-09:
i am going through this to create a multi boot sd which had Ubuntu and Kail....so here is the problem i flashed 2 different SD with IMG files i had on my hard drive both for arm64 and they both work on the PI4 individually....

unfortunately i cannot just install the OS directly from pinn, Kali does not exist and for some reason the Ubuntu that i have gotten through berryboot and similar seem to have a issue with the screen i have to use, the version i have of Ubuntu seem to work with the display so those are some constraints even when using the berry boot with the 2 os i have only ubuntu displays not kali for some reason so i wanted to try pinn since i seem to have more luck with it right from boot in terms of using display

so the issue is i used the RPI flasher to write the img to SD both ubantu and kali but in the instruction on 

https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN

it say for step 2 under **Backup Raspbian** that there should be 2 partitions root/boot if you flash with RPI flasher or Etcher. It only shows you the boot partition mounted not the root hence cant seem to get past step 2

**1)** so i was wondering do i have to add default OS that Pinn provides or Noobs provides in order for to be able to create Multi Boot SD card as per the instruction on the link above? and if i flash a OS that is no in the list in Pinn or Noobs i am out of luck? do i need to flash it with a different program?

basicaly if i have flash 2 OS That do not show root partition only boot can i Multi Boot SD card (also using win 10 to see if both partition show up as mount)

**1a)** does Linux have a way to show the second root partition some how even if win 10 does not?

**2)** is it possible to stick a img file on a USB and have pinn read the usb and put the OS on the SD to make a duel boot SD.....i imagen this is not available yet but maybe i don't know some thing?

procount, 2021-02-09:
Err.... You're going about this all wrong.
PINN does not support an OS that was created for BerryBoot. 
It does not support .img files.
Any OS you want first needs to be converted into .tar.xz format so that PINN can install it.

I have done this for about 60 OSes already, including Ubuntu and Kali.
What you should do is:
1. Install PINN to your SD card.
2. Connect your RPi to the internet and boot PINN.
3. Select Ubuntu and Kali and install them to your SD card.
4. Add any specific drivers you need for your displays to each of those installed OSes.

You really don't want to go down the https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN route unless there is no other way. But in this case I have already done the hard work for you,so just use the provided Ubuntu and Kali images.

lurch, 2021-02-09:
I don't mean this disrespectfully, but I think https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN is intended for advanced users, and from your questions it seems like you don't have that level of knowledge yet?

So yes, I think in your case you should just stick to the OSes that PINN offers you, and not try to do anything "clever" by "flash it with a different program".

> does Linux have a way to show the second root partition some how even if win 10 does not?

Yes, Linux supports the ext4 filesystem typically used on a root partition, but Windows doesn't support ext4.

mop999x, 2021-02-10:
> Err.... You're going about this all wrong.
> PINN does not support an OS that was created for BerryBoot.
> It does not support .img files.
> Any OS you want first needs to be converted into .tar.xz format [...]

**ya i figured you cant use Barry boot image with pinn, being able to use a img file to install was more wishful thinking on my part....so you  are suggesting either install the rpi4 versions of Ubuntu and kail vie web option on PINN or insert the ones you have created** 

> I don't mean this disrespectfully, but I think https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN is intended for advanced users, and from [...]

**ya i am new, i am mostly doing things the hard way to learn new stuff, found the whole multi boot thing interesting..

so based on what you are saying if i stick the sd card in to the usb adapter and stick that into a Linux distro on PC (virtualbox) or RPI4. The root and the boot should show up and i may have to mount them?  i ask because i have ordered the adapter but dont have it yet, and if everything goes good i can start compression and insert of json files etc....

so ya the issue i am having was and wanted to figure out the answer to was how to get the boot and root partition to show and mount, i only saw the boot on windows and was confused a bit since i don't own the adapter.....perhaps i should have been clear about the issue i was having. If there is more confusion, both of you, i can try to clear it up let me know. 

thanks for the help really both of you, cant wait to get away from windows a bit**

lurch, 2021-02-10:
No need to mess about with web URLs or localhost files. Simply download PINN and write it to an SD card, boot it up in your RPi, connect it to the internet using either Ethernet or Wifi, and PINN will present you with a list of OSes (compatible with your model of RPi) that it can then download, install and boot for you. I think you might have been underestimating how easy and automated PINN makes the whole process? :grinning:  No need for manually mounting partitions or editing JSON files etc...

> so based on what you are saying if i stick the sd card(using the win10 laptop built in sd reader so far) in to the usb adapter and stick that into a Linux distro on PC (virtualbox) or RPI4. The root and the boot should show up and i may have to mount them?

Possibly. I don't have Windows on any of my computers any more, but _if_ it allows VirtualBox to get "raw access" to the USB SD card-reader (rather than going through the Windows filesystem), then yes that should work.
An alternative approach is to write a LiveLinux distro (e.g. Ubuntu) onto a USB flash drive using Etcher, and then use that to boot your laptop into Linux (it'll boot entirely off the USB and won't touch your HDD) and that'll then let you access your laptop's SD reader from within Linux.

> found the whole multi boot thing interesting.

As I mentioned in the other issue, have a look at https://github.com/raspberrypi/noobs/wiki/Standalone-partitioning-explained and https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained if you want to understand how it all works at a low-level.
But NOOBS and PINN are designed so that you don't _need_ to understand how it all works, it just does everything for you automatically.

jharris1993, 2021-03-19:
Having completed some extensive research on this exact issue, there are two main pathways here:

1.  If all you want to do is install these O/S's on a suitably sized SD card, the suggestion to install PINN and then choose the OS's to install is the way to go. PINN does all the hard work and makes everything nice and tidy.

2. If you want to use a pre-existing and/or specifically pre-configured O/S, (assuming a raspbian/debian/ubuntu flavored distro here), you can spin your own PINNified package using the link on how to create a custom package for multibooting.

The original PINN readme.md (PINN is not NOOBS), also contains a good description on how to create a custom package.  Procount has recently updated the instructions so that it should be rather painless if you don't mind getting your hands dirty.

As you have discovered, PINN uses a special way of packaging the O/S to install that includes not only the OS itself, but specific metadata that allows PINN to know how to install it.  This is similar to the way package maintainers create .deb or .rpm packages - the software itself along with enough information so that the system knows what it's doing.

procount, 2021-04-07:
Closing due to inactivity

closed 2021-04-07 by procount